espressif/arduino-esp32 #4851: xSemaphoreCreateRecursiveMutexStatic cannot be used due to missing xQueueCreateMutexStatic

gsuberland, 2021-02-23:
### Hardware:
Board:							NodeMCU-32S (ESP32)
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Computer OS:                                        Windows 10

### Description:
Attempting to use [`xSemaphoreCreateRecursiveMutexStatic`](https://www.freertos.org/xSemaphoreCreateRecursiveMutexStatic.html) in a sketch throws a linker error: 

```
\Temp\arduino_build_484718\sketch\horus.ino.cpp.o:(.literal._Z5setupv+0x48): undefined reference to `xQueueCreateMutexStatic'
\Temp\arduino_build_484718\sketch\horus.ino.cpp.o: In function `setup()':
\Temp\arduino_build_484718\sketch/configurable.h:183: undefined reference to `xQueueCreateMutexStatic'
collect2.exe: error: ld returned 1 exit status
exit status 1
```

The `configurable.h` script has no code at line 183; it's the end of the file. I checked this in the actual file in the temp folder. This appears to be a red herring. The issue is called when attempting to use the `xSemaphoreCreateRecursiveMutexStatic` function.

Support for this FreeRTOS feature is supposed to be enabled by setting [`configSUPPORT_STATIC_ALLOCATION`](https://www.freertos.org/a00110.html#configSUPPORT_STATIC_ALLOCATION) and [`configUSE_RECURSIVE_MUTEXES`](https://www.freertos.org/a00110.html#configUSE_RECURSIVE_MUTEXES) in the build flags. I did this via `platform.local.txt`, along with some other flags I needed:

```
extras.defines=-DCONFIG_SUPPORT_STATIC_ALLOCATION=1 -DconfigSUPPORT_STATIC_ALLOCATION=1 -DconfigUSE_MUTEXES=1 -DconfigUSE_RECURSIVE_MUTEXES=1 -DconfigUSE_COUNTING_SEMAPHORES=1
build.defines={extras.defines}
compiler.cpp.extra_flags={extras.defines}
compiler.c.elf.extra_flags={extras.defines}
compiler.c.extra_flags={extras.defines}
compiler.S.extra_flags={extras.defines}
```

This should enable both static and recursive semaphores, and all FreeRTOS reference states that this should work. I can see that the flags are passed as part of the compilation output.

Looking into the code here, the actual implementation of `xQueueCreateMutexStatic` seems to be missing. There's a forward declaration [in queue.h](https://github.com/espressif/arduino-esp32/blob/a0ddd8a16e744b04ee58a7015742b582bc11eea7/tools/sdk/include/freertos/freertos/queue.h#L1401) but nothing else. The main FreeRTOS code branch includes the actual implementation.

Note that the dynamic recursive version of the function, `xSemaphoreCreateRecursiveMutex`, does appear to be present and working when these config flags are set. It's just the static ones that are missing.

### Sketch: 
```cpp

#include <Arduino.h>
#include <freertos/freertos.h>
#include <freertos/queue.h>
#include <freertos/semphr.h>

SemaphoreHandle_t mutexHandle;
StaticSemaphore_t mutexStatic;

void setup() {
    mutexHandle = xSemaphoreCreateRecursiveMutexStatic(&mutexStatic);
}

void loop() {
}
```

phantomkingx, 2021-03-11:
Seems to me that there is no way any to dynamically build freertos configuration since the assemblies are pre-built as /tools/sdk/lib/libfreertos.a and there is no freertos source code provided anywhere except for the header files in /tools/sdk/include/freertos/freertos/.  Or am I missing something?

igrr, 2021-03-11:
This can probably be fixed next time Arduino libraries are built, by enabling CONFIG_FREERTOS_SUPPORT_STATIC_ALLOCATION in sdkconfig used by esp32-arduino-lib-builder. @me-no-dev WDYT?

me-no-dev, 2021-03-11:
yup. on my to-do list

me-no-dev, 2021-03-17:
the flag has been enabled in the lib builder and we are waiting for IDF v3.3.5 to be released, so the builder can pick it up and send the update to https://github.com/espressif/arduino-esp32/pull/4936 for merging :) When that happens, Arduino 1.0.6 will be released as well

me-no-dev, 2021-03-18:
Change is now in the `idf-release/v3.3` branch of this repository. Could you please test and confirm that everything is OK?

me-no-dev, 2021-03-26:
https://github.com/espressif/arduino-esp32/releases/tag/1.0.6 includes CONFIG_FREERTOS_SUPPORT_STATIC_ALLOCATION

closed 2021-03-26 by me-no-dev