espressif/arduino-esp32 #4852: Add the option to disable Bluedroid in the sdkconfig

dominsch, 2021-02-23:
I have a working port of the ArduinoBLE stack running on the esp32 in the ESP-IDF, which can be used as an easier alternative to bluedroid or nimble. However it doesn't work in the Arduino IDE because bluedroid is initialized by default.
I would like there to be a way to set the bluetooth host to controller_only in the sdkconfig or alternatively a way to deinitialize bluedroid and initialize my own stack.
I have tried using the esp_bt_controller_deinit() and esp_bt_mem_release() functions but that didn't solve the problem.

### Hardware:
Board:							generic
Core Installation version:			1.0.5
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:				       any
Computer OS:                                        Windows 10

### Description:
The Arduino IDE has bluedroid enabled by default.

### Sketch:  
```cpp

#include "esp_bt.h"

void setup() {
  Serial.begin(115200);
  delay(1000);
  esp_err_t ret;
  esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
  if ((ret = esp_bt_controller_init(&bt_cfg)) != ESP_OK) {
      Serial.printf("Bluetooth controller initialize failed: %s\r\n", esp_err_to_name(ret));
      return;
  }
  Serial.printf("Bluetooth controller initialized\r\n");
}

void loop() {
}
```

### Debug Messages:
```
Bluetooth controller initialize failed: ESP_ERR_INVALID_STATE
```

chegewara, 2021-02-23:
1. there is no way to control any option in sdkconfig using arduino IDE, all is precompiled in idf libraries,
2. if it would be disabled in sdkconfig you could not use ble at all,
3. try this:
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-bt.c#L34-L80

dominsch, 2021-02-23:
Thanks for the quick response.
I figured it out by looking at how the nimble library does it.
In the file NimBLEDevice.cpp they have the following lines:

```C
#ifdef ARDUINO_ARCH_ESP32
        // make sure the linker includes esp32-hal-bt.c so ardruino init doesn't release BLE memory.
        btStarted();
#endif
```
I don't know what exactly that does but my code works now.

```C
#include "esp_bt.h"
#include "esp32-hal-bt.h"

btStarted();
esp_bt_controller_mem_release(ESP_BT_MODE_CLASSIC_BT);
esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
bt_cfg.mode = ESP_BT_MODE_BLE; // this line is important
esp_bt_controller_init(&bt_cfg);
esp_bt_controller_enable(ESP_BT_MODE_BLE); // has to match the config

closed 2021-02-23 by dominsch