espressif/arduino-esp32 #4855: WebServer.h fails to redirect

wuzamarine, 2021-02-24:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do? submit html form
Describe your system( Hardware, computer, O/S, core version, environment). Linux Android and SUSE
Describe what is failing. 
server.sendHeader("Location", String("/"), true);
fails intermittently. mostly fail. 

Show the shortest possible code that will duplicate the error. https://controlc.com/c7032fa7
Show the EXACT error message(it doesn't work is not enough). https://i.imgur.com/TZyO9Wc.png
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.
https://arduino.stackexchange.com/posts/81632/edit

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.8.2
IDE name:						Arduino IDE
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					115200
Computer OS:                                        Ubuntu, SUSE

### Description:
Describe your problem here

I am trying to create a simple form and upon user clicking submit, connection will reset "most of the time" before user is redirected.

server.sendHeader("Location", String("/"), true);

appears to fail 80% of the time

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <IRremote.h>
    
    //IRsend IrSender;
    
    // On the Zero and others we switch explicitly to SerialUSB
    #if defined(ARDUINO_ARCH_SAMD)
    #define Serial SerialUSB
    #endif
    
    int led_pin = 3; //pin is set to d4 on board but pin 3 here. Just the way it works out.
    
    //new networking stuff
    
    #include <WiFi.h>
    #include <WiFiClient.h>
    #include <WebServer.h>
    #include <ESPmDNS.h>
    
    const char *ssid = "";
    const char *password = "";
    
    //used for counting cycles later. Safety feature for overwatering
    int CYCLE0;
    int CYCLE1;
    
    
    WebServer server(80);
    
    // Current time
    unsigned long currentTime = millis();
[... 139 lines cut ...]
        });
        server.onNotFound(handleNotFound);
        server.begin();
        Serial.println("HTTP server started");
    
    }
    
    void loop() {
        
        //new networking stuff
        server.handleClient();
        Serial.print("\nesp32s infrared AC server device 2");
    
        //esp_task_wdt_reset();
        delay(1500);
    }
### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```
https://i.imgur.com/TZyO9Wc.png

The connection was reset

The connection to the server was reset while the page was loading.

me-no-dev, 2021-02-24:
you forgot the output on the serial port with debug enabled :)

wuzamarine, 2021-02-24:
@me-no-dev 
Forgive me but, how do you enable verbose debugging on the serial port? I have never seen this done before.

The connection reset for the URL is the most verbose output I can find.

Here is the serial output.

10:37:33.303 -> esp32s infrared AC server device 2
10:37:34.795 -> esp32s infrared AC server device 2
10:37:36.287 -> esp32s infrared AC server device 2
10:37:37.813 -> esp32s infrared AC server device 2
10:37:39.306 -> esp32s infrared AC server device 2
10:37:40.799 -> esp32s infrared AC server device 2
10:37:42.291 -> esp32s infrared AC server device 2AC was turned off:	8000
10:37:43.888 -> 
10:37:43.922 -> esp32s infrared AC server device 2
10:37:45.414 -> esp32s infrared AC server device 2
10:37:46.907 -> esp32s infrared AC server device 2
10:37:48.399 -> esp32s infrared AC server device 2
10:37:49.926 -> esp32s infrared AC server device 2

It just dies in the background with no error but what's in the URL.

If you can tell me what debuggers you would like, I'll be more than happy to run them. I just need to know what they are.

There is a wireshark image of a capture session in the URL presented.

lbernstone, 2021-02-24:
Set your board to "ESP32 Dev Module", and you will see the core debug menu under tools.
You should set `WiFi.setSleep(false);` for a device you intend to use as for web service.  This should take about 750-1000ms off your response times.

me-no-dev, 2021-02-24:
@lbernstone there is a `delay(2500);` in the handler :)

wuzamarine, 2021-02-24:
@me-no-dev  Thank you so much

closed 2021-07-09 by stale[bot]