espressif/arduino-esp32 #4857: How to check that a file has been opened successfully in the SD library?

JRVeale, 2021-02-24:
### Hardware:
Board:							Huzzah32
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						Platform.io
Flash Frequency:					?40Mhz?
PSRAM enabled:                                      ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                         Windows 10

### Description:
Hi,

I've always thought that the way to check that an `SD.open()` was succesful was as follows.

```
File myFile = SD.open("/path/to/file.txt", FILE_WRITE);
if(!myFile)
{
    // Oh no, myFile failed to open successfully
}
```

This is corroborated by all the examples I find online, and by the examples within the SD library. The Arduino documentation states that a `File` will evaluate as `false` if the open hasn't been successful - so I'd expect the above to work.

However, if I try to test this by removing my microSD card between writes (say with the esp32 set to write to the file every 2 seconds), the above code doesn't pick up the error at all. I get a serial print of the error from _vfs_api.cpp_ letting me know that the open has failed, but my `if(!myFile)` test doesn't work and so my code doesn't spot the error at all.

Is this a bug, or a specific case in which the above check isn't meant to work?

Is there a working method of checking if file opening has been successful?

Thanks!

me-no-dev, 2021-02-25:
it does sound like a bug. please post example code and uart output.

JRVeale, 2021-02-25:
Thanks for the quick reply. Here's some test code. The delays are to allow time to eject the SD card where needed for testing.

```
#include <Arduino.h>

#include "SPI.h"
#include "SD.h"

#define CARD_DETECT_PIN GPIO_NUM_34
#define CHIP_SELECT_PIN GPIO_NUM_4
#define TEST_FILE "/test.txt"

void setup()
{
  Serial.begin(115200);
  SPI.begin();

  pinMode(CARD_DETECT_PIN, INPUT);

  Serial.print("ver: ");
  Serial.println(esp_get_idf_version());

  Serial.println("Checking card present");
  // Check card is present
  if (digitalRead(CARD_DETECT_PIN))
  {
    // No SD Card
    Serial.println("Setup()\tNo SD Card present");
    ESP_ERROR_CHECK(ESP_FAIL);
  }

  delay(2000);

  Serial.println("Initing SD");
  if (!SD.begin(CHIP_SELECT_PIN))
  {
    Serial.println("Setup()\tSD init failed.");
    ESP_ERROR_CHECK(ESP_FAIL);
  }

  delay(2000);

  // Create file
  Serial.println("Creating file");
  File file = SD.open(TEST_FILE, FILE_WRITE);
  if (!file)
  {
    Serial.println("File failed to create");
    ESP_ERROR_CHECK(ESP_FAIL);
  }
  file.print("\n\n\nHeader\n");
  file.close();
}

int i = 0;

void loop()
{
  Serial.print("Writing line ");
  Serial.println(i);
  File file = SD.open(TEST_FILE, FILE_APPEND);
  if (!file)
  {
    Serial.println("File failed to open");
  }
  else
  {
    file.print("Line ");
    file.println(i);
    file.close();
  }

  delay(1000);
  i++;
}
```

Here's my UART output (I ejected the SD card after printing line 9):

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
ver: v3.2.3-14-gd3e562907
Checking card present
Initing SD
Creating file
Writing line 0
Writing line 1
Writing line 2
Writing line 3
Writing line 4
Writing line 5
Writing line 6
Writing line 7
Writing line 8
Writing line 9
Writing line 10
Writing line 11
Writing line 12
Writing line 13
Writing line 14
Writing line 15
Writing line 16
Writing line 17
Writing line 18
Writing line 19
Writing line 20
Writing line 21

```

And here's the contents of the file on the SD card:

```

Header
Line 0
Line 1
Line 2
Line 3
Line 4
Line 5
Line 6
Line 7
Line 8
Line 9

```

Interestingly, this minimal example doesn't even evoke the error logging prints from _vfs_api.cpp_, my actual code however does and I get the "fopen(FILE_NAME) failed" error prints over UART

me-no-dev, 2021-03-01:
this should now be fixed here: https://github.com/espressif/arduino-esp32/commit/1cf1c8eb79415cbf4e98da19cf1ca60564fc3d83

Please check and confirm :)

hwang59, 2021-03-09:
Could you also commit this to the idf-v4.2 branch for the SD.h file?

me-no-dev, 2021-03-09:
4.2 will be merged into master soon and the change will propagate itself :)

closed 2021-07-09 by stale[bot]