espressif/arduino-esp32 #4860: Failed assertion on esp_sleep_get_touchpad_wakeup_status()

Leo-C, 2021-02-25:
Hi,
I have a strange behaviour: changing the order of some settings in a demo included into Arduino ESP32 package, code fail.

### System

Board: ESP32 TTGO T-call
IDE name: Arduino IDE 1.8.13
Board library: ESP32 v1.0.5
Flash Frequency: 80Mhz
Upload Speed: 921600
Computer OS: Windows 10

### Tested Sketch

```
/*
Deep Sleep with Touch Wake Up
=====================================
This code displays how to use deep sleep with
a touch as a wake up source and how to store data in
RTC memory to use it over reboots

This code is under Public Domain License.

Author:
Pranav Cherukupalli <cherukupallip@gmail.com>
*/

#define Threshold 40 /* Greater the value, more the sensitivity */

RTC_DATA_ATTR int bootCount = 0;
touch_pad_t touchPin;
bool before = true;

/*
Method to print the reason by which ESP32
has been awaken from sleep
*/
void print_wakeup_reason(){
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch(wakeup_reason)
  {
    case ESP_SLEEP_WAKEUP_EXT0 : Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case ESP_SLEEP_WAKEUP_EXT1 : Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case ESP_SLEEP_WAKEUP_TIMER : Serial.println("Wakeup caused by timer"); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD : Serial.println("Wakeup caused by touchpad"); break;
    case ESP_SLEEP_WAKEUP_ULP : Serial.println("Wakeup caused by ULP program"); break;
    default : Serial.printf("Wakeup was not caused by deep sleep: %d\n",wakeup_reason); break;
  }
}

/*
Method to print the touchpad by which ESP32
has been awaken from sleep
*/
void print_wakeup_touchpad(){
  touchPin = esp_sleep_get_touchpad_wakeup_status();

  switch(touchPin)
  {
    case 0  : Serial.println("Touch detected on GPIO 4"); break;
    case 1  : Serial.println("Touch detected on GPIO 0"); break;
    case 2  : Serial.println("Touch detected on GPIO 2"); break;
    case 3  : Serial.println("Touch detected on GPIO 15"); break;
    case 4  : Serial.println("Touch detected on GPIO 13"); break;
    case 5  : Serial.println("Touch detected on GPIO 12"); break;
    case 6  : Serial.println("Touch detected on GPIO 14"); break;
    case 7  : Serial.println("Touch detected on GPIO 27"); break;
    case 8  : Serial.println("Touch detected on GPIO 33"); break;
    case 9  : Serial.println("Touch detected on GPIO 32"); break;
    default : Serial.println("Wakeup not by touchpad"); break;
  }
}

void callback(){
  //placeholder callback function
}

void setup(){
  Serial.begin(115200);
  delay(1000); //Take some time to open up the Serial Monitor
  
  //Increment boot number and print it every reboot
  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));
  
  //Setup interrupt on Touch Pad 3 (GPIO15)
  if (before)
    touchAttachInterrupt(T5, callback, Threshold);
  
  //Print the wakeup reason for ESP32 and touchpad too
  print_wakeup_reason();
  print_wakeup_touchpad();
  
  //Setup interrupt on Touch Pad 3 (GPIO15)
  if (!before)
    touchAttachInterrupt(T5, callback, Threshold);
  
  //Configure Touchpad as wakeup source
  esp_sleep_enable_touchpad_wakeup();
  
  //Go to sleep now
  Serial.println("Going to sleep now");
  esp_deep_sleep_start();
  Serial.println("This will never be printed");
}

void loop(){
  //This will never be reached
}
```

If `before` is `true`, then `esp_sleep_get_touchpad_wakeup_status()` contained in `print_wakeup_touchpad()` fail by following assertion:
`assertion "ret == ESP_OK && "wakeup reason is RTC_TOUCH_TRIG_EN but SENS_TOUCH_MEAS_EN is zero"" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/sleep_modes.c", line 460, function: esp_sleep_get_touchpad_wakeup_status`
apparently because `touchAttachInterrupt()` is called before `esp_sleep_get_touchpad_wakeup_status()`;
otherwise, if `before` is `false` (`touchAttachInterrupt()` is called after `esp_sleep_get_touchpad_wakeup_status()` like in the original example contained in Arduino ESP32 package), code works without problems.

Anyone can understand the root cause of this strange behaviuor?

Many thanks in advance,
Leonardo

chegewara, 2021-02-25:
Is there any reason you call attachInterrupt before checking wakeup reason?
Most likely attach interrupt is making internal calls to GPIO registers and its setting/clearing them.

Leo-C, 2021-02-25:
Hi,
some functions of this example are encapsulated in a class written by me, and method order calling is not so easy to define because can depends by many external factors.

If order can be inverted without problems, for me is easier ...

L

chegewara, 2021-02-25:
Sorry, but i cant see any reason to not check wakeup reason as the very first call after setup serial. Whatever is the reason of wakeup, you can continue in setup with your code based on result. You can check here what registers are responsible for this ssert:
https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf

Leo-C, 2021-02-25:
OK, thanks. I'll try to rearrrange my code.

But I still cannot understand why in text of assertion is stated that SENS_TOUCH_MEAS_EN is zero ...

L

chegewara, 2021-02-25:
Like i said, probably when you call attackInterrupts, in esp-idf ISR setup is causing to set this register value to 0.

closed 2021-07-09 by stale[bot]