espressif/arduino-esp32 #4865: WiFi connects successfully to... Nothing??

etherfi, 2021-02-27:
### Description:

I have an ESP32 DevBoard here. I have previously flashed it with code that successfully connects to my home network. For some reason, it seems to be storing it somewhere and remembering, even though this is a totally different sketch.

I realize a chip-erase will get rid of the saved WiFi configuration, but I want it not saved in the first place. How do I make it stop??

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
#include <WiFi.h>

void setup()
{
    Serial.begin(115200);
    delay(10);

    Serial.println("Connecting to Nothing!");
    WiFi.begin();

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected.");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
}

void loop()
{}
```

### Debug Messages:
```
19:15:44.193 -> ets Jun  8 2016 00:22:57
19:15:44.193 -> 
19:15:44.193 -> rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
19:15:44.239 -> configsip: 0, SPIWP:0xee
19:15:44.239 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
19:15:44.239 -> mode:DIO, clock div:1
19:15:44.239 -> load:0x3fff0018,len:4
19:15:44.239 -> load:0x3fff001c,len:1216
19:15:44.239 -> ho 0 tail 12 room 4
19:15:44.239 -> load:0x40078000,len:10944
19:15:44.239 -> load:0x40080400,len:6388
19:15:44.239 -> entry 0x400806b4
19:15:44.467 -> Connecting to Nothing!
19:15:44.558 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
19:15:44.602 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
19:15:44.695 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
19:15:44.695 -> [W][WiFiGeneric.cpp:390] _eventCallback(): Reason: 203 - ASSOC_FAIL
19:15:45.012 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
19:15:45.012 -> [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
19:15:45.012 -> [D][WiFiGeneric.cpp:418] _eventCallback(): STA IP: 192.168.1.102, MASK: 255.255.255.0, GW: 192.168.1.1
19:15:45.102 -> .
19:15:45.102 -> WiFi connected.
19:15:45.102 -> IP address: 
19:15:45.102 -> 192.168.1.102
```

me-no-dev, 2021-03-01:
calling `WiFi.disconnect(false, true);` before `WiFi.begin();`will do the trick. Not calling `WiFi.begin();` without arguments will also overwrite the stored credentials.

TheRealDJ, 2021-03-02:
The Wifi credentials are stored in the Wifi chip. Calling `WiFi.begin()` will use the previously set credentials.

lbernstone, 2021-03-02:
You can also call `WiFi.persistent(false);` before you connect to prevent it from saving credentials.

closed 2021-03-04 by etherfi