espressif/arduino-esp32 #4869: ESP32 WiFi.status() 1 trying to connect to Raspberry Pi ad-hoc network

jamcneilGH, 2021-02-28:
### Hardware:

|<B>Board</B> ESP32-WROVER
|<B>Version/Date</B>6/8/2016 (?)
|<B>IDE name</B>|Arduino IDE 1.8.13
|<B>Flash Frequency</B>80Mhz
|<B>PSRAM enabled</B>yes
|<B>Upload Speed</B>921600
|<B>Computer OS</B>Mac OSX 10.11.6

### Description:
I am trying to connect an ESP32-CAM or an ESP32-WROVER to a Raspberry Pi 4 (open) ad-hoc network using WiFi.begin(ssid). I have tested it with a modification of the WiFiScan sketch.  On the ESP32 side I have successfully connected to my home network (Xfinity supplied router with WPA2_PSK authentication) using WiFi.begin(ssid,passwd).  On the Raspberry Pi side I have successfully connected my Mac-Pro (OSX 10.11.6) to the ad-hoc network. I have also successfully connected another RPi to the other RPi ad-hoc network, but that required termination of the WPA supplicant client: sudo wpa-cli terminate.

### Sketch: 
// MODIFIED WiFiScan sketch to include a WiFi connection

#include "WiFi.h"
#include <esp_log.h>
#include <esp_wifi.h>
#include <esp_wifi_internal.h>

#define  WIFI_SSID "pi3" // SSID of Raspberry Pi ad-hoc network
#define WIFI_PASS "" // NOT USED FOR OPEN AD-HOC NETWORK

int iSSID = 99; // INDEX OF WIFI NETWORKS IN MY NEIGHBORHOOD

void setup()
{
    Serial.begin(115200);
    esp_log_level_set("wifi", ESP_LOG_VERBOSE);
    // Set WiFi to station mode and disconnect from an AP if it was previously connected
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(100);
    Serial.println("Setup done");
}

void loop()
{
    Serial.println("scan start");
    // WiFi.scanNetworks will return the number of networks found
    int n = WiFi.scanNetworks();
    Serial.println("scan done");
    if (n == 0) 
    {
        Serial.println("no networks found");
    } else 
    {
        Serial.print(n); Serial.println(" networks found");
        for (int i = 0; i < n; ++i) 
        {
            // Print SSID and RSSI for each network found
            String encryptionTypeDescription = translateEncryptionType(WiFi.encryptionType(i));          
            Serial.print(i + 1); Serial.print(": "); Serial.println(WiFi.SSID(i));
            Serial.print("      Signal strength: ");Serial.println(WiFi.RSSI(i));
            Serial.print("      MAC address: "); Serial.println(WiFi.BSSIDstr(i));
            Serial.print("      Encryption: "); Serial.println(encryptionTypeDescription);
            if(WiFi.SSID(i) == WIFI_SSID){iSSID = i;}
            delay(10);
        }
    }
    Serial.printf("  SSID index: %u\n",iSSID +1);

// Start WiFi....
  WiFi.begin(WIFI_SSID);
  Serial.println("Connecting Wifi...");
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
      delay(1000);
      Serial.print(" WiFi.status: ");Serial.println(WiFi.status());
  }
  Serial.println("");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
//          ... WiFi started
  delay(5000);
  WiFi.disconnect();
} 

String translateEncryptionType(wifi_auth_mode_t encryptionType) 
{
  switch (encryptionType) 
  {
    case (WIFI_AUTH_OPEN):
      return "Open";
    case (WIFI_AUTH_MAX):
      return "AUTH_MAX";
    case (WIFI_AUTH_WEP):
      return "WEP";
    case (WIFI_AUTH_WPA_PSK):
      return "WPA_PSK";
    case (WIFI_AUTH_WPA2_PSK):
      return "WPA2_PSK";
    case (WIFI_AUTH_WPA_WPA2_PSK):
      return "WPA_WPA2_PSK";
    case (WIFI_AUTH_WPA2_ENTERPRISE):
      return "WPA2_ENTERPRISE";
  }
  return "NONE";
}

### Serial Output with Log and Debug Messages:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[D][esp32-hal-psram.c:47] psramInit(): PSRAM enabled
I (1440) wifi: wifi driver task: 3ffb57a4, prio:23, stack:3584, core=0
I 997e4b
I (2933) wifi: config NVS flash: enabled
I (2933) wifi: config nano formating: disabled
I (2934) wifi(2933) wifi: wifi firmware version: 7: Init dynamic tx buffer num: 32
I (2938) wifi: Init data frame dynamic rx buffer num: 32
I (2943) wifi: Init management frame dynamic rx buffer num: 32
I (2949) wifi: Init management short buffer num: 32
I (2953) wifi: Init static rx buffer size: 1600
I (2958) wifi: Init static rx buffer num: 16
I (2961) wifi: Init dynamic rx buffer num: 32
I (3021) wifi: mode : sta (e8:db:84:12:7a:10)
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
Setup done
scan start
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 1 - SCAN_DONE
scan done
7 networks found
1: pi3
      Signal strength: -28
      MAC address: DC:A6:32:D0:AB:D1
      Encryption: Open
....
<removed info on other networks from this output>
....
SSID index: 1
Connecting Wifi...
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
 WiFi.status: 1
 WiFi.status: 1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
 WiFi.status: 1
 WiFi.status: 1
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
 WiFi.status: 1

....ad infinitum

jamcneilGH, 2021-03-01:
Update:  The issue persists with an ad-hoc network created on a Mac Pro using the "Create Network" function from the WiFi dropdown menu. To summarize the ESP32 can connect to the secured (WPA2_PSK) network but not the open ad-hoc networks. Any suggestions would be appreciated.

TheRealDJ, 2021-03-02:
I don't believe that you can connect an ESP32 to Ad-Hoc Wifi. Can't you setup the RPi as an Access Point?

jamcneilGH, 2021-03-03:
Thank you for the response. I am surprised and a bit puzzled that the ESP32 is able to connect to a secured network but not an open one. I have not read that anywhere. Do you have any documentation to which you can point me?  Thanks for the suggestion of setting up the RPi as an Access Point. I will need to research whether that will enable a solution to my ultimate goal of being able to transfer files (FTP) from the ESP to the RPi without needing a router. Thanks.

lbernstone, 2021-03-03:
:+1: on @TheRealDJ answer.  Use the Pi as an access point with hostapd.
An open network is not the same as an unbounded one.  Security is a completely separate issue.
Ad hoc networking is an unbounded (and uncontrolled) form of mesh networking.  This means that the information needed to maintain an ad hoc network grows factorially.  If you keep in mind that the esp32 only allows 4 clients on an AP due to memory constraints, it becomes clear why ad hoc is not supported.
esp32 supports a [proprietary mesh network](https://docs.espressif.com/projects/esp-idf/en/v3.3.4/api-reference/network/esp_mesh.html).  This allows you to build a network out organically in a large factory environment, while maintaining control of the flow patterns and keeping the routing bloat under control.

jamcneilGH, 2021-03-04:
Thanks for the information on Ad-hoc networks and ESP32.  I got the AP running on the RPi. It works fine with my Mac Pro (assigns and IP address with both ssh and ftp working) however, while the ESP32 now connects to the RPi's AP, the ftp service fails to authenticate (suggestions?) This is a new problem which should have it's own issue if I can't figure it out; so I plan to close this issue. Thanks for your assistance.

closed 2021-03-04 by jamcneilGH