espressif/arduino-esp32 #4871: Can't get ESP32 working with Arduino IDE on macOS Big Sur: xtensa-esp32-elf-g++: error

wouterds, 2021-02-28:
I keep getting this when trying to install a program to my Wemos Lolin32 Lite ESP32 development board on macOS Big Sur with latest version of the Arduino IDE. It works however on an older macOS device.

**Console output:**
```
xtensa-esp32-elf-g++: error: Lite: No such file or directory
xtensa-esp32-elf-g++: error: Lite": No such file or directory
exit status 1
Error compiling for board WEMOS LOLIN32 Lite.
```

**System:** macOS Big Sur 11.2.2
**Arduino IDE:** 1.8.13
**arduino-esp32:** 1.0.5

**Steps to reproduce:**
1. Download Arduino IDE
2. Install Arduino IDE app
3. Add ESP32 to boards manager
4. Install ESP32 boards
5. Select board type & serial port
6. Try to write basic blink program to board

Anyone that got it working on macOS Big Sur? Am I missing something?

(I recorded above steps to reproduce in case anyone could help out by just watching me do it https://youtu.be/IsvjP5e27ZQ)

wouterds, 2021-03-02:
Some extra information of my other machine where this is still working:

**System:** macOS Catalina 10.15.5
**Arduino IDE:** 1.8.13
**arduino-esp32:** 1.0.4

TheRealDJ, 2021-03-02:
There is an issue in the 1.0.5 **boards.txt** (as well as the master) with this device. There is a space in the `lolin32-lite.build.board` setting.
```
lolin32-lite.build.board=LOLIN32 Lite
``` 
should be changed to something without a space, like
```
lolin32-lite.build.board=LOLIN32_Lite
```

wouterds, 2021-03-06:
I can confirm that works, can we please release 1.0.6 with this fix? 🙏

me-no-dev, 2021-03-06:
@wouterds we will :)

TheRealDJ, 2021-06-30:
Looks like this was included in release 1.0.6 (https://github.com/espressif/arduino-esp32/releases/tag/1.0.6). @wouterds Can you confirm 1.0.6 works as expected then close the ticket?

closed 2021-09-19 by stale[bot]