espressif/arduino-esp32 #4881: ConnectionResetError: [Errno 104] Connection reset by peer : ota esp32

aszymkiw, 2021-03-02:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			?
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Ubuntu?

### Description:
Hello, 

I want to update my esp32 module via OTA. I use espota.py with parameters but i et a message : ConnectionResetError: [Errno 104] Connection reset by peer or i get as messaeg : error uploading if i let try except in the python program.

I don't know how to resolve, i didn't find my problem on forums, tthat's why i ask your help !

My esp32 gets an ip address, i arrive ping his IP. And before this error i have this logs : 

`09:26:41 [DEBUG]: Options: {'esp_ip': '192.168.1.13', 'host_ip': '0.0.0.0', 'esp_port': 3232, 'host_port': 43886, 'auth': '', 'image': '/home/aszymkiw/Arduino/sketch_jan06a/test3/test3.ino.esp32-gateway.bin', 'spiffs': False, 'debug': True, 'progress': True, 'timeout': 10}
09:26:41 [INFO]: Starting on 0.0.0.0:43886
09:26:41 [INFO]: Upload size: 1595088
Sending invitation to 192.168.1.13 
09:26:41 [INFO]: Waiting for device...
Uploading: [                                                            ] 0% Traceback (most recent call last):
`
Thank you for your help !!

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
//---OTA SETUP----------------------------------------------------------

  // Port defaults to 3232
  //ArduinoOTA.setPort(3232);

  // Hostname defaults to esp3232-[MAC]
  // ArduinoOTA.setHostname("myesp32");

  // No authentication by default
  // ArduinoOTA.setPassword("admin");

  // Password can be set with it's md5 value as well
  // MD5(admin) = 21232f297a57a5a743894a0e4a801fc3
  // ArduinoOTA.setPasswordHash("21232f297a57a5a743894a0e4a801fc3");

  ArduinoOTA.onStart([]() {
    otaRunning = true;
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH)
      type = "sketch";
    else // U_SPIFFS
      type = "filesystem";

    // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
    Serial.println("Start updating " + type);
  }).onEnd([]() {
    otaRunning = false; Serial.println("\nEnd");
  }).onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
    timerWrite(timer, 0); //reset timer (feed watchdog)
  })
  .onError([](ota_error_t error) {
    otaRunning = false;
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });
  ArduinoOTA.begin();
```

### Debug Messages:
ConnectionResetError: [Errno 104] Connection reset by peer

```

```

lbernstone, 2021-03-02:
1) Make sure you are running the latest release (1.0.5)
2) Increase timeout- `ArduinoOTA.setTimeout(3000);`
3) Set core debugging to verbose and post the detailed log.

aszymkiw, 2021-03-03:
Thank you for your answer, i resolved my problem : in fact i had to added the spiffs module.

In the console add the option : -s

closed 2021-03-03 by aszymkiw