espressif/arduino-esp32 #4887: I2C SDA and SCL Pull-Ups Can't Be Disabled

intakenick, 2021-03-03:
I've designed a custom PCB using the ESP-WROVER-IE module for very low power (battery) operation. 
In order to reduce quiescent draw, I've set up a peripheral power rail turned on/off by a GPIO-driven high-side mosfet.

The issue is this:

- There are I2C peripherals tied to the AUX Rail with external pullup resistors
- When the rail is on, everything is good, but the WROVER goes to deep sleep, the SCL and SDA pins remain at 3.3V which bleeds through the pullup-resistors into the AUX rail and causes unwanted current draw.

I have tried everything to disable the internal pullup resistors but can not seem to do it. Using the Wire library, one wire.begin() is called, the SCl and SDA gpios stay high until the board is reset, even in deep sleep.

Is there a function that will allow me to disable these pullups or at lease shut down the I2C bus before deep sleep?

chegewara, 2021-03-03:
Are you sure its not I2C device/sensor is pulling it up with on board hardware resistors?

intakenick, 2021-03-03:
Yes, when I remove the SSD1306 the power stops bleeding over (there are
onboard pullups to VCC).

Also, I put a pause in firmware before calling ssd1306.begin() and used
digitalWrite to set the pins low. This works fine, but once they are at
high, they stay high. Even in deep sleep.

--

Nick Johnson
Intake

(914) 906-3527 [mobile]

nick@myintakepro.com
------------------------------------------------------------------------------------------------

Automating preventive care. (myintakepro.com)

NOTICE OF TRANSMISSION: This email and any files transmitted with it are
confidential and intended solely for the use of the individual or entity to
whom they are addressed. If you are not the named addressee please notify
the sender immediately and delete this e-mail from your system. If you are
not the intended recipient you are notified that disclosing, copying,
distributing or taking any action in reliance on the contents of this
information is strictly prohibited.

chegewara, 2021-03-03:
With hardware pullups you can do nothing, for GPIO you can read about power domains:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html#power-down-of-rtc-peripherals-and-memories

intakenick, 2021-03-03:
Typed that wrong - the pull-ups are on the SSD1306 board and pull up to what it sees as VCC (in this case my aux rail). There is no hardware pullup to actual VCC - voltage is coming through the SDA/SCL lines and polluting the rail via it's pullups.  I'll will look at the power domains, thanks!

intakenick, 2021-03-03:
I've managed to solve this issue. Details here: https://arduino.stackexchange.com/questions/81759/cant-turn-off-internal-i2c-pull-ups-on-esp32/81775#81775

Basically, disable the I2C peripheral completely by calling this: 
```
    periph_module_disable(PERIPH_I2C0_MODULE);
    digitalWrite(SDA, 0);
    digitalWrite(SCL, 0);
```
Then just call display.begin() or wire.begin() (if not using the ssd1306) to restart the device.

stickbreaker, 2021-03-03:
@intakenick 
All you should need to do is call the destructor to release the resources allocated to `Wire()`, configure the SDA,SCL for your power down requirements, then after the Power Down recovery re-allocate and configure for your needs. 
```c++
// code before powerdown

~Wire(); // release all resources held by I2C hardware.

//  by default SDA and SCL are configured as INPUT | PULLUP to maintain valid I2C bus conditions 

PinMode(SDA,INPUT);
PinMode(SCL,INPUt);

// code after low power recovery , the Wire() object must be reconsructed:

Wire = TwoWire(0);
Wire.begin(SDA,SCL,clockSpeed);

```
Chuck.

closed 2021-03-04 by Kiznoh