espressif/arduino-esp32 #4888: ESP32-WROOM-32E - Secure Simple Pairing deactivated doesn't build!

RMOX97, 2021-03-03:
### Hardware:
Board:							ESP32-WROOM-32E
AT version:			                      2.2.0.0	
SDK version:				                  4.2	
Flash Frequency:					40Mhz 
Upload Speed:					115200
Computer OS:                                         Windows 10

### Description:
I would like to enable pairing pin in Classic Bluetooth SPP on ESP32.

After enabling the AT commands for bluetooth (via menuconfig) I would like to disable the "Secure Simple Pairing" in order to activate the PIN request for the bluetooth connection of a device.
the commands executed are the following:

_Component config --> AT --> [*] AT bt command support
Component config --> Bluetooth -->  Bluedroid Options --> [ ] Secure Simple Paring_

After saving the build fails. I attach below what it returns.

### Sketch
[SecureSimpleParing_ERRORI.txt](https://github.com/espressif/arduino-esp32/files/6075567/SecureSimpleParing_ERRORI.txt)

chegewara, 2021-03-03:
Are you sure its the right place to post this issue/ask this question?
In arduino IDE there is no menuconfig.

RMOX97, 2021-03-03:
Sorry I didn't notice. I moved it to the correct section

closed 2021-03-03 by RMOX97