espressif/arduino-esp32 #4889: Is there a reprogrammable ID option that can survice erase/ reprogramming (Question, not an issue)

kapyaar, 2021-03-03:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do 
Describe your system (Hardware, computer, O/S, core version, environment)  
Describe what is failing  
Show the shortest possible code that will duplicate the error  
Show the EXACT error message (it doesn't work is not enough)  
Then if someone is interested and knowledgeable you might get a answer.  All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
[ExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder)

----------------------------- Remove above -----------------------------

### Hardware:
|||||||
|:---|---|---|---|---|---|
|<B>Board</B>|ESP32 Dev Module|node32|ttgo_lora|ESP32-S2-Saola|Custom w/ ESP32-S2-WROVER 16MB|
|<B>Version/Date</B>|1.0.4|2.0.0|0badbeef|11/jul/2017|today's master|
|<B>IDE name</B>|Arduino IDE|Atom + Platform.io|IDF component|VSCode|
|<B>Flash Frequency</B>|40Mhz|80Mhz|
|<B>PSRAM enabled</B>|yes|no|
|<B>Upload Speed</B>|115200|
|<B>Computer OS</B>|Windows 10|Mac OSX|Ubuntu|

### Description:
This is a generic question, not an issue. 

I am working on a project that uses firmware and spiffs. There is a need to have a unique factory programmable (ideally reprogrammable) id. Something like eFuses. So far, my thought is that since I am not using bluetooth for my project, I could reprogram the BT MAC address with a unique ID (which hopefully?) would stay even when the board is erased/ reprogrammed on firmware and spiffs. Is this a reasonable approach? Or is there a better way?

Thanks for any help/ clues.

lbernstone, 2021-03-03:
The base mac address is unique on each device. https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Esp.cpp#L349-L354
[NVS/Preferences](https://github.com/espressif/arduino-esp32/tree/master/libraries/Preferences) does not get wiped when the firmware or spiffs is flashed, but it can get erased in serious error conditions.

kapyaar, 2021-03-03:
@lbernstone Thanks. Yes I did read about the base MAC and the incremental structure for other MAC addresses. That is ( so far) where I am heading to. I plan to use either BT or Ethernet MAC last octet. It would have been great if there was something similar without having to mess with MAC address as such. Hence the question.

closed 2021-03-04 by kapyaar