espressif/arduino-esp32 #4897: Why are there explanatory comments in the code, nor examples of reading data other than printing a string?

rowifi, 2021-03-05:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do  
Describe your system (Hardware, computer, O/S, core version, environment)  
Describe what is failing  
Show the shortest possible code that will duplicate the error  
Show the EXACT error message (it doesn't work is not enough)  
Then if someone is interested and knowledgeable you might get a answer.  All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
[ExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder)

----------------------------- Remove above -----------------------------

### Hardware:
|||||||
|:---|---|---|---|---|---|
|<B>Board</B>|ESP32 Dev Module|node32|ttgo_lora|ESP32-S2-Saola|Custom w/ ESP32-S2-WROVER 16MB|
|<B>Version/Date</B>|1.0.4|2.0.0|0badbeef|11/jul/2017|today's master|
|<B>IDE name</B>|Arduino IDE|Atom + Platform.io|IDF component|VSCode|
|<B>Flash Frequency</B>|40Mhz|80Mhz|
|<B>PSRAM enabled</B>|yes|no|
|<B>Upload Speed</B>|115200|
|<B>Computer OS</B>|Windows 10|Mac OSX|Ubuntu|

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

me-no-dev, 2021-03-05:
there are examples showing the full functionality. What are you looking for I am not sure? https://github.com/espressif/arduino-esp32/blob/master/libraries/AsyncUDP/examples/AsyncUDPClient/AsyncUDPClient.ino#L22

rowifi, 2021-03-05:
I'm trying to broadcast a byte array of known length to a defined port.
Using broadcastTo();
Also I was trying to uncouple the Onpacket callback function to be a separate function outside the initial setup   it would read clearer for me.

me-no-dev, 2021-03-05:
something like this?
```cpp
AsyncUDP udp;

void myOnPacket(AsyncUDPPacket packet) {
    Serial.print("UDP Packet Type: ");
    Serial.print(packet.isBroadcast()?"Broadcast":packet.isMulticast()?"Multicast":"Unicast");
    Serial.print(", From: ");
    Serial.print(packet.remoteIP());
    Serial.print(":");
    Serial.print(packet.remotePort());
    Serial.print(", To: ");
    Serial.print(packet.localIP());
    Serial.print(":");
    Serial.print(packet.localPort());
    Serial.print(", Length: ");
    Serial.print(packet.length());
    Serial.print(", Data: ");
    Serial.write(packet.data(), packet.length());
    Serial.println();
    //reply to the client
    packet.printf("Got %u bytes of data", packet.length());
}

bool doConnect(IPAddress& ip, uint16_t port) {
    if(udp.connect(IPAddress(192,168,1,100), 1234)) {
        udp.onPacket(myOnPacket);
        return true;
    }
    return false;
}

void setup()
{
    Serial.begin(115200);
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    if (WiFi.waitForConnectResult() != WL_CONNECTED) {
        Serial.println("WiFi Failed");
        while(1) {
            delay(1000);
        }
    }
    if(doConnect(IPAddress(192,168,1,100), 1234)) {
        Serial.println("UDP connected");
        //Send broadcast on port 1234
        udp.broadcastTo("Anyone here?", 1234);
    }
}
```

rowifi, 2021-03-05:
Ah.. Yes. Different from the server  think. Thanks.
But I Cant workout how to do something like this..    
```
int localPortout = 1234;
int localPortin = 1235;

udp.broadcastTo("Anyone here?",1234); // THIS WORKS

udp.broadcastTo( &UDPcomms.bytes[0], localPortout);  // ** Needs a size otherwise stops at zero null
```
And I'm using different ports for send and receive   so the class local port I think gets set when listening, need to be defined in the call?

rowifi, 2021-03-05:
solved.

closed 2021-03-05 by rowifi