espressif/arduino-esp32 #4898: MAJ OTA keeps the previous programm

aszymkiw, 2021-03-05:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.3
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Ubuntu

### Description:
Hello,

The last week i had a problem to maj in OTA but i must added a flag : SPIFFS.
I wanted today to try to maj but i saw that my gateway ESP32 keeps the same previous programm and don't take my bin file that i send it.

I use espota.py to make the update. And i see the loading and the percent without any error. At the end, i can read "Success".

Thank you for your help !

me-no-dev, 2021-03-05:
what is MAJ OTA? The flag SPIFFS means that it will update the SPIFFS partition, not the firmware

aszymkiw, 2021-03-05:
thank you for your answer.
Ha okay, i want to update my esp32 via OTA.

me-no-dev, 2021-03-05:
you should start with what you have done so far. What command did you execute, what was the output in terminal, what was the output of the serial port of the ESP? If you can't update it, that is because it's finding something wrong. What partition scheme did you select?

aszymkiw, 2021-03-05:
I executed the espota.py with the bin are generated by arduino and if i put -s : no error and if i don't write : error.

me-no-dev, 2021-03-05:
you gave me exactly none of the information I asked for :) I need exact command and the output of it. Also the log from the ESP itself and which partition scheme you used.

aszymkiw, 2021-03-05:
the command : python3 espota.py -i IP -f FILE.bin -r -d -p 3232 (if i don't add -s, it don't work)

the partition without -s in the commande none with the -s the spiffs.

I hope so that it is that you want (sorry for my english)

me-no-dev, 2021-03-05:
the command yes! thanks!

Now the partitions I asked you have selected in the board menu when you were building the sketch (or you used the default one). Also, please post the output of the command! And the log from the ESP itself! Asking for last time ;) If you do not provide all of the necessary information, there is no way to help. OTA is working fine on my end.

aszymkiw, 2021-03-05:
i use noota (large app), but i thought that it was possible to update this by a command.

And i don't have log.

me-no-dev, 2021-03-05:
Why do you think the partition scheme is named `noota ` :) it's because it does not have another partition for OTA to work ;) You can only update it through UART

closed 2021-03-05 by me-no-dev