espressif/arduino-esp32 #4899: ESP_IDF Arduino component errror during compiling

DheeranYuvaraj, 2021-03-05:
Hardware:
ESP32-Wroom-32D

Description:
I am trying to use Arduino components for ESP-IDF. While compiling i am getting the below error. After cloning latest  Arduino components only this error. Please check the error message under debug section. May i know how to solve this?  

I followed the official procedure available in github. [https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md](url)

### Sketch: 
```
#include "Arduino.h"

void setup() 
{
   Serial.begin(115200);
}
void loop()
{
   Serial.println("Hello World");
   delay(2000);
}
```

### Debug Messages:
```
C:/esp32/esp_idf-arduino/hello_world/components/arduino/libraries/Update/src/HttpsOTAUpdate.cpp: In member function 'void HttpsOTAUpdateClass::begin(const char*, const char*, bool)':

C:/esp32/esp_idf-arduino/hello_world/components/arduino/libraries/Update/src/HttpsOTAUpdate.cpp:91:12: error: 'struct esp_http_client_config_t' has no member named 'skip_cert_common_name_check'
     config.skip_cert_common_name_check = skip_cert_common_name_check;
            ^
make[1]: *** [/c/esp32/esp-idf-v3.3/make/component_wrapper.mk:290: libraries/Update/src//HttpsOTAUpdate.o] Error 1
make: *** [C:/esp32/esp-idf-v3.3/make/project.mk:552: component-arduino-build] Error 2

```

me-no-dev, 2021-03-05:
which version IDF?

DheeranYuvaraj, 2021-03-06:
@me-no-dev 

ESP-IDF-v3.3

lbernstone, 2021-03-06:
release version 1.0.5 uses IDF 3.3.4- [git commit 7a85334d8](https://github.com/espressif/arduino-esp32/blob/1.0.5/tools/sdk/include/config/sdkconfig.h#L399)
I have built a [docker container](https://hub.docker.com/r/lbernstone/esp32-arduino-lib-builder) which contains exactly (and only) the code necessary to build release 1.0.5.

DheeranYuvaraj, 2021-03-08:
Hi,

After switching to IDF 3.3.4 also error while compiling WiFiClientSecure libraries. Check the below debug message. 

`C:/esp32/esp_idf-arduino/hello_world/components/arduino/libraries/WiFiClientSecure/src/ssl_client.cpp:23:4: error: #error "Please configure IDF framework to include mbedTLS -> Enable pre-shared-key ciphersuites and activate at least one cipher"
 #  error "Please configure IDF framework to include mbedTLS -> Enable pre-shared-key ciphersuites and activate at least one cipher"
    ^
C:/esp32/esp_idf-arduino/hello_world/components/arduino/libraries/WiFiClientSecure/src/ssl_client.cpp: In function 'int start_ssl_client(sslclient_context*, const char*, uint32_t, int, const char*, const char*, const char*, const char*, const char*, bool)':
C:/esp32/esp_idf-arduino/hello_world/components/arduino/libraries/WiFiClientSecure/src/ssl_client.cpp:165:67: error: 'mbedtls_ssl_conf_psk' was not declared in this scope
                  (const unsigned char *)pskIdent, strlen(pskIdent));
                                                                   ^
make[1]: *** [/c/esp32/esp-idf-v3.3.4/make/component_wrapper.mk:290: libraries/WiFiClientSecure/src//ssl_client.o] Error 1
make: *** [C:/esp32/esp-idf-v3.3.4/make/project.mk:552: component-arduino-build] Error 2
`

me-no-dev, 2021-03-08:
copy `tools/sdk/sdkconfig` to your project root and recompile. It will work

DheeranYuvaraj, 2021-03-09:
@me-no-dev 

In ESP-IDF-V3.3.4 under **tools** there is no folder named **sdk**.