espressif/arduino-esp32 #4901: How to do an i2c complex command and receive the response?

arjenhiemstra, 2021-03-05:
### Hardware:
Board:							ESP-WROOM-32D
Core Installation version:			1.0.5
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Mac OSX

### Description:
I would like to make a complex request to an i2c device and receive a reply. 
Sending the request works but how to receive the reply? May this i a stupid question and I'm completely overlooking something obvious but i've looked for a solution but can't find one. 

An approach I was thinking of is send the request, switch to slave with a callback to receive the data. This would work for the ESP8266 Arduino but seems not available for ESP32 Arduino but maybe I'm overlooking something.

### Sketch: 
```cpp
uint8_t command[] = { 0x82, 0x80, 0x90, 0xE0, 0x04, 0x00, 0x8A };

Wire.beginTransmission(byte(0x41));
for (uint8_t i = 1; i < sizeof(command); i++) {
  Wire.write(command[i]);
}
uint8_t err = Wire.endTransmission(true);

//response will be formatted like:
//0x80,0x82,0x90,0xE0,0x01,0x12,0x00,0x01,0x00,0x14,0x12,0x0B,0x00,0x00,0x00,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0x00,0x62

//code below does not receive any data, is there another way to receive the data?

char c[32] = {};
uint8_t i = 0;
  
Wire.requestFrom(byte(0x41), 25);
while (Wire.available()) {
  c[i] = Wire.read();
  i++;
  if (i > sizeof(c)) break;
}

```

stickbreaker, 2021-03-05:
@arjenhiemstra 
   I don't believe the Write transaction actually succeeded. you did not do any error checking,  Try this code:
```c++
// code with error checking 

uint8_t command[] = { 0x82, 0x80, 0x90, 0xE0, 0x04, 0x00, 0x8A };

Wire.beginTransmission(byte(0x41));
Wire.write(command,sizeof(command));
uint8_t err = Wire.endTransmission(true);
if(err != 0){ // error has occurred
  Serial.printf("I2C Write Failed = %d (%s)\n",err,Wire.getErrorText(err));
} 
else { // Successful command to device, now readback data

//response will be formatted like:
//0x80,0x82,0x90,0xE0,0x01,0x12,0x00,0x01,0x00,0x14,0x12,0x0B,0x00,0x00,0x00,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0x00,0x62

//code below does not receive any data, is there another way to receive the data?

   char c[32] = {};
   uint8_t i = 0;
  
   err = Wire.requestFrom(byte(0x41), 25); // err will either be 0(error) or the number of bytes received
   if(err==0) {
	  Serial.printf("I2C Read Failed = %d (%s)\n",Wire.lastError(),Wire.getErrorText(Wire.lastError()));
   }
   else { // err will never be greater than 25
        while (i< err ) {
            c[i] = Wire.read();
            i++;
        }
   }
}

```
Here is the same functionality using the block handling `Wire()` methods:
```c++
/* Optimized code with error checking, since you have your byte sequence defined in
  an array, use the block handling Wire methods
  */

uint8_t command[] = { 0x82, 0x80, 0x90, 0xE0, 0x04, 0x00, 0x8A };

i2c_err_t err = Wire.writeTransmission(0x41, command, sizeof(command),true);

/* i2c_err_t will return:  0 .. 8 
const char ERRORTEXT[] =
    "OK\0"     // successful
    "DEVICE\0"	// hardware failure in ESP32, usually to IRQ available (unusual)
    "ACK\0"		// Slave Device did not acknowledge I2C address
    "TIMEOUT\0"	// SCL held low longer than 12ms by slave
    "BUS\0"		// I2C bus was not valid ( usually returned because SCL or SDA are held low ,bus shored to ground
    "BUSY\0"	// I2C bus in use by other master device(external to esp32) or SLAVE is holding SCL
    "MEMORY\0"	// unable to allocate internal buffer
    "CONTINUE\0"	// Wire buffers commands until SEND_STOP is set to TRUE, the ESP32 cannot execute a ReSTART operation until a Wire.endTransmission(true), or Wire.requestFrom(id,true); 
    "NO_BEGIN\0" 	// Wire.write() issued without a Wire.begin()
    "\0";
*/

if(err != 0){ // error has occurred, report it.
  Serial.printf("I2C Write Failed = %d (%s)\n",err,Wire.getErrorText(err));
} 
else { // Successful command to device, now readback data

    char c[32];
    uint32_t count=0;
    err = Wire.readTransmission(0x41,c,sizeof(c),true,&count);
    if(err!=0) {
       Serial.printf("I2C Read Failed = %d (%s)\n",err,Wire.getErrorText(err));
    }
    else { 
     // data is in 'c' already, up to 32 bytes, actual number of bytes received is in 'count'
        Serial.printf(" I2C received %d bytes of data\n",count);
        uint32_t i = 0;
        while(i< count){
            Serial.printf(" c[%02d]=%02x (%c)\n",i,c[i],((c[i]>=32)&&(c[i]<=127)?c[i],'.'));
             i++;
        }
	
//response will be formatted like:
//0x80,0x82,0x90,0xE0,0x01,0x12,0x00,0x01,0x00,0x14,0x12,0x0B,0x00,0x00,0x00,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0xFF,0x00,0x62
}
```

Chuck.

arjenhiemstra, 2021-03-05:
Thanks a lot Chuck for the very elaborate answer! That's far more than I hoped for.

There is an error indeed: I2C Read Failed = 2 (ACK)

So it seems that the sending of the command went fine (which is confirmed by the fact that the target device processes on other i2c send only commands) but transmission is not being ACKed. Need to look into this further why.

arjenhiemstra, 2021-04-16:
Yesterday I finally found the issue. For others that run in a comparable situation, the reason was twofold:
1. this is a multi master setup, the device i'm trying te get working needs to send as master and receive the reply as slave, this is not easily possible with the Arduino-ESP32. Using the IDF functions could solve this but that didn't work either due to 
2. there was a clock stretching issue that caused the master write to fail. Adding i2c_set_timeout(i2c_port_num, 0xFFFFF) to the master write function solved this

closed 2021-04-16 by arjenhiemstra