espressif/arduino-esp32 #4903: GPIO36 "fake" falling interrupts

stuartpittaway, 2021-03-06:
### Hardware:
Board:							ESP32-WROOM-32U
Core Installation version:			PlatformIO 5.1.1a3
IDE name:						Platform.io
Computer OS:                                        Windows 10

### Description:
I'm using version 3.10005.210223 of platformio/framework-arduinoespressif32.  If I attach a falling interrupt to GPIO36 it will continuously trigger.  The pin is pulled high externally via a touch screen.  Verified with scope.

Swapping back to version 3.10004.210126 fixes this issue and the interrupt returns to normal, only triggering on genuine transitions of the pin.

Describe your system( Hardware, computer, O/S, core version, environment).
Windows 10, Platform IO

me-no-dev, 2021-03-06:
interesting... there has been no change to the code for GPIO and interrupts here. I would guess that there has been a change in the wifi driver that is causing the issue to show, or do you not have WiFi/ADC on?

felmue, 2021-03-06:
Hello guys

I noticed the same thing on an M5Paper device which uses GPIO36 as interrupt from the touch screen. I get constant 'ghost' interrupts.

And I can confirm that it only happens if WiFi is active. As soon as I invoke a `WiFi.disconnect()` the ghost interrupts go away. When I invoke a `WiFi.begin()` the issue starts again.

I only see the issue with framework-arduinoespressif32 - 3.10005.210223 (1.0.5).

With the previous version, e.g. framework-arduinoespressif32 - 3.10004.201016 (1.0.4) the issue goes away.

Thanks
Felix

me-no-dev, 2021-03-09:
This is in line with the hardware issue with IOs 36 and 39 mentioned in the ESP32-ECO and changes in the WiFi driver.

me-no-dev, 2021-03-09:
I saw somewhere mentioned that `WiFi.setSleep(false);` after connection is established could help.

felmue, 2021-03-09:
Hello @me-no-dev 

thank you for the information. I can confirm that disabling the WiFi sleep mode prevents the 'ghost' interrupts. E.g.
```
WiFi.setSleep(WIFI_PS_NONE);
```
Thanks
Felix

stuartpittaway, 2021-03-09:
> With the previous version, e.g. `framework-arduinoespressif32 - 3.10004.201016 (1.0.4)` the issue does not happen.

Same here, with SDK Version v3.2.3-14-gd3e562907 - everything is good.

@me-no-dev do you have a link to the hardware issue/bug reference you mentioned about the ESP-ECO?

me-no-dev, 2021-03-09:
@stuartpittaway https://www.espressif.com/sites/default/files/documentation/eco_and_workarounds_for_bugs_in_esp32_en.pdf

closed 2021-03-09 by stuartpittaway

innovationcentralph, 2022-10-21:
> Hello @me-no-dev
> 
> thank you for the information. I can confirm that disabling the WiFi sleep mode prevents the 'ghost' interrupts. E.g.
> 
> ```
> WiFi.setSleep(WIFI_PS_NONE);
> ```
> 
> Thanks Felix

Thank you so much for this. 

This solved my issue

rozrabiak, 2025-01-18:
Hi, above solution not work for me.
Any ideas?
```
Chip is ESP32-D0WDQ6 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
```