espressif/arduino-esp32 #4904: CAN bus driver broken 1.0.4 -> 1.0.5

obdevel, 2021-03-06:
### Hardware:
Board:							DevkitC - see below
Core Installation version:			1.0.5
IDE name:						Arduino IDE 1.8.13
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        macOS 10.13.6

### Description:

Code that previously worked in 1.0.4 now fails in 1.0.5. 

Specifically, the can_driver_install() function returns ESP_ERR_INVALID_STATE, which implies the driver is already installed. The code compiles as expected but fails with this error at runtime.

Reverting to 1.0.4 (via boards manager) corrects the problem.

I know that Espressif has renamed the CAN driver to TWAI in recent versions of the IDF, probably to avoid paying the licence fee to Bosch :) This has resulted in renaming of functions, macros, structs, etc. This hasn't fed through to the IDF version underneath 1.0.4 or 1.0.5.

Thanks.
---------------

esptool.py v2.8-dev
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 3c:71:bf:f1:1e:f0
Uploading stub...
Running stub...
Stub running...
Warning: ESP32 has no Chip ID. Reading MAC instead.
MAC: 3c:71:bf:f1:1e:f0
Hard resetting via RTS pin...

me-no-dev, 2021-03-06:
CAN driver comes directly from IDF. In 1.0.5 this is from recent IDF v3.3. I suggest you post an issue there :)

obdevel, 2021-03-06:
Thanks. I've raised an issue there and I'll post again here if there is a recommended IDF version to fix this error. I really, really don't want to have to patch the Arduino core as I have non-technical users who would not be able to do this.

https://github.com/espressif/esp-idf/issues/6669

obdevel, 2021-06-20:
The CAN driver has been deprecated in the IDF in favour of TWAI. However, this not included with the current Arduino core (as at 1.0.6). As it stands, this core includes a driver that is no longer supported and no longer works.

(It's effectively the same code with a different name, presumably to avoid paying the licence. )

Can you base the next release on a copy of the IDF that is configured with the TWAI driver ? Thanks.

me-no-dev, 2021-06-22:
the latest beta is based on IDF master. Can't get newer than that :)

obdevel, 2021-06-22:
Does the TWAI driver need to be explicitly included or enabled inIDF ? 

The facts, as far as I understand them, because I don't use IDF:

- CAN driver worked perfectly in 1.0.4 and earlier
- Espressif changed the name from CAN to TWAI and deprecated the CAN driver, but it's the same chip peripheral
- CAN driver stopped working in 1.0.5 (see above)
- TWAI files not present in 1.0.5 (e.g. driver #includes) so I can't compile with this driver

All because Bosch charge a licence fee to use the name 'CAN' and Espressif want to avoid this :(

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/twai.html

closed 2021-09-06 by stale[bot]