espressif/arduino-esp32 #4905: v1.05 board library breaks a number of things

JimDrewGH, 2021-03-07:
Board: Any
Core Installation version: 1.05
IDE name: Arduino IDE

Since upgrading to v1.05 board library, none of the nifty video projects from Rossumur (https://github.com/rossumur), or any derivatives, will now work.  The ESP_8_Bit and ESPFlix are projects that output color composite video.  No video is now output at all with ESP_8_bit (but compilation is fine) and ESPFlix now generates this error when compiling:

`
C:\Users\Main\Documents\Arduino\ESPFLIX\espflix.ino: In function 'void wifi_join(const char*, const char*)':
espflix:203:54: error: cannot convert 'esp_interface_t' to 'wifi_interface_t' for argument '1' to 'esp_err_t esp_wifi_set_config(wifi_interface_t, wifi_config_t*)'
     esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config);
                                                      ^
C:\Users\Main\Documents\Arduino\ESPFLIX\espflix.ino: In function 'std::__cxx11::string wifi_ssid()':
espflix:212:53: error: cannot convert 'esp_interface_t' to 'wifi_interface_t' for argument '1' to 'esp_err_t esp_wifi_get_config(wifi_interface_t, wifi_config_t*)'
     esp_wifi_get_config(ESP_IF_WIFI_STA,&wifi_config);
                                                     ^
exit status 1
cannot convert 'esp_interface_t' to 'wifi_interface_t' for argument '1' to 'esp_err_t esp_wifi_set_config(wifi_interface_t, wifi_config_t*)'
`
Going back to v1.02, v1.03, or v1.04 everything works fine.

lbernstone, 2021-03-07:
Those are upstream changes.  The maintainer of those libraries will need to update their code, just as the maintainer of this repository had to make changes to work with the framework.

JimDrewGH, 2021-03-08:
The ESP_8_Bit app is hung in a loop much like other devices that have been recently described here in other reports.  I can't imagine this is something that has to be fixed by every programmer moving forwards.  One would expect a certain level of backwards compatibility with previous board versions.  Clearly things like the compile error are due to changes in the framework.  Where is that documented?

igrr, 2021-03-08:
@JimDrewGH Sorry for this issue, it was introduced in ESP-IDF for C++ code when changing the definition of wifi_interface_t. We haven't noticed it until recently because most of our test and example code in IDF is in C, where conversions between different enum types are allowed. We will find a solution for this issue and try to get it implemented in the next bugfix release of Arduino-esp32.

espxiehang, 2021-03-08:
Hi @JimDrewGH , In order to avoid this issue, I will reverse the change of `wifi_interface_t`.
 in your project, changing `ESP_IF_WIFI_STA` to `WIFI_IF_STA` can solve this issue.
And after `wifi_interface_t` changes revert, it can also work normally.

JimDrewGH, 2021-03-09:
Thanks for the note guys!  I look forward to a future release.  I am hoping that also fixes the composite video problem for the projects mentioned in my original post.  This issues seems to be involving the setup of the I2S and APLL.

me-no-dev, 2021-03-18:
Change is now in the `idf-release/v3.3` branch of this repository. Could you please test and confirm that everything is OK?

JimDrewGH, 2021-03-20:
I would be happy to test this, but I am not sure what I should be doing to get the latest release (3.3).   Keep in mind that I am using whatever the Arduino IDE installs but I can grab and replace files.  I just need to know where they are and where they go.  :)

lbernstone, 2021-03-20:
Uninstall from board manager, then install the [development repository](https://github.com/espressif/arduino-esp32#installation-instructions)

JimDrewGH, 2021-03-20:
OK, thanks!

closed 2021-07-11 by stale[bot]