espressif/arduino-esp32 #4906: How to set CONFIG_ARDUINO_LOOP_STACK_SIZE?

torntrousers, 2021-03-07:
### Hardware:
Board:							Heltec ESP32 
Core Installation version:			1.0.5

### Description:
Updating to the ESP32 Arduino core 1.0.5 I see there is now a define for CONFIG_ARDUINO_LOOP_STACK_SIZE in main.cpp, how can I set that? I've tried the sketch below but it doesn't get picked up.  

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
#define CONFIG_ARDUINO_LOOP_STACK_SIZE 8192*4

void setup() {
  Serial.begin(115200); Serial.println();
  Serial.println(uxTaskGetStackHighWaterMark(NULL));
}

void loop() {
}
```

### Debug Messages:

That sketch outputs 7524, whereas if I just update the stack size in main.cpp then uxTaskGetStackHighWaterMark does show the change.

Thanks for any pointers.

me-no-dev, 2021-03-08:
You can not change it from within ArduinoIDE, but you should check first why you would need such a large stack in the loop task. That is not normal/OK

torntrousers, 2021-03-08:
I'm using mutual TLS with BearSSL or MbedTLS, both of which seem to use a lot of stack.
Presently I've just been going into main.cpp and updating the stack size in there, I guess I can continue with that but just wonder seeing as there is the new CONFIG_ARDUINO_LOOP_STACK_SIZE define if there is some better way.

Is there some other config place where I could set CONFIG_ARDUINO_LOOP_STACK_SIZE  and have the Arduino environment pickup the changed value?

Probably not the right place to ask but ... or what about with platformio - does that have way to set CONFIG_ARDUINO_LOOP_STACK_SIZE  ?

chegewara, 2021-03-08:
32kB is a huge stack size. Maybe solution would be to do something like that:
- in setup create new task with stack size you need,
- in setup delete default looperTask with `vTaskDelete(NULL)`
- use that task just like internal arduino loopeTask,

torntrousers, 2021-03-08:
I've not tried fine tunning that size to workout what a minimal value is, just the default doesn't work and 8192*4 does.

I just saw in the latest code it now has this define specifically coded to look like you could set it from somewhere else - https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/main.cpp#L20-L22 - so thought I'd ask where that somewhere else is.

Should I just close this ticket if its not actually possible?

obones, 2021-05-17:
People over at PlatformIO have suggested two possible changes:
https://community.platformio.org/t/esp32-stack-configuration-reloaded/20994/2

It would be very nice if there was a user configurable option for this.

torntrousers, 2021-05-17:
There is this: https://github.com/espressif/arduino-esp32/pull/5173

closed 2021-08-04 by stale[bot]