espressif/arduino-esp32 #4910: Error upload code in esp32-s2!

Javito68, 2021-03-07:
Hello everyone, I have the following error when uploading code:

-Arduino 1.8.13 in windows 10
-Hardware: https://es.aliexpress.com/item/1005001922366220.html?spm=a2g0s.9042311.0.0.36d463c0nw301j
-code: https://github.com/volca/esp-soil/blob/pio/src/main.ino
[image]
[image]
I have installed the Arduino core for the ESP32 la branche esp32s2 according to:
[image]

What am I doing wrong? What do I have to do to make it work?
Thanks !!!

chegewara, 2021-03-07:
It is not error. You entered download mode with buttons and after flashing you have rest with button. It is how it works.

Javito68, 2021-03-07:
Thanks for answering chegewara, how do I get rest with button? what is rest button?

a greeting!

chegewara, 2021-03-07:
Sorry, typo. Its `reset`.

Javito68, 2021-03-07:
chegewara  Could you help me with the scheme, I don't see the reset button.
[schematic (1).pdf](https://github.com/espressif/arduino-esp32/files/6097682/schematic.1.pdf)

Javito68, 2021-03-07:
K1, K2 o S1?

chegewara, 2021-03-07:
K2. Or you can just unplug from power source (USB and battery).

Javito68, 2021-03-07:
if I press K2 button the com port is disabled
[image]

Javito68, 2021-03-07:
What would be the correct procedure to upload the program? Thanks!

Javito68, 2021-03-07:
If I keep K1 pressed and press k2 then the com port is enabled but I enter download mode ..., something escapes me

chegewara, 2021-03-07:
Its all fine, windows just change com port when you enter download mode. The screenshots shows you already flashed device, so i dont see any problems you may have.

Javito68, 2021-03-07:
well in the end I have achieved it, I have changed from uart to usb cdc in the IDE:
[image]

thanks chegewara, you always learn something. Salut!

chegewara, 2021-03-07:
Thats strange, because it is not working to me on windows, and not only to me, thats why i didnt mention it.

Javito68, 2021-03-07:
Well, this is how it is working and the port is enabled. All the best!

[image]

ozgurbostan, 2021-03-07:
Hi @Javito68,

Have you used USB CDC feature without any button pressig? and with Win 10?

Javito68, 2021-03-10:
Hi ozgurbostan, I'm sorry for answering late. I've had a job, I'll explain how I proceed:
  -Arduino ide in USB CDC mode
-To load the program: press button IO0 + EN (reset)
-To work, press the EN button (reset)
it works -> OK

ozgurbostan, 2021-03-10:
Thanks @Javito68 for your answer.

Javito68, 2021-03-10:
now I have another problem reading battery values, see thread: help to read battery in esp32-s2.
This I close.
Thanks!!!

closed 2021-03-10 by Javito68