espressif/arduino-esp32 #4912: Bluetooth connect timeout when coexistance with WiFi  (OTA)

LutzO-o, 2021-03-08:
### Hardware:
Board:							ESP32 Dev Module // Module independend 
Core Installation version:			1.0.5 fail. 1,.0.4. works fine 
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                      no
Partition Scheme:                                    Minimal SPIFFS (Huge App) with OTA
Upload Speed:					921600
Computer OS:                                         Windows 10

### Description:
Hi, 
updated esp boardmanager to version to 1.0.5, 
apps using BluetoothSerial and OTA update(WiFi) together are not longer working. Al compiles fine. The Bluetooth connection gets an timeout while conneting to an android phone. 
Same apps work fine with 1.0.4. (no problems at al ! also while all services activ all the Time )
I merged the BasicOTA and the BluetoothSerial SerialToSerial sampel an recreated the Problem.
See below, regards Lutz 

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <WiFi.h>
#include <ESPmDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#include <BluetoothSerial.h>
#include <esp_coexist.h>

const char* ssid = "your net";  
const char* password = "your key"; 

BluetoothSerial SerialBT;

void setup() {
  esp_coex_preference_set(ESP_COEX_PREFER_BT);
  Serial.flush(); // time to Sync.... 
  Serial.begin(115200);
  Serial.println("Booting");
  SerialBT.begin("ESP32test"); //Bluetooth device name
  Serial.println("The device started, now you can pair it with bluetooth!");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  ArduinoOTA
    .onStart([]() {
      String type;
      if (ArduinoOTA.getCommand() == U_FLASH)
        type = "sketch";
      else // U_SPIFFS
        type = "filesystem";

      // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
      Serial.println("Start updating " + type);
    })
    .onEnd([]() {
      Serial.println("\nEnd");
    })
    .onProgress([](unsigned int progress, unsigned int total) {
      Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
    })
    .onError([](ota_error_t error) {
      Serial.printf("Error[%u]: ", error);
      if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
      else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
      else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
      else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
      else if (error == OTA_END_ERROR) Serial.println("End Failed");
    });

  ArduinoOTA.begin();

  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  ArduinoOTA.handle();
  if (Serial.available()) {
    SerialBT.write(Serial.read());
  }
  if (SerialBT.available()) {
    Serial.write(SerialBT.read());
  }
  delay(20);
}
```

### Debug Messages:
'''
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```
NONE!! 

Error message on BTSerial Phone:
Conneting to ESP2test ...
Connetion failed: read failed, socket might closed or timeout, read ret: -1

me-no-dev, 2021-03-08:
Can you please try the latest `master`? Problem should be fixed :)

closed 2021-03-08 by me-no-dev

LutzO-o, 2021-03-09:
Hi, tested last master, now BT connect works. Regards Lutz

me-no-dev, 2021-03-09:
nice :)