espressif/arduino-esp32 #4914: Compilation failure in idf-release/v4.2 branch SOC_GPIO_PIN_COUNT undeclared

william-ferguson-au, 2021-03-09:
### Hardware:
Board:							ESP32-Wrover
Core Installation version:			ESP-IDF release/v4.2  
                                                                Arduino-Esp32 idf-release/v4.2
IDE name:						CLion but compiling using idf.py from ESP-IDF command env.
Flash Frequency:					NA
PSRAM enabled:                                      NA
Upload Speed:					NA
Computer OS:                                        Windows 10

### Description:
Attempting to compile an app that uses arduino-esp32 as a component.
But it fails because SOC_GPIO_PIN_COUNT is not declared.

Fails with
```cpp
In file included from ../components/arduino-esp32/cores/esp32/esp32-hal.h:74,
                 from ../components/arduino-esp32/cores/esp32/esp32-hal-cpu.c:26:
../components/arduino-esp32/cores/esp32/esp32-hal-gpio.h:78:44: error: 'SOC_GPIO_PIN_COUNT' undeclared here (not in a function); did you mean 'GPIO_PIN_COUNT'?
 extern const esp32_gpioMux_t esp32_gpioMux[SOC_GPIO_PIN_COUNT];
                                            ^~~~~~~~~~~~~~~~~~
                                            GPIO_PIN_COUNT

```

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include <esp_log.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>

extern "C" {
    void app_main();
}

static const char *TAG = "main";

void app_main(void) {
    ESP_LOGI(TAG, "Logger initialised");

    while (true) {

        vTaskDelay(300 / portTICK_PERIOD_MS);
    }
}
```

This appears to have been introduced in 6ffe081fd2d6df033cc8b428b92db14eb7a67485
Where it switched from 

```cpp
#include "soc/gpio_caps.h"
```
to 
```cpp
#include "soc/soc_caps.h"
```

NB esp-idf release/v4.2 does not include soc/soc_caps.h

lbernstone, 2021-03-09:
idf-release/v4.2 should be compiled against esp-idf master.  I have built a docker container ready to compile at https://hub.docker.com/r/lbernstone/esp32-arduino-lib-builder if you want a shortcut.

william-ferguson-au, 2021-03-09:
Why would I expect to build arduino-esp32:idf-release/v4.2 against esp-idf:master instead of esp-idf:release/v4.2?
Where is that documented?

I would really prefer to limit the amount of change that my app is exposed to to a particular esp-idf release.
Especially since I am currently struggling with various bugs and incompatibilities in both projects.

lbernstone, 2021-03-09:
The 4.2 branch is highly experimental, and subject to frequent changes.  Documentation is only in the code.  https://github.com/espressif/arduino-esp32/blob/idf-release/v4.2/tools/sdk/esp32/include/config/sdkconfig.h#L629

william-ferguson-au, 2021-03-09:
Ok, let me put this another way then.

If I want to use esp-idf:release/v4.2 which version of arduino-esp32 should
I use?

me-no-dev, 2021-03-09:
can't use Arduino with IDF 4.2. Arduino will jump to 4.3 instead

william-ferguson-au, 2021-03-09:
Sorry I didn't understand either of those comments. @me-no-dev @lbernstone 

@me-no-dev  I actually need to compile against esp-idf:release/v4.3
So what version of arduino-esp32 should I compile against?

@lbernstone what does this mean?
"If you checkout esp-idf at the commit your desired version of arduino shows as CONFIG_ARDUINO_IDF_COMMIT, then it will compile."

lbernstone, 2021-03-09:
```
cd $IDF_PATH
git pull
git checkout 2bfdd036b
```

william-ferguson-au, 2021-03-09:
@ibernstone I don't want to jump to some arbitrary commit on esp-idf:master
I will be using esp-idf:4.3

I want to know what version of arduino-esp32 I should be using with esp-idf:4.3

lbernstone, 2021-03-10:
You can use whatever version you want, you just have to resolve all the conflicts yourself.
If you want someone else to do that for you, then you have to use the versions they have matched up.  For the current commit of idf-release/v4.2 that would be esp-idf @ 2bfdd036b

william-ferguson-au, 2021-03-10:
Right.

So I only need arduino-esp32 to support u8g2.

So for those following after me, my solution was to take arduino-esp32:v1.0.5rc7, remove all libraries except Core, SPI and Wire and then fix the 3 small compilation failures.

For what it's worth 
- I think the versioning used for arduino-esp32 is misguided. It suggests to developers to use versions that they shouldn't use.
- And requiring your arduino-esp32 version to dictate which esp-idf version you must use is backwards. For a component that is so closely linked to and dependent upon esp-idf to not have releases that conform to idf-release releases is just bizarre.

me-no-dev, 2021-03-10:
there are changes in ESP-IDF that prevent Arduino to support all IDF versions with one codebase (or code will become spaghetti and unmanageable). Arduino is really meant to be used with the precompiled IDF that it includes.

wladwnt, 2021-03-13:
> 
> 
> idf-release/v4.2 should be compiled against esp-idf master. I have built a docker container ready to compile at https://hub.docker.com/r/lbernstone/esp32-arduino-lib-builder if you want a shortcut.

@lbernstone: Are you using  idf-release/v4.2 with arduino-esp32 as a component? If yes, what is your expirience, is it arther buggy or not?

brandonros, 2021-03-28:
Something is broken somewhere.

```
git clone -b idf-release/v4.2 --recursive https://github.com/espressif/arduino-esp32.git components/arduino
docker run --rm -v $PWD:/project -w /project espressif/idf:release-v4.2 idf.py build
```

yields

```
Brandons-MacBook-Air:esp32 brandonros$ docker run --rm -v $PWD:/project -w /project espressif/idf:release-v4.2 idf.py build
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
Detecting the Python interpreter
Checking "python" ...
Python 3.6.9
"python" has been detected
Adding ESP-IDF tools to PATH...
Using Python interpreter in /opt/esp/python_env/idf4.2_py3.6_env/bin/python
Checking if Python packages are up to date...
Python requirements from /opt/esp/idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /opt/esp/idf/components/esptool_py/esptool
  /opt/esp/idf/components/espcoredump
  /opt/esp/idf/components/partition_table
  /opt/esp/idf/components/app_update
  /opt/esp/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin
  /opt/esp/tools/xtensa-esp32s2-elf/esp-2020r3-8.4.0/xtensa-esp32s2-elf/bin
  /opt/esp/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin
  /opt/esp/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin
  /opt/esp/tools/cmake/3.16.4/bin
  /opt/esp/tools/openocd-esp32/v0.10.0-esp32-20200709/openocd-esp32/bin
  /opt/esp/python_env/idf4.2_py3.6_env/bin
  /opt/esp/idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

[0/1] Re-running CMake...
-- ccache will be used for faster recompilation
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /project/sdkconfig
-- App "main" version: 1
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /opt/esp/idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /project/build/esp-idf/esp32/esp32_out.ld
-- Adding linker script /opt/esp/idf/components/esp32/ld/esp32.project.ld.in
-- Adding linker script /opt/esp/idf/components/esp32/ld/esp32.peripherals.ld
-- Components: app_trace app_update arduino asio bootloader bootloader_support bt cbor coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_ipc esp_local_ctrl esp_netif esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos heap idf_test jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /opt/esp/idf/components/app_trace /opt/esp/idf/components/app_update /project/components/arduino /opt/esp/idf/components/asio /opt/esp/idf/components/bootloader /opt/esp/idf/components/bootloader_support /opt/esp/idf/components/bt /opt/esp/idf/components/cbor /opt/esp/idf/components/coap /opt/esp/idf/components/console /opt/esp/idf/components/cxx /opt/esp/idf/components/driver /opt/esp/idf/components/efuse /opt/esp/idf/components/esp-tls /opt/esp/idf/components/esp32 /opt/esp/idf/components/esp_adc_cal /opt/esp/idf/components/esp_common /opt/esp/idf/components/esp_eth /opt/esp/idf/components/esp_event /opt/esp/idf/components/esp_gdbstub /opt/esp/idf/components/esp_hid /opt/esp/idf/components/esp_http_client /opt/esp/idf/components/esp_http_server /opt/esp/idf/components/esp_https_ota /opt/esp/idf/components/esp_https_server /opt/esp/idf/components/esp_ipc /opt/esp/idf/components/esp_local_ctrl /opt/esp/idf/components/esp_netif /opt/esp/idf/components/esp_ringbuf /opt/esp/idf/components/esp_rom /opt/esp/idf/components/esp_serial_slave_link /opt/esp/idf/components/esp_system /opt/esp/idf/components/esp_timer /opt/esp/idf/components/esp_websocket_client /opt/esp/idf/components/esp_wifi /opt/esp/idf/components/espcoredump /opt/esp/idf/components/esptool_py /opt/esp/idf/components/expat /opt/esp/idf/components/fatfs /opt/esp/idf/components/freemodbus /opt/esp/idf/components/freertos /opt/esp/idf/components/heap /opt/esp/idf/components/idf_test /opt/esp/idf/components/jsmn /opt/esp/idf/components/json /opt/esp/idf/components/libsodium /opt/esp/idf/components/log /opt/esp/idf/components/lwip /project/main /opt/esp/idf/components/mbedtls /opt/esp/idf/components/mdns /opt/esp/idf/components/mqtt /opt/esp/idf/components/newlib /opt/esp/idf/components/nghttp /opt/esp/idf/components/nvs_flash /opt/esp/idf/components/openssl /opt/esp/idf/components/partition_table /opt/esp/idf/components/perfmon /opt/esp/idf/components/protobuf-c /opt/esp/idf/components/protocomm /opt/esp/idf/components/pthread /opt/esp/idf/components/sdmmc /opt/esp/idf/components/soc /op [... cut ...]
 extern const esp32_gpioMux_t esp32_gpioMux[SOC_GPIO_PIN_COUNT];
                                            ^~~~~~~~~~~~~~~~~~
                                            GPIO_PIN_COUNT
ninja: build stopped: subcommand failed.
Executing action: all (aliases: build)
Running ninja in directory /project/build
Executing "ninja all"...
ninja failed with exit code 1
Your environment is not configured to handle unicode filenames outside of ASCII range. Environment variable LC_ALL is temporary set to C.UTF-8 for unicode support.
```

chegewara, 2021-03-28:
Arduino idf_release/v4.2 branch is not compatible with esp-idf v4.2 branch.
You have to use a particular commit from esp-idf master branch. Almost for sure this one:
https://github.com/espressif/esp-idf/commit/2bfdd036b

laolux, 2021-06-14:
@william-ferguson-au 

> So for those following after me, my solution was to take arduino-esp32:v1.0.5rc7, remove all libraries except Core, SPI and Wire and then fix the 3 small compilation failures.

Would you mind sharing the result? I am currently running into the same issue with GxEPD2 and somehow stuck. Would be easier to start from your result. Thanks.

william-ferguson-au, 2021-06-14:
@laolux have a look at https://github.com/william-ferguson-au/arduino-esp32/tree/trinity-v9-1.0.5rc7

laolux, 2021-06-15:
@william-ferguson-au that's pretty nice. I use it as a basis to make my own reduced branch of arduino-esp32 where I remove all unneeded stuff (for me). That should make it easier for me to keep up with changes here without breaking my builds.

On another note, I found that using the release v4.3 branch of the idf works a lot better with arduino-esp32 master. Specifically, I used 88c2b69c68d3258d846c6e43f8cba329ccaa5898 of idf and ea236e28e53d248b98bf00a69b98f9cdb02cf54f of arduino-esp32 and it almost worked. Only needed to comment out `eth_phy = esp_eth_phy_new_ksz8081(&phy_config);` in `libraries/WiFi/src/ETH.cpp`.

closed 2021-09-06 by stale[bot]