espressif/arduino-esp32 #4915: Bluetooth connect timeout after Android phone app disconnect and reconnect (IDFGH-4889)

cattledogGH, 2021-03-09:
ESP32 Dev Module, Arduino Core 1.0.5, 

BluetoothSerial.h included with core library properties shows
`name=BluetoothSerial
version=1.0
author=Evandro Copercini
maintainer=Evandro Copercini
sentence=Simple UART to Classical Bluetooth bridge for ESP32
paragraph=
category=Communication
url=
architectures=esp32`

When using BluetoothSerial.h and library example SerialToSerialBT with an Android phone (Kai Morich[ SerialBluetoothTerminal](https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal&hl=en_US&gl=US)) the phone app can not reconnect after disconnect without restarting the ESP32.

Initial connection and example code work as expected. Disconnect phone app. Phone will not reconnect without ESP32 reset.

I believe that this issue is related to this issue which was repaired in the esp-idf
[https://github.com/espressif/esp-idf/issues/1537](url)

Request that the reconnection error be corrected in BluetoothSerial.h for ESP32.

projectgus, 2021-03-09:
Arduino Core 1.0.5 is built with ESP-IDF release/v3.3 branch commit 7a85334d8

cattledogGH, 2021-03-09:
This failure to reconnect after disconnect does not occur when using Arduino Core version 1.0.4. 

Something was broken with the update to 1.0.5

lbernstone, 2021-03-09:
See https://github.com/espressif/arduino-esp32/issues/4912

cattledogGH, 2021-03-09:
"See #4912"  WiFi does not need to be involved for the disconnect/reconnect behavior.

This issue may also be related to #4919

cattledogGH, 2021-03-09:
I can not confirm that the file revisions of #4912 correct the issue of failure to reconnect when using the BluetoothSerial library example SerialToSerialBT with an Android phone.
I installed the revised BluetoothSerial.h file and the revised esp32-hal-bt.c files into core 1.0.5  but there was not change in the failure to reconnect after a disconnect without restarting the Arduino.
As previously noted, this behaviour does not occur with core version 1.0.4

me-no-dev, 2021-03-10:
I do not have an Android phone, but some fixes have been added to current master (and will come in 1.0.6), so I would ask you if you can give it a go and confirm?

cattledogGH, 2021-03-10:
@me-no-dev
>I do not have an Android phone, but some fixes have been added to current master (and will come in 1.0.6), so I would ask you if you can give it a go and confirm?

I have tested the changes in these files, and they do not solve the problem.
[https://github.com/espressif/arduino-esp32/commit/4d95e3a7ea072ac61ff0eb24b3304db47de96712](url)

Is there something else you would like me to test?

me-no-dev, 2021-03-10:
@cattledogGH I would suggest that you try directly with master

cattledogGH, 2021-03-10:
I have only loaded the Arduino core using the Boards Manager of the Arduino IDE.  What is the process for getting the master into the packages of the Arduino 15 folder? I am using Windows 10. Arduino IDE 1.8.13

me-no-dev, 2021-03-10:
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md but uninstall from board manager first

cattledogGH, 2021-03-10:
Yes, I believe the current master fixes the issue.

I followed the instructions referenced in the previous comment, and I think that I have successfully installed the current master, but I'm not quite sure where to pull a version number from the compiling report to absolutely confirm what was used.

With the sketch from the SerialBluetooth library "SerialToSerialBluetooth"  and the current master, I can connect and disconnect repeatedly without restarting the ESP 32.

I look forward to Arduino Core 1.0.6. available through the Board Manager.

lyusupov, 2021-03-13:
It was found out that an update of [this libbt.a](https://github.com/espressif/arduino-esp32/blob/e8311b00ae2575426d75637fc9664c84d9b91e2d/tools/sdk/lib/libbt.a) binary from https://github.com/espressif/arduino-esp32/commit/e8311b00ae2575426d75637fc9664c84d9b91e2d is sufficient to make a temporary fix for this issue.

gfo1983, 2021-03-20:
> It was found out that an update of [this libbt.a](https://github.com/espressif/arduino-esp32/blob/e8311b00ae2575426d75637fc9664c84d9b91e2d/tools/sdk/lib/libbt.a) binary from [e8311b0](https://github.com/espressif/arduino-esp32/commit/e8311b00ae2575426d75637fc9664c84d9b91e2d) is sufficient to make a temporary fix for this issue.

great. this will really solve the problem. thank

kressdev, 2021-03-20:
> https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md but uninstall from board manager first

Can confirm the issue and this solution is working.

barafael, 2021-03-20:
Is there a rough estimate on when this fix will be generally available e.g. in platformio?

me-no-dev, 2021-03-23:
probably by the end of this week

cattledogGH, 2021-03-27:
Closed with release of Arduino Core 1.0.6 available through Boards Manager

ketan, 2021-04-04:
Is this available on platformio too?

barafael, 2021-04-04:
Works just fine for me via platformio core 5.1.0 in the Arduino env. Thanks everyone!

oraclerob, 2021-04-12:
I have the latest 1.0.6 installed via pio and can confirm this is still an issue. I'm testing using a BT GPS receiver and the following happens when I switch the device off and on:

`[V][BluetoothSerial.cpp:290] esp_spp_cb(): ESP_SPP_DATA_IND_EVT len=38 handle=129`
`[I][BluetoothSerial.cpp:255] esp_spp_cb(): ESP_SPP_CLOSE_EVT`
`[I][BluetoothSerial.cpp:765] connect(): master : remoteName`

Have the following check to manually reconnect but it never gets past the connect() message on the next attempt it just seems to hang.

`if (!SerialBT.connected()) { ............. `

FernandoGarcia, 2022-03-21:
Hi!

I'm trying to use the function `esp_coex_preference_set(ESP_COEX_PREFER_BT);` but after connect to ESP32 access point I'm getting this crash loop:

```
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[E][WiFiGeneric.cpp:137] wifiLowLevelInit(): esp_wifi_init 4353
[E][BluetoothSerial.cpp:865] isReady(): BT is not initialized. Call begin() first
Starting software access point
[E][WiFiAP.cpp:109] softAP(): passphrase too short!
Starting captive portal on 192.168.4.1
abort() was called at PC 0x401932d3 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x4008f694:0x3ffd31e0 0x4008f911:0x3ffd3200 0x401932d3:0x3ffd3220 0x4019331a:0x3ffd3240 0x401929ef:0x3ffd3260 0x40192d26:0x3ffd3280 0x40192a8d:0x3ffd32a0 0x401af79a:0x3ffd32c0 0x401b0019:0x3ffd3300 0x400e78d7:0x3ffd3330 0x400e8f48:0x3ffd3350 0x400dfa2e:0x3ffd3370 0x400d3a9e:0x3ffd3390 0x400ecf38:0x3ffd33e0 0x40090986:0x3ffd3400
```

I'm using this code:

```
#define BLUETOOTH_NAME "Parking Sensor"
#define BLUETOOTH_PINCODE "0000"

#include <ESP8266React.h>
#include "BluetoothSerial.h"
#include <ArduinoJson.h>
#include <esp_coexist.h>

BluetoothSerial SerialBT;

AsyncWebServer server(80);
ESP8266React esp8266React(&server);

unsigned start_millis;

void bluetoothHandle(String msg);

void setup()
{
  esp_coex_preference_set(ESP_COEX_PREFER_BT); // Priority for bluetooth

  Serial.begin(115200);
  
  esp8266React.begin();
  server.begin();

  SerialBT.setPin(BLUETOOTH_PINCODE);
  SerialBT.begin(BLUETOOTH_NAME);  
}

void loop()
{
  String msg;

  esp8266React.loop();

  if (SerialBT.available())
  {
    esp8266React.loop();
    start_millis = millis();
    msg = SerialBT.readString();

    bluetoothHandle(msg);
  }
}

void bluetoothHandle(String msg)
{
  char charBuf[100];
  msg.toCharArray(charBuf, 100);

  DynamicJsonDocument Json(100);

  Serial.print("t1 = ");
  Serial.print(millis() - start_millis);
  Serial.print(", ");

  String mode[2] = {"reverse", "break"};
  String sensor[8] = {"a", "b", "c", "d", "e", "f", "g", "h"};

  Json["mode"] = mode[random(0, 2)];
  Json["sensor"]   = sensor[random(0, 8)];
  Json["distance"] = random(0, 200);

  serializeJson(Json, SerialBT);

  Serial.print("cmd = ");
  Serial.println(charBuf);
  Serial.print("res = ");
  serializeJson(Json, Serial);
  Serial.print(", t2 = ");
  Serial.println(millis() - start_millis);
}
```

The ESP8266React lib (framework) is [here](git@github.com:FernandoGarcia/esp8266-react.git).

Webserver and bluetooth can work fine separately without `esp_coex_preference_set(ESP_COEX_PREFER_BT)` function but together just bluetooth works fine.

I have an ESP32-Wroover IE 8MB and I'm using PIO 5.2.5 with espressif32 3.5.0.

Can someone give some help?

Best regards.

VojtechBartoska, 2022-04-06:
Guys , do you still face this issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1). Take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

cattledogGH, 2022-04-07:
I can connect and disconnect multiple times with no issues using 2.0.3RC1 with Arduino IDE 1.8.15. I do not see the problem of having to restart the esp32 to reconnect. I am using Kai Morich's Serial Bluetooth Terminal on an Android phone with Android 11.

No issues as well with 2.0.2.

VojtechBartoska, 2022-04-07:
@cattledogGH Thanks for testing, I'm, then closing this issue. If still needed, you can reopen it.

closed 2022-04-07 by VojtechBartoska

italocjs, 2023-05-03:
Hello, i'm experiencing the same issue,  using ESPIDF v4.4.4 + Arduino as component (v.2.0.8 - master branch from a few hours ago).  No error messages, First connection works fine, second needs to reset the board.

I have enabled the debug level for the bluetooth stuff in menuconfig, and attached the full terminal log (with some comments) https://pastebin.com/vA2uSbJG  and here is the sdkconfig https://pastebin.com/iM2egQDf

I've created an task that check if an client has disconnectd (state machine checking every second), it it had, i stop the SerialBT, set the callback and start the SerialBT object. this helped me to be able to connect again.

WebDust21, 2024-07-30:
Having the same issue with Arduino-ESP32 v3.0.3: I can connect fine the first time, but after disconnecting, it is impossible to re-establish a connection without resetting the ESP32.

@italocjs Your suggestion did the trick: when we know that the BT connection has been lost, disable and re-enable the BluetoothSerial module and it is able to reconnect again.

Pseudo-code here...the "flagBtWasConnected" is a one-shot so it doesn't infinitely reset the BT module if no connection is present.
```
void BtDisconnectEvent() {
    if (flagBtWasConnected) {
	    // widely known bug with Arduino-ESP32 BT Classic / BluetoothSerial means that
	    //after a disconnect, we cannot reconnect to any device.
	    // Crude solution: manually restart the BT functionality upon a disconnect.
	    SerialBT.end();
	    delay(50);
	    SerialBT.begin(btDeviceName, false, true);
    }
    flagBtWasConnected = false;
}
```

italocjs, 2024-07-31:
> Having the same issue with Arduino-ESP32 v3.0.3: I can connect fine the first time, but after disconnecting, it is impossible to re-establish a connection without resetting the ESP32.
>
> @italocjs [...]

Classis bluetooth have a lot of issues. i gave up on it and have switched to nimble + nordic uart service. 

on Classic BT i had issues with specific phones manufacturers not connecting or causing the device to crash, lib had a huge footprint, require weird undocumented workarounds.    Anyway, if you can, take a look at Nimble+NUS.

WebDust21, 2024-08-01:
The problem: I'm developing a "compatible replacement" for an obsolete BT board that runs over BT Classic (does not support BLE).  The requirement is to seamlessly integrate into the existing ecosystem with zero changes.

So there's no way around the use of BT Classic for this project.