espressif/arduino-esp32 #4916: ESP32-Mqtt was not using Azure SSL Certificate

selvamani1991, 2021-03-09:
ESP32-Mqtt was not connected using Azure SSL Certificate

I am using the below code for connecting with ssl certificate was not working.

/*
  Wifi secure connection example for ESP32
  Running on TLS 1.2 using mbedTLS
  Suporting the following chipersuites:
  "TLS_ECDHE_ECDSA_WITH_AES_256_GCM_SHA384","TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384","TLS_DHE_RSA_WITH_AES_256_GCM_SHA384","TLS_ECDHE_ECDSA_WITH_AES_256_CCM","TLS_DHE_RSA_WITH_AES_256_CCM","TLS_ECDHE_ECDSA_WITH_AES_256_CBC_SHA384","TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA384","TLS_DHE_RSA_WITH_AES_256_CBC_SHA256","TLS_ECDHE_ECDSA_WITH_AES_256_CBC_SHA","TLS_ECDHE_RSA_WITH_AES_256_CBC_SHA","TLS_DHE_RSA_WITH_AES_256_CBC_SHA","TLS_ECDHE_ECDSA_WITH_AES_256_CCM_8","TLS_DHE_RSA_WITH_AES_256_CCM_8","TLS_ECDHE_ECDSA_WITH_CAMELLIA_256_GCM_SHA384","TLS_ECDHE_RSA_WITH_CAMELLIA_256_GCM_SHA384","TLS_DHE_RSA_WITH_CAMELLIA_256_GCM_SHA384","TLS_ECDHE_ECDSA_WITH_CAMELLIA_256_CBC_SHA384","TLS_ECDHE_RSA_WITH_CAMELLIA_256_CBC_SHA384","TLS_DHE_RSA_WITH_CAMELLIA_256_CBC_SHA256","TLS_DHE_RSA_WITH_CAMELLIA_256_CBC_SHA","TLS_ECDHE_ECDSA_WITH_AES_128_GCM_SHA256","TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256","TLS_DHE_RSA_WITH_AES_128_GCM_SHA256","TLS_ECDHE_ECDSA_WITH_AES_128_CCM","TLS_DHE_RSA_WITH_AES_128_CCM","TLS_ECDHE_ECDSA_WITH_AES_128_CBC_SHA256","TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA256","TLS_DHE_RSA_WITH_AES_128_CBC_SHA256","TLS_ECDHE_ECDSA_WITH_AES_128_CBC_SHA","TLS_ECDHE_RSA_WITH_AES_128_CBC_SHA","TLS_DHE_RSA_WITH_AES_128_CBC_SHA","TLS_ECDHE_ECDSA_WITH_AES_128_CCM_8","TLS_DHE_RSA_WITH_AES_128_CCM_8","TLS_ECDHE_ECDSA_WITH_CAMELLIA_128_GCM_SHA256","TLS_ECDHE_RSA_WITH_CAMELLIA_128_GCM_SHA256","TLS_DHE_RSA_WITH_CAMELLIA_128_GCM_SHA256","TLS_ECDHE_ECDSA_WITH_CAMELLIA_128_CBC_SHA256","TLS_ECDHE_RSA_WITH_CAMELLIA_128_CBC_SHA256","TLS_DHE_RSA_WITH_CAMELLIA_128_CBC_SHA256","TLS_DHE_RSA_WITH_CAMELLIA_128_CBC_SHA","TLS_ECDHE_ECDSA_WITH_3DES_EDE_CBC_SHA","TLS_ECDHE_RSA_WITH_3DES_EDE_CBC_SHA","TLS_DHE_RSA_WITH_3DES_EDE_CBC_SHA","TLS_DHE_PSK_WITH_AES_256_GCM_SHA384","TLS_DHE_PSK_WITH_AES_256_CCM","TLS_ECDHE_PSK_WITH_AES_256_CBC_SHA384","TLS_DHE_PSK_WITH_AES_256_CBC_SHA384","TLS_ECDHE_PSK_WITH_AES_256_CBC_SHA","TLS_DHE_PSK_WITH_AES_256_CBC_SHA","TLS_DHE_PSK_WITH_CAMELLIA_256_GCM_SHA384","TLS_ECDHE_PSK_WITH_CAMELLIA_256_CBC_SHA384","TLS_DHE_PSK_WITH_CAMELLIA_256_CBC_SHA384","TLS_PSK_DHE_WITH_AES_256_CCM_8","TLS_DHE_PSK_WITH_AES_128_GCM_SHA256","TLS_DHE_PSK_WITH_AES_128_CCM","TLS_ECDHE_PSK_WITH_AES_128_CBC_SHA256","TLS_DHE_PSK_WITH_AES_128_CBC_SHA256","TLS_ECDHE_PSK_WITH_AES_128_CBC_SHA","TLS_DHE_PSK_WITH_AES_128_CBC_SHA","TLS_DHE_PSK_WITH_CAMELLIA_128_GCM_SHA256","TLS_DHE_PSK_WITH_CAMELLIA_128_CBC_SHA256","TLS_ECDHE_PSK_WITH_CAMELLIA_128_CBC_SHA256","TLS_PSK_DHE_WITH_AES_128_CCM_8","TLS_ECDHE_PSK_WITH_3DES_EDE_CBC_SHA","TLS_DHE_PSK_WITH_3DES_EDE_CBC_SHA","TLS_RSA_WITH_AES_256_GCM_SHA384","TLS_RSA_WITH_AES_256_CCM","TLS_RSA_WITH_AES_256_CBC_SHA256","TLS_RSA_WITH_AES_256_CBC_SHA","TLS_ECDH_RSA_WITH_AES_256_GCM_SHA384","TLS_ECDH_RSA_WITH_AES_256_CBC_SHA384","TLS_ECDH_RSA_WITH_AES_256_CBC_SHA","TLS_ECDH_ECDSA_WITH_AES_256_GCM_SHA384","TLS_ECDH_ECDSA_WITH_AES_256_CBC_SHA384","TLS_ECDH_ECDSA_WITH_AES_256_CBC_SHA","TLS_RSA_WITH_AES_256_CCM_8","TLS_RSA_WITH_CAMELLIA_256_GCM_SHA384","TLS_RSA_WITH_CAMELLIA_256_CBC_SHA256","TLS_RSA_WITH_CAMELLIA_256_CBC_SHA","TLS_ECDH_RSA_WITH_CAMELLIA_256_GCM_SHA384","TLS_ECDH_RSA_WITH_CAMELLIA_256_CBC_SHA384","TLS_ECDH_ECDSA_WITH_CAMELLIA_256_GCM_SHA384","TLS_ECDH_ECDSA_WITH_CAMELLIA_256_CBC_SHA384","TLS_RSA_WITH_AES_128_GCM_SHA256","TLS_RSA_WITH_AES_128_CCM","TLS_RSA_WITH_AES_128_CBC_SHA256","TLS_RSA_WITH_AES_128_CBC_SHA","TLS_ECDH_RSA_WITH_AES_128_GCM_SHA256","TLS_ECDH_RSA_WITH_AES_128_CBC_SHA256","TLS_ECDH_RSA_WITH_AES_128_CBC_SHA","TLS_ECDH_ECDSA_WITH_AES_128_GCM_SHA256","TLS_ECDH_ECDSA_WITH_AES_128_CBC_SHA256","TLS_ECDH_ECDSA_WITH_AES_128_CBC_SHA","TLS_RSA_WITH_AES_128_CCM_8","TLS_RSA_WITH_CAMELLIA_128_GCM_SHA256","TLS_RSA_WITH_CAMELLIA_128_CBC_SHA256","TLS_RSA_WITH_CAMELLIA_128_CBC_SHA","TLS_ECDH_RSA_WITH_CAMELLIA_128_GCM_SHA256","TLS_ECDH_RSA_WITH_CAMELLIA_128_CBC_SHA256","TLS_ECDH_ECDSA_WITH_CAMELLIA_128_GCM_SHA256","TLS_ECDH_ECDSA_WITH_CAMELLIA_128_CBC_SHA256","TLS_RSA_WITH_3DES_EDE_CBC_SHA","TLS_ECDH_RSA_WITH_3DES_EDE_CBC_SHA","TLS_ECDH_ECDSA_WITH_3DES_EDE_CBC_SHA","TLS_RSA_PSK_WITH_AES_256_GCM_SHA384","TLS_RSA_PSK_WITH_AES_256_CBC_SHA384","TLS_RSA_PSK_WITH_AES_256_CBC_SHA","TLS_RSA_PSK_WITH_CAMELLIA_256_GCM_SHA384","TLS_RSA_PSK_WITH_CAMELLIA_256_CBC_SHA384","TLS_RSA_PSK_WITH_AES_128_GCM_SHA256","TLS_RSA_PSK_WITH_AES_128_CBC_SHA256","TLS_RSA_PSK_WITH_AES_128_CBC_SHA","TLS_RSA_PSK_WITH_CAMELLIA_128_GCM_SHA256","TLS_RSA_PSK_WITH_CAMELLIA_128_CBC_SHA256","TLS_RSA_PSK_WITH_3DES_EDE_CBC_SHA","TLS_PSK_WITH_AES_256_GCM_SHA384","TLS_PSK_WITH_AES_256_CCM","TLS_PSK_WITH_AES_256_CBC_SHA384","TLS_PSK_WITH_AES_256_CBC_SHA","TLS_PSK_WITH_CAMELLIA_256_GCM_SHA384","TLS_PSK_WITH_CAMELLIA_256_CBC_SHA384","TLS_PSK_WITH_AES_256_CCM_8","TLS_PSK_WITH_AES_128_GCM_SHA256","TLS_PSK_WITH_AES_128_CCM","TLS_PSK_WITH_AES_128_CBC_SHA256","TLS_PSK_WITH_AES_128_CBC_SHA","TLS_PSK_WITH_CAMELLIA_128_GCM_SHA256","TLS_PSK_WITH_CAMELLIA_128_CBC_SHA256","TLS_PSK_WITH_AES_128_CCM_8","TLS_PSK_WITH_3DES_EDE_CBC_SHA","TLS_EMPTY_RENEGOTIATION_INFO_SCSV"]
  2017 - Evandro Copercini - Apache 2.0 License.
*/

#include <WiFiClientSecure.h>

const char* ssid     = "xxx";     // your network SSID (name of wifi network)
const char* password = "xxx"; // your network password

const char*  server = "xxx.azure-devices.net";  // Server URL

// www.howsmyssl.com root certificate authority, to verify the server
// change it to your server root CA
// SHA1 fingerprint is broken now!

const char* test_root_ca= \
"-----BEGIN CERTIFICATE-----\n" \
"MIIDxDCCAqwCCQDyrMM0eCtQEDANBgkqhkiG9w0BAQsFADCBozELMAkGA1UEBhMC\n" \
"SU4xCzAJBgNVBAgMAktBMRIwEAYDVQQHDAlCYW5nYWxvcmUxEzARBgNVBAoMCkJy\n" \
"YWluQ2hpbGQxGTAXBgNVBAsMEEF6dXJlU21hcnREZXZpY2UxFzAVBgNVBAMMDlNt\n" \
"YXJ0RGV2aWNlQmN0MSowKAYJKoZIhvcNAQkBFht2aWpheS5uQGJyYWluY2hpbGR0\n" \
"ZWNoLmNvbSAwHhcNMjAwNDEyMDYxNzM2WhcNMzEwNjMwMDYxNzM2WjCBozELMAkG\n" \
"A1UEBhMCSU4xCzAJBgNVBAgMAktBMRIwEAYDVQQHDAlCYW5nYWxvcmUxEzARBgNV\n" \
"BAoMCkJyYWluQ2hpbGQxGTAXBgNVBAsMEEF6dXJlU21hcnREZXZpY2UxFzAVBgNV\n" \
"BAMMDlNtYXJ0RGV2aWNlQmN0MSowKAYJKoZIhvcNAQkBFht2aWpheS5uQGJyYWlu\n" \
"Y2hpbGR0ZWNoLmNvbSAwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQDM\n" \
"9BSby0MzYNe4FpezzhUwkthQl6a41zuf6bpp42jUbagGjmLvBf36By6dUd4qQN9n\n" \
"DELU/51TLr0v3lWRyli9dz9gaD3l0l8jw7mT5TaaxgWi6A8o288T8Vlf2QKPwH8Y\n" \
"4wejrVpWp4hUxX5SA6y5a4XTe1UBHyKDiAAa3SGUu7gzw6zau+6HE45sQhzBZoke\n" \
"fApDQvYmexjecIfqGpNTNxhvKENTt/luNBRevsRr5uEp+7D2mwVHWQxz321EvtJX\n" \
"8GePNpzpavBzfQaXBIg8N8Ump33NEzPcfCvlVjY/GCTdjDV47RvFCB45hgoU5y1P\n" \
"voSEu60ju4fOnKK5z9OlAgMBAAEwDQYJKoZIhvcNAQELBQADggEBAAqrBglSslip\n" \
"kmrilWXTRWwfc9goH+BbgHvSLJJI6/LEsLVkHPYE+Id4lvyFih8JMTgn5QVqDdtK\n" \
"dqE8MN792VQBmDMMWamRuQLWAjZi3zrRlT24X1S0lQncx418z4gIJL7rugfdkkdc\n" \
"v77BXQaog3N7ErEo0ye8AajgcnNixe5oYzI0jrC/7TFpo6r6me5yKz2GLq/RIpSq\n" \
"8pT2prkw0pCJWBHqT7pt/f2ZxPp5wKFKqOXdWJSW957hDcVobgYUzfARtBA1pgmv\n" \
"JdA4Wz2GMmzdhKGlNKG0wXjesSYSyyijKfNHQzRZ04SM4kDk5Xf9sm1igWvrOIbU\n" \
"B6qOPjRQhDA=\n" \
"-----END CERTIFICATE-----\n";

// You can use x.509 client certificates if you want
const char* test_client_key =  \
"-----BEGIN RSA PRIVATE KEY-----\n" \
"MIIEpAIBAAKCAQEAvrsh1YjvyGB90UroWLX383OrtrfS5jabzIAl99W0gjQWpN8E\n" \
"Ccq8thffWzq6+EoJrjrMshqLrgwuupsKyYAxZ8pEBMu/7/6P2ROk/9huycP9MP/c\n" \
"nexpfqlTrlrabDFG4YEbFNGmQ0eFhGjR4G+ymn6WGGzcG6c+Wd7YgseMI7MPz8W0\n" \
"EUXSmQMYa0BIrFfwQ4j5cnJL2L6ylenc6rwoVG1Zzlvx+1ZgtOuIMjZDzyfwD2OU\n" \
"grB8Spjg8RI8CPoBeF8TMwnNghPfSJZ2Hl7Kq6S8Wxo67qfUVBVtfA8lE/8dvO16\n" \
"txBxc7xDnZDJD6bQP+qZl81wLjzu6/e2H6Cd7wIDAQABAoIBAFFbW1RNhBKR7jk/\n" \
"XZqxjCGBnqZjBQgEFTMHGqr7snKzq1EKfBt9Kj/2MOUmSSit1DFIjZYPMrO8KJaa\n" \
"meBX2TsvkRIGo0nI6S0sbv5fIOrZuf+TtFR8yocnzTUbNf5eiqCYRKf9wZEDShWv\n" \
"7I5yttml03p5vO3iyfrlzq8CVWnIM03lDgjgVRvgqPMTL/ZpOCkbSuMGPlwTVXXR\n" \
"CTBH2SNg3SRROKMYM4sJWGbbCeB55VkryIosfpSkqeOxJ2Kon5j9PNOt+3M80/pV\n" \
"T/t4gRtp9IlFu2AfGbibSiyBs2/Zx0VPJUlPvJX57k88EtuCZz/ZOxc3ryRKrJN1\n" \
"wMg70ikCgYEA8l4m82UFivtPkE7fcmNtLUTilwh9v5Fkl9HTcJx0uppa47+wnJHU\n" \
"+s1uBh9O1yjaxJS9bBupNsNZc7/93SP6e6m2/mBP8AR3SmmY7bJTw2hnN8caMSVW\n" \
"tm57NY6kLNF5BU6bZiBQ6YqFExOBJ4PGp8LOhYDnQ6NuA9zcVZjvxW0CgYEAyXV2\n" \
"iIA8hnjAqkxq61B2EhXxUigCqAFDQGJHt98F92it+GeRU5iWBjSWwrvPuiCMSRNl\n" \
"bPKB0gzPCjag8FGK9B0hHApCYpgQuZ5x80P0txkWYP1dYB2+zsKCr32WMaw8pDPD\n" \
[... cut ...]
    client.println("Host: www.howsmyssl.com");
    client.println("Connection: close");
    client.println();

    while (client.connected()) {
      String line = client.readStringUntil('\n');
      if (line == "\r") {
        Serial.println("headers received");
        break;
      }
    }
    // if there are incoming bytes available
    // from the server, read them and print them:
    while (client.available()) {
      char c = client.read();
      Serial.write(c);
    }

    client.stop();
  }
}

void loop() {
  // do nothing
}

me-no-dev, 2021-03-09:
Could you please clarify the problem? Why should ESP-MQTT use the Azure certificates? And are you trying to connect to `www.howsmyssl.com` using Azure certs?

selvamani1991, 2021-03-10:
Hi,
I refer this link, kindly review it
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiClientSecure/examples/WiFiClientSecure/WiFiClientSecure.ino

me-no-dev, 2021-03-10:
enable debug in Arduino IDE and post the log from Serial. The example as given can not be tested so I can not know what breaks where

me-no-dev, 2021-03-10:
in the board menu in Arduino IDE there is an option to enable debug. Set it to "verbose" and try again to get the log. You can paste it here instead of screenshot

selvamani1991, 2021-03-10:
Hi,
I got this commands when i enable debug-verbose,

...ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
[E][esp32-hal-cpu.c:170] setCpuFrequencyMhz(): Can not switch to 240 MH⸮⸮⸮⸮⸮CPU frequency rated for 160MHz.
[D][esp32-hal-cpu.c:189] setCpuFrequencyMhz(): PLL: 320 / 2 = 160 Mhz, APB: 80000000 Hz
Attempting to connect to SSID: Libre
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
.[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
..[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.1.61, MASK: 255.255.255.0, GW: 192.168.1.1
Connected to Libre

Starting connection to server...
[V][ssl_client.cpp:59] start_ssl_client(): Free internal heap before TLS 280372
[V][ssl_client.cpp:65] start_ssl_client(): Starting socket
[V][ssl_client.cpp:104] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:113] start_ssl_client(): Setting up the SSL/TLS structure...
[V][ssl_client.cpp:129] start_ssl_client(): Loading CA cert
[V][ssl_client.cpp:178] start_ssl_client(): Loading CRT cert
[V][ssl_client.cpp:187] start_ssl_client(): Loading private key
[V][ssl_client.cpp:197] start_ssl_client(): Setting hostname for TLS session...
[V][ssl_client.cpp:212] start_ssl_client(): Performing the SSL/TLS handshake...
[E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():216]: (-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[E][WiFiClientSecure.cpp:127] connect(): start_ssl_client: -9984
[V][ssl_client.cpp:265] stop_ssl_socket(): Cleaning SSL connection.

Failed.

me-no-dev, 2021-03-10:
seems that your certificates are wrong :) `(-9984) X509 - Certificate verification failed, e.g. CRL, CA or signature check failed`

selvamani1991, 2021-03-10:
Ok, how to i connect the mqtt without certificate verification using the above code.

me-no-dev, 2021-03-10:
Remove:
```cpp
  client.setCACert(test_root_ca);
  client.setCertificate(test_client_cert); // for client verification
  client.setPrivateKey(test_client_key);  // for client verification
```

Add:
```cpp
  client.setInsecure();
```

selvamani1991, 2021-03-10:
thanks, but after i have added the setInsecure(), also mqtt was not connected,

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
[E][esp32-hal-cpu.c:170] setCpuFrequencyMhz(): Can not switch to 240 MH⸮f⸮⸮⸮⸮CPU frequency rated for 160MHz.
[D][esp32-hal-cpu.c:189] setCpuFrequencyMhz(): PLL: 320 / 2 = 160 Mhz, APB: 80000000 Hz
Attempting to connect to SSID: Libre
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
.[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
.[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.1.61, MASK: 255.255.255.0, GW: 192.168.1.1
Connected to Libre

Starting connection to server...
[V][ssl_client.cpp:59] start_ssl_client(): Free internal heap before TLS 280352
[V][ssl_client.cpp:65] start_ssl_client(): Starting socket
[V][ssl_client.cpp:104] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:113] start_ssl_client(): Setting up the SSL/TLS structure...
[I][ssl_client.cpp:127] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
[V][ssl_client.cpp:197] start_ssl_client(): Setting hostname for TLS session...
[V][ssl_client.cpp:212] start_ssl_client(): Performing the SSL/TLS handshake...
[V][ssl_client.cpp:233] start_ssl_client(): Verifying peer X.509 certificate...
[V][ssl_client.cpp:242] start_ssl_client(): Certificate verified.
[V][ssl_client.cpp:257] start_ssl_client(): Free internal heap after TLS 238336
[V][ssl_client.cpp:295] send_ssl_data(): Writing HTTP request with 23 bytes...
[V][ssl_client.cpp:265] stop_ssl_socket(): Cleaning SSL connection.

Failed.

me-no-dev, 2021-03-10:
your log shows that everything went fine. where is the code that prints "Failed."?

selvamani1991, 2021-03-11:
Hi, I have changed the code, 
 Serial.println("\nStarting connection to server...");
  if (!client.connect(server, 8883)){
    Serial.println("Failed");
 }

selvamani1991, 2021-03-11:
But i am using this below code, it shows Faild message.

#include <WiFiClientSecure.h>
#include <PubSubClient.h>

const char *ssid     = "xxx";     // your network SSID (name of wifi network)
const char *password = "xxx"; // your network password

const char *server = "xxx.azure-devices.net";  // Server URL

const char* test_root_ca= \
"-----BEGIN CERTIFICATE-----\n" \

"-----END CERTIFICATE-----\n";

// You can use x.509 client certificates if you want
const char* test_client_key =  \
"-----BEGIN RSA PRIVATE KEY-----\n" \

"-----END RSA PRIVATE KEY-----\n";   //to verify the client

const char* test_client_cert =  \
"-----BEGIN CERTIFICATE-----\n" \

"-----END CERTIFICATE-----\n";  //to verify the client

WiFiClientSecure client;
PubSubClient pubsubclient(client);

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(115200);
  delay(100);

  Serial.print("Attempting to connect to SSID: ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);

  // attempt to connect to Wifi network:
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    // wait 1 second for re-trying
    delay(1000);
  }

  Serial.print("Connected to ");
  Serial.println(ssid);

  client.setCACert(test_root_ca);
  client.setCertificate(test_client_cert); // for client verification
  client.setPrivateKey(test_client_key);  // for client verification
  client.setInsecure();
  pubsubclient.setServer(server, 8883);
  
  Serial.println("\nStarting connection to server...");
  if (!pubsubclient.connected()) {
    String clientId = "emdi11111";
    if (pubsubclient.connect(clientId.c_str())) {
        Serial.println("\nConnected.");        
    } else {
        Serial.print("\nFailed.");
    }
  }
}

void loop() {
  // do nothing
}

-----------------------------------

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
[E][esp32-hal-cpu.c:170] setCpuFrequencyMhz(): Can not switch to 240 MH⸮⸮⸮⸮⸮CPU frequency rated for 160MHz.
[D][esp32-hal-cpu.c:189] setCpuFrequencyMhz(): PLL: 320 / 2 = 160 Mhz, APB: 80000000 Hz
Attempting to connect to SSID: Libre
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
.[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
..[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.1.61, MASK: 255.255.255.0, GW: 192.168.1.1
Connected to Libre

Starting connection to server...
[V][ssl_client.cpp:59] start_ssl_client(): Free internal heap before TLS 280512
[V][ssl_client.cpp:65] start_ssl_client(): Starting socket
[V][ssl_client.cpp:104] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:113] start_ssl_client(): Setting up the SSL/TLS structure...
[I][ssl_client.cpp:127] start_ssl_client(): WARNING: Skipping SSL Verification. INSECURE!
[V][ssl_client.cpp:197] start_ssl_client(): Setting hostname for TLS session...
[V][ssl_client.cpp:212] start_ssl_client(): Performing the SSL/TLS handshake...
[V][ssl_client.cpp:233] start_ssl_client(): Verifying peer X.509 certificate...
[V][ssl_client.cpp:242] start_ssl_client(): Certificate verified.
[V][ssl_client.cpp:257] start_ssl_client(): Free internal heap after TLS 238484
[V][ssl_client.cpp:295] send_ssl_data(): Writing HTTP request with 23 bytes...
[V][ssl_client.cpp:265] stop_ssl_socket(): Cleaning SSL connection.

Failed.

me-no-dev, 2021-03-11:
I would suggest you go and ask in the pubsub repo. Client looks like it connected fine

selvamani1991, 2021-03-11:
I am already asked, but still no reply

TheRealDJ, 2021-03-11:
I don't believe PubSubClient supports TLS. Check out [https://github.com/256dpi/arduino-mqtt](https://github.com/256dpi/arduino-mqtt) `arduino-cli lib install MQTT`

TheRealDJ, 2021-03-11:
FYI: The correct Azure Devices CA cert:
```
// FILE=azure-devices-ca.pem
// CN=*.azure-devices.net
// Subject=CN=*.azure-devices.net
const char* azure_ca = "-----BEGIN CERTIFICATE-----\n" \
"MIIH0TCCBbmgAwIBAgITfwAFBfBz+770HaDrHwAAAAUF8DANBgkqhkiG9w0BAQsF\n" \
"ADBPMQswCQYDVQQGEwJVUzEeMBwGA1UEChMVTWljcm9zb2Z0IENvcnBvcmF0aW9u\n" \
"MSAwHgYDVQQDExdNaWNyb3NvZnQgUlNBIFRMUyBDQSAwMjAeFw0yMDEyMTgyMzMy\n" \
"MTlaFw0yMTEyMTgyMzMyMTlaMB4xHDAaBgNVBAMMEyouYXp1cmUtZGV2aWNlcy5u\n" \
"ZXQwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQDUJZGi5my91i9Gl49y\n" \
"3w2kzUhOPLAm0mh0O8W6ejBk8Fd3MwrMISmcziwJUU008Parj+1phxeQ+3t9NtrJ\n" \
"gxTLZ8aQSMs354kaFtAFSX8OEwO/DVOkzmk8eA8vvHQO6vSBURyaRHPL0QXm4jeS\n" \
"GvYkm7xzSp3rEkwdj5JtAADY9nsUK4evaeqbjw9WltcNzsLp7AS494fRr6IoJBYc\n" \
"dtNKNmBk7pZa8F5jVeaRkLiHJAPzdCgW92/BNP2oOpJoKKO3NqwnELj6aQQT/Mre\n" \
"jVWhMrMW0HYfoFaE9YBUOxNYczujtVQ7axB4st/8w5vA2XuN9DApH24SG4Qwn3LM\n" \
"ta5VAgMBAAGjggPVMIID0TCCAQMGCisGAQQB1nkCBAIEgfQEgfEA7wB1APZclC/R\n" \
"dzAiFFQYCDCUVo7jTRMZM7/fDC8gC8xO8WTjAAABdng7osUAAAQDAEYwRAIgPXvt\n" \
"5+7HwZNWliR/LqeHiIi1SnGBiEZic3ETk2jUKLcCIClZnhQ3U6suV6ipdKxXvV7t\n" \
"uN+lqyiAaiQFHWuw3iSfAHYAVYHUwhaQNgFK6gubVzxT8MDkOHhwJQgXL6OqHQcT\n" \
"0wwAAAF2eDuj9AAABAMARzBFAiEAguWhJ90gbxqkW12F864ajWDoVP7wOSDFp95a\n" \
"9rltiGkCIH/+fpbcsviodJemnMLQSKRTd+nAVDrKez0z9NIc+DQcMCcGCSsGAQQB\n" \
"gjcVCgQaMBgwCgYIKwYBBQUHAwEwCgYIKwYBBQUHAwIwPgYJKwYBBAGCNxUHBDEw\n" \
"LwYnKwYBBAGCNxUIh9qGdYPu2QGCyYUbgbWeYYX062CBXYWGjkGHwphQAgFkAgEl\n" \
"MIGHBggrBgEFBQcBAQR7MHkwUwYIKwYBBQUHMAKGR2h0dHA6Ly93d3cubWljcm9z\n" \
"b2Z0LmNvbS9wa2kvbXNjb3JwL01pY3Jvc29mdCUyMFJTQSUyMFRMUyUyMENBJTIw\n" \
"MDIuY3J0MCIGCCsGAQUFBzABhhZodHRwOi8vb2NzcC5tc29jc3AuY29tMB0GA1Ud\n" \
"DgQWBBSiff1VoAT5EnBQundkRKWFvmi36zALBgNVHQ8EBAMCBLAwXQYDVR0RBFYw\n" \
"VIITKi5henVyZS1kZXZpY2VzLm5ldIIaKi5hbXFwd3MuYXp1cmUtZGV2aWNlcy5u\n" \
"ZXSCISouc3UubWFuYWdlbWVudC1henVyZS1kZXZpY2VzLm5ldDCBsAYDVR0fBIGo\n" \
"MIGlMIGioIGfoIGchk1odHRwOi8vbXNjcmwubWljcm9zb2Z0LmNvbS9wa2kvbXNj\n" \
"b3JwL2NybC9NaWNyb3NvZnQlMjBSU0ElMjBUTFMlMjBDQSUyMDAyLmNybIZLaHR0\n" \
"cDovL2NybC5taWNyb3NvZnQuY29tL3BraS9tc2NvcnAvY3JsL01pY3Jvc29mdCUy\n" \
"MFJTQSUyMFRMUyUyMENBJTIwMDIuY3JsMFcGA1UdIARQME4wQgYJKwYBBAGCNyoB\n" \
"MDUwMwYIKwYBBQUHAgEWJ2h0dHA6Ly93d3cubWljcm9zb2Z0LmNvbS9wa2kvbXNj\n" \
"b3JwL2NwczAIBgZngQwBAgEwHwYDVR0jBBgwFoAU/y9/4Qb0OPMt7SWNmML+DvZs\n" \
"/PowHQYDVR0lBBYwFAYIKwYBBQUHAwEGCCsGAQUFBwMCMA0GCSqGSIb3DQEBCwUA\n" \
"A4ICAQAHRynzSqOyNpQh9NguvLWv+w+/Syz9bLVDtKSCEkOaElQsfC/3WlZIDUOS\n" \
"DiUXA3jjFhRgoG7fE4zW5Vmhbo+/7a922CT3u/LtNh6DqLR7cHqLCe3i/CpIPfRn\n" \
"t7LB7MABSbEPHhaZ+B5kgd2CLsT66eQbKKo4SXFnL/hdG1SCjI0YGTuxPd0oGHg5\n" \
"htCWdoGueumvFezy/eo8VDSiFggyavNNWHUD4qcbKFqOdXFtGDa7+rOREV5i37k7\n" \
"j2sGVSuyyqiC+lA5gYsu2Ex5p+oGBjEag+gmSPTiXM/SRJ4m1zhK1mEHgbypwyEE\n" \
"/JUbVneDQfaJV0Hu/9BuHDPG+71HTMx0RUh8RLfP8UJiIJXJRuN+MX6w2pL9YmOW\n" \
"X60YnsEOAHp+PjCgFWTqT1UEsDBcYbf9YiOWYOl8b2veR01kFD8EvdWv0jdtvfpx\n" \
"FQbIZ1VTxu258yQt91yXHvzRuWxWOaDLAqbKZVByEb1iMwgzxCJVJkASDQ0zxZzf\n" \
"q7LiScLJwPgWjZQupttcgMkhHH1cAcGT4HYK57onXWqwi1Z25tIPYWs9cKTBswya\n" \
"SO4wf6f0WhjM9KmGsFyrKrSCM0N4shwmM8p+PJw5dO9o7b3d5Fn0BqtCURvU/mg4\n" \
"Xu2Orf4bogcTSsnHpke6s7/GU0MwvepsCrwEgsfBSkdQLdM90g==\n" \
"-----END CERTIFICATE-----\n";
```