espressif/arduino-esp32 #4917: Arduino IDE reports Error in <Arduino.h>

ericlangel, 2021-03-09:
In included file: unknown type name '_read'
C:\Users\eric\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.5/tools/sdk/include/newlib\sys/reent.h:195:36:
note: error occurred hereclang(unknown_typename)

Project compiles, but the #include <Arduino.h> is underlined Red and show the Error above

me-no-dev, 2021-03-09:
please show some code that can replicate the issue. We do not see it on our end

ericlangel, 2021-03-09:
[image]

You dont need Code.

Also a new sketch shows this Error.

me-no-dev, 2021-03-09:
you are using Beta IDE. Also does this prevent you from compiling and running the sketch?

ericlangel, 2021-03-09:
Compiling is fine,

i thought that it could be an Error (missing include or something else)
maybe the related functions are unused?

me-no-dev, 2021-03-09:
This is the line that causes the issue: https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/newlib/sys/reent.h#L195

`_READ_WRITE_RETURN_TYPE` expands to `int` which is not an unknown typename :) I guess that `clang` for the beta IDE is not setup correctly and is not picking the correct definition.

closed 2021-03-09 by me-no-dev