espressif/arduino-esp32 #4918: Wifi STA+SoftAP have DHCP Server listening and answering to STA interface

cpainchaud, 2021-03-09:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
Arduino IDE Version :   10805
ESP SDK version:        v3.2.3-14-gd3e562907
IDE name:						Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					940000
Computer OS:                                        Windows 10

### Description:
When STA mode is on along with SoftAP, DHCP server will respond on STA interface as well which creates network issues.
If I add this code: tcpip_adapter_dhcps_stop(TCPIP_ADAPTER_IF_AP) then ofc it stops happening but AP interface is not serving DHCP anymore.
Then if I use tcpip_adapter_dhcps_stop(TCPIP_ADAPTER_IF_STA) it just has no effect.

Logs are showing plenty of:
dhcps: send_offer>>udp_sendto result 0

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

cpainchaud, 2021-03-11:
@me-no-dev same story with the following version:
Arduino IDE Version :	10805
ESP SDK version:	v3.3.4-432-g7a85334d8

me-no-dev, 2021-03-11:
can you please provide minimal example to debug this?

cpainchaud, 2021-03-11:
I supppose it should do it:

```cpp
setup{
WiFi.enableAP(true);
if( !WiFi.softAP("MyAp", "") )
          Serial.println("WIFI AP start FAILED");

WiFi.config(IPAddress((uint32_t) 0), IPAddress((uint32_t) 0), IPAddress((uint32_t) 0));
WiFi.begin("HomeNetwork"), "password");
WiFi.setAutoConnect(true);
WiFi.setAutoReconnect(true);
}
```

me-no-dev, 2021-03-11:
I see the messages but the packets are not going to the STA interface at all

[image]

cpainchaud, 2021-03-11:
Interesting! I was assuming this device was the issue because of these messages and didn't fire a tcpdump.

I think it's worth fixing/ignoring the requests anyways to be 100%clean, your call...

And now I need to find out who's sending these bogus DHCP ...

thank you

closed 2021-07-09 by stale[bot]