espressif/arduino-esp32 #4921: Is running BLE Server and Tasks together... is this possible?

PDE-Paul, 2021-03-09:
Hi,

I need to know is running BLE Server using the following

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLE2902.h>

and Tasks as per below...possible

  xTaskCreatePinnedToCore(
   Task1code,   /* Task function. */
   "RGBTask",     /* name of task. */
   100000,       /* Stack size of task */
   NULL,        /* parameter of the task */
   1,           /* priority of the task */
   &Task1,      /* Task handle to keep track of created task */
   1);          /* pin task to core 1 */      

So far I can only get either one running without the other present, any pointers?

I am using a 16MB ESP32-WROOM-32
Model: 1; Features: 0x32; Cores: 2; Revision: R1; SDK: v3.3.4-432-g7a85334d8
Arduino 1.8.13

Any pointers will be most appreciated.....

Cheers Paul

lbernstone, 2021-03-09:
See https://github.com/espressif/arduino-esp32/blob/1.0.5/libraries/ESP32/examples/FreeRTOS/FreeRTOS.ino
You must put delays in your tasks to allow other processes CPU time to run.

PDE-Paul, 2021-03-11:
Thank you Ibernstone, I got it working. Many Thanks.

closed 2021-03-11 by PDE-Paul