espressif/arduino-esp32 #4925: Is websocket not supported in v1.0.5?

CelliesProjects, 2021-03-10:
### Question:
Is websocket not supported in v1.0.5?
Or do I need to include another header file? Or some other way to fix this?

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.5
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Ubuntu

### Description:
When I add the http server and try to use websockets I get the compilation error `'httpd_ws_frame_t' was not declared in this scope`
See the sketch below.

### Sketch:
```cpp
#include <Arduino.h>
// https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_http_server.html

#include <WiFi.h>
#include <esp_http_server.h>

/*
 * Structure holding server handle
 * and internal socket fd in order
 * to use out of request send
 */
struct async_resp_arg {
    httpd_handle_t hd;
    int fd;
};

/*
 * This handler echos back the received ws data
 * and triggers an async send if certain message received
 */
static esp_err_t echo_handler(httpd_req_t *req)
{
    uint8_t buf[128] = { 0 };
    httpd_ws_frame_t ws_pkt;
    memset(&ws_pkt, 0, sizeof(httpd_ws_frame_t));
    ws_pkt.payload = buf;
    ws_pkt.type = HTTPD_WS_TYPE_TEXT;
    esp_err_t ret = httpd_ws_recv_frame(req, &ws_pkt, 128);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "httpd_ws_recv_frame failed with %d", ret);
        return ret;
    }
    ESP_LOGI(TAG, "Got packet with message: %s", ws_pkt.payload);
    ESP_LOGI(TAG, "Packet type: %d", ws_pkt.type);
    if (ws_pkt.type == HTTPD_WS_TYPE_TEXT &&
        strcmp((char*)ws_pkt.payload,"Trigger async") == 0) {
        return trigger_async_send(req->handle, req);
    }

    ret = httpd_ws_send_frame(req, &ws_pkt);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "httpd_ws_send_frame failed with %d", ret);
    }
    return ret;
}

static const httpd_uri_t ws = {
        .uri        = "/ws",
        .method     = HTTP_GET,
        .handler    = echo_handler,
        .user_ctx   = NULL,
        .is_websocket = true
};

esp_err_t root_handler(httpd_req_t *req)
{
  /* Send a simple response */
  const char* resp = (const char*) req->user_ctx;
  httpd_resp_send(req, resp, HTTPD_RESP_USE_STRLEN);
  return ESP_OK;
}

esp_err_t get_handler(httpd_req_t *req)
{
  /* Send a simple response */
  const char* resp = "URI GET Response";
  httpd_resp_send(req, resp, HTTPD_RESP_USE_STRLEN);
  return ESP_OK;
}

const  httpd_uri_t root_get = {
  .uri      = "/",
  .method   = HTTP_GET,
  .handler  = root_handler,
  .user_ctx = (char*)"Hello World!"
};

const httpd_uri_t uri_get = {
  .uri      = "/uri",
  .method   = HTTP_GET,
  .handler  = get_handler,
  .user_ctx = NULL
};

httpd_handle_t start_webserver(void)
{
  /* Generate default configuration */
  httpd_config_t config = HTTPD_DEFAULT_CONFIG();

  /* Empty handle to esp_http_server */
  httpd_handle_t server = NULL;

  /* Start the httpd server */
  if (httpd_start(&server, &config) == ESP_OK) {
    /* Register URI handlers */
    httpd_register_uri_handler(server, &root_get);
    httpd_register_uri_handler(server, &uri_get);
  }
  /* If server failed to start, handle will be NULL */
  return server;
}

void stop_webserver(httpd_handle_t server)
{
  if (server) {
    /* Stop the httpd server */
    httpd_stop(server);
  }
}

void setup() {
  Serial.begin(115200);

  WiFi.begin("huiskamer","0987654321");
  while (!WiFi.isConnected()) {
    delay(10);
  }

  Serial.println(WiFi.localIP());
  
  
  httpd_handle_t server = start_webserver();

  if (!server) {
    Serial.println("Could not start server");
  }
  
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(2);
}
```

### Compilation Messages:
```
'httpd_ws_frame_t' was not declared in this scope
```

me-no-dev, 2021-03-10:
ESP32-Arduino 1.0.5 uses IDF v3.3. You are quoting the latest IDF doc, which is ahead of IDF 3.3. I guess it's not backported to to v3.3

closed 2021-03-10 by me-no-dev

CelliesProjects, 2021-05-19:
@me-no-dev I just started testing 2.0.0-alpha.

I see that `CONFIG_HTTPD_WS_SUPPORT` is still not set?

`# CONFIG_HTTPD_WS_SUPPORT is not set` -> L651 in `.arduino15/packages/esp32/hardware/esp32/2.0.0-alpha1/tools/sdk/esp32`