espressif/arduino-esp32 #4928: I2C communication problem between ESP32 and STM8

ozgurkoc0, 2021-03-10:
### Hardware:

Board: Custom w/ ESP32-WROVER-E 
IDE name: Arduino IDE
Flash Frequency: 80Mhz
Upload Speed: 921600
Computer OS: Windows 10

### Description:
I have a ESP32-WROVER-E and a STM8S003F3. ESP32 is master and STM8 is slave. STM8 runs at 3.3V.

I want to communicate with I2C between them but I am getting the following debug messages. I used 10K, 4.7K and 1K pull-up resistors but nothing changed. If I set the ESP32 I2C clock frequency to 62.5KHz or below, it works correctly and I don't get errors. Why am I getting error at 100KHz? 

ESP32 worked at 100KHz with any sensor, I tested it.
I ran the same code at 100KHz between Arduino Mega and STM8 and it worked correctly, no errors.

### Sketch:  
```cpp
#include <Wire.h>
#include <Arduino.h>

void setup() {

  Wire.begin(4,15,100000);
  Serial.begin(9600);

}

void loop() {
  
  Wire.beginTransmission(0x3);
  Wire.write(0x00);
  Wire.endTransmission();

  delay(1);

  if(Wire.requestFrom(0x3,1))
    Wire.read();

  delay(10); 
}
```

### Debug Messages:
```
[D][esp32-hal-i2c.c:1344] i2cProcQueue():  Busy Timeout start=0x2f0a, end=0x2f3c, =50, max=50 error=1
[E][esp32-hal-i2c.c:318] i2cDumpI2c(): i2c=0x3ffbebe8
[I][esp32-hal-i2c.c:319] i2cDumpI2c(): dev=0x60013000 date=0x16042000
[I][esp32-hal-i2c.c:321] i2cDumpI2c(): lock=0x3ffb81cc
[I][esp32-hal-i2c.c:323] i2cDumpI2c(): num=0
[I][esp32-hal-i2c.c:324] i2cDumpI2c(): mode=1
[I][esp32-hal-i2c.c:325] i2cDumpI2c(): stage=3
[I][esp32-hal-i2c.c:326] i2cDumpI2c(): error=1
[I][esp32-hal-i2c.c:327] i2cDumpI2c(): event=0x3ffb84c0 bits=0
[I][esp32-hal-i2c.c:328] i2cDumpI2c(): intr_handle=0x3ffb84f0
[I][esp32-hal-i2c.c:329] i2cDumpI2c(): dq=0x3ffb849c
[I][esp32-hal-i2c.c:330] i2cDumpI2c(): queueCount=1
[I][esp32-hal-i2c.c:331] i2cDumpI2c(): queuePos=0
[I][esp32-hal-i2c.c:332] i2cDumpI2c(): errorByteCnt=0
[I][esp32-hal-i2c.c:333] i2cDumpI2c(): errorQueue=0
[I][esp32-hal-i2c.c:334] i2cDumpI2c(): debugFlags=0x00000000
[I][esp32-hal-i2c.c:311] i2cDumpDqData(): Debug Buffer not Enabled
[I][esp32-hal-i2c.c:354] i2cDumpInts(): Debug Buffer not Enabled
[W][esp32-hal-i2c.c:1419] i2cCheckLineState(): invalid state sda(4)=1, scl(15)=0
[D][esp32-hal-i2c.c:1427] i2cCheckLineState(): Recovered after 1 Cycles
[E][esp32-hal-i2c.c:1434] i2cCheckLineState(): Bus Invalid State, TwoWire() Can't init sda=1, scl=0
```

chegewara, 2021-03-10:
What version/branch of arduino-esp32 are you using?

ozgurbostan, 2021-03-10:
Hi @chegewara,

We are working on same project with @ozgurkoc0, so I can give you the answer in terms of him; it's ESP32 Arduino 1.0.4 (IDF 3.2) Release.

me-no-dev, 2021-03-10:
I can suggest to try two things:
1. Update to 1.0.5
2. Use some other GPIO instead of 15

4.7 Kohm Pull-ups should be fine

ozgurbostan, 2021-03-10:
Hi @me-no-dev,

Why you recommend to try another GPIO instead of 15?

me-no-dev, 2021-03-11:
it is a bootstrap pin and could be causing issues. it was just a suggestion

ozgurkoc0, 2021-03-11:
Hi @me-no-dev,
I updated to 1.0.5 and used another GPIO pins (SDA: 23 & SCL: 22) but nothing changed. I'm getting the same debug message.

me-no-dev, 2021-03-11:
Maybe @stickbreaker can shed some light?

ozgurbostan, 2021-03-16:
Hi @stickbreaker, you have any comment? 

It's a really weird situation related to i2c clock speed. When it is tuned over 62.5kHz, it gets stuck. Lower than that value, there is no problem at all. All the functions and transactions are working and executing properly. The problem arises in "i2cProcQueue()" function while branching to get eBits value (ISR related). 

If you have ever encountered such a problem and can recommend us a solution, it would be really appreciated.

Thanks in advance for your support and interest.

dalbert2, 2021-04-23:
Not sure if this helps, but I am using the ESP32 with an STM32G0 co-processor and I2C works just fine at 1MHz (Fast Mode Plus), so it doesn't seem likely that it's a problem in the ESP32 Wire implementation.  I'm using 4k7 pull-ups.

closed 2021-07-08 by stale[bot]