espressif/arduino-esp32 #4929: BluetoothSerial let a device to be discoverable only with Android BT settings, any other App doesn't find the device

lunard, 2021-03-10:
Hello to all, I'm using an ESP32  (az-delivery Dev Kit C v4) with BluetoothSerial.h and I want to use an Android app to discover the ESP32.

It seems not possible to discover the ESP32 via app.
I tested the issue using:
- BLE Scanner
- Serial Bluetooth Terminal
- my own Flutter app, using the package flutter_blue

With the default Android BT setting app, I'm able to discover (and pair) the ESP32, but not with any other Android app.

Someone could give me some suggestions? Thanks!

lbernstone, 2021-03-10:
https://github.com/espressif/arduino-esp32/tree/master/libraries/BluetoothSerial

chegewara, 2021-03-10:
> * BLE Scanner

BluetoothSerial is classic BT, so you need bluetooth classic app on adroid.

lunard, 2021-03-10:
> https://github.com/espressif/arduino-esp32/tree/master/libraries/BluetoothSerial

Thanks, but I already used the I follower

lunard, 2021-03-10:
> > * BLE Scanner
> 
> BluetoothSerial is classic BT, so you need bluetooth classic app on adroid.

But @chegewara I found lot of example using BluetoothSerial.h on ESP32 side and flutter_blue (but also Serial Bluetooth Terminal as test app) on Android side

closed 2021-03-10 by lunard

chegewara, 2021-03-10:
https://github.com/espressif/arduino-esp32/tree/master/libraries/BluetoothSerial#bluetooth-serial-library
> Bluetooth Serial Library
A simple Serial compatible library using ESP32 classical bluetooth (SPP)