espressif/arduino-esp32 #4930: Serial.readbytes() return wrong

hicham-vv, 2021-03-11:
### Hardware:
Board: esp32 az-delivery-devkit-v4							?
framework:                                              arduino
IDE name:	                                        Platform.io
Upload Speed:					4800 used  for the communication bettwen SIM800 and my esp32
Computer OS:                                        Windows 10

### Description:
hello guys, hope you doing good, 
i am new with esp32, and i am facing a problem with the serial port number of my esp32, 
any way i am sending a firmware.bin file to my esp32 via the serial port number 2, and for that i am reading 1024bytes per AT command ( because my SIM800 can't go max than that ), 
so in the begining everything is good, i can read until  like 118kb, but after that when i try readbytes from the serial, something went wrong , i got always 0 as return of the function Serial2.readbytes(mybuffer,1024).

the size of my firmware is about 200kbytes.

i tried to restart the serail2.port but nothing happened.

here its my sketch of reading data.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
// boucle to read all the data of the firmware
for (int i=0;i<ii;i++){

  Serial2.println("AT+FTPGET=2,1024");
  Serial2.readBytes(rembytes,36); // remove some bytes unuseful
  mastaille=Serial2.readBytes(masbytes,1024); // reading the 1024 bytes of the firmware.bin 
  Serial.println(mastaille); // print the retrun of Serial.readbytes ( the number of bytes readed)
  Serial2.readBytes(rembytes,36);// remove some bytes unuseful

  if(mastaille==1024) // check if mastaille ==1024 and then write the bytes to the file
  file.write(masbytes,1024);
  else Serial.println("something wrong");

  Serial.println(i);
  

  }

chegewara, 2021-03-11:
Did you confirm that SIM800 is sending anything when start reading `mastaille==0`?
I think you may have wrong logic in your `for` loop code. What you do is:
- discard 36 bytes, // do you really need to discard this every iteration?
- read 1024 bytes,
- discard 36 bytes,  // do you really need to discard this every iteration?
- loop.

As far as i understand you are requesting 1024 bytes from SIM800:
`Serial2.println("AT+FTPGET=2,1024");`
then you read from UART 36 bytes and discard, then you read 1024 bytes from UART, but here should be only `1024 - 36` left, and at the end you read another 36 bytes from UART. That seems to be not right.

hicham-vv, 2021-03-11:
Hei @chegewara, thank you for your reply , 

here it's why i need to remove the 36 bytes of the begining, 
when i command the AT+FTPGET = 2,1024 rx data got all those bytes 

so i need to remove the two line than start reading the 1024 data needed , and after that 

**AT+FTPGET=2,1024
+FTPGET: 2,1024**

01234567890123456789012345678901234
5678901234567890…..1234

**OK
+FTPGET:1,0**

chegewara, 2021-03-11:
Ok, in that case all is fine, but i would add some code that is checking if you really get `OK`, because there may be some error in communication with SIM800 you are missing. Also it would be good to confirm somehow, maybe with logic analyzer, that SIM800 is sending data when `Serial2` is reading 0 bytes.

So, my suggestion is to add code to print AT "header" (36 bytes in front) and "footer" (last 36 bytes) to see in logs whats going on.

hicham-vv, 2021-03-11:
ok i think i found the problem, when i send AT+FTPGET=2,1024 for the 116 time, i receive 
**AT+FTPGET=2,1024

ERROR**
and i found that is mean ""Read data after FTP session is stopped. ""

chegewara, 2021-03-11:
Probably timeout, but you have to check this out. As for this issue, i think it is resolved and can be closed.

hicham-vv, 2021-03-11:
Yes thank you, i will check that , i appreciate your help 
i will close it

closed 2021-03-11 by hicham-vv