espressif/arduino-esp32 #4931: HTTPClient crash on GET() for url with 1 redirect

cpainchaud, 2021-03-11:
### Hardware:
Board:							ESP32 Dev Module
Arduino IDE Version :	10805
ESP SDK version:	v3.3.4-432-g7a85334d8
IDE name:						PIO
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                   Windows 10

### Description:
Crash happens when downloading URL with http.setFollowRedirects(followRedirects_t::HTTPC_STRICT_FOLLOW_REDIRECTS);
This URL indeeds forward the client once to the read URL where the binary file is.
Note that the crash does not occur if I give it the relocated URL directly.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

void setup()
{
  Serial.begin(115200);
  delay(10);

  WiFi.config(IPAddress((uint32_t) 0), IPAddress((uint32_t) 0), IPAddress((uint32_t) 0));
  WiFi.begin("my_network", "my_password");
  WiFi.setAutoConnect(true);
  WiFi.setAutoReconnect(true);
  Serial.println("wait a bit for Wifi to be up");
  delay(10000);

  HTTPClient http;

  Serial.println("Requesting web page now!");

  http.setFollowRedirects(followRedirects_t::HTTPC_STRICT_FOLLOW_REDIRECTS);
  http.begin("https://github.com/cpainchaud/RFLink/releases/download/nightly/esp32-firmware.bin");
  int httpCode = http.GET();

}

void loop()
{

}
```

### Debug Messages:
```
If you have a Guru Meditation Error or Backtrace, ***please decode it***:
  0  0x400e1235:0x3ffe7f00 in WiFiClient::connect(char const*, unsigned short, int) at /packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiClient.cpp:572
  1  0x400f0876:0x3ffe7f30 in HTTPClient::connect() at /packages/framework-arduinoespressif32/libraries/HTTPClient/src/HTTPClient.cpp:925
  2  0x400f08f7:0x3ffe7f60 in HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at /packages/framework-arduinoespressif32/libraries/HTTPClient/src/HTTPClient.cpp:925
  3  0x400f0a73:0x3ffe7fb0 in HTTPClient::GET() at /packages/framework-arduinoespressif32/libraries/HTTPClient/src/HTTPClient.cpp:925
  4  0x400d7777:0x3ffe7fd0 in RFLink::OTA::downloadFromUrl(char const*) at RFLink/13_OTA.cpp:29

```

me-no-dev, 2021-03-11:
I know that PIO can use the development version of this repo. Please check in their docs how to do it and try against that. I added a fix exactly for redirects

cpainchaud, 2021-03-11:
platform = https://github.com/platformio/platform-espressif32.git should do it, i will give it a try !

cpainchaud, 2021-03-11:
Nope :/

Arduino IDE Version :	10805
ESP SDK version:	v3.3.4-436-g0bfff0b25

  0  0x00000000:0x3ffe7e60 in ?? ??:0
  #1  0x400f18d9:0x3ffe7e80 in HTTPClient::connect() at C:/Users/christophe.painchaud/.platformio/packages/framework-arduinoespressif32/libraries/HTTPClient/src/HTTPClient.cpp:925
  2  0x400f1a4b:0x3ffe7eb0 in HTTPClient::sendRequest(char const*, unsigned char*, unsigned int) at C:/Users/christophe.painchaud/.platformio/packages/framework-arduinoespressif32/libraries/HTTPClient/src/HTTPClient.cpp:925
  3  0x400f1bc7:0x3ffe7f00 in HTTPClient::GET() at C:/Users/christophe.painchaud/.platformio/packages/framework-arduinoespressif32/libraries/HTTPClient/src/HTTPClient.cpp:925

me-no-dev, 2021-03-16:
Took me a while but I figured it out :) fixed in the commit above

closed 2021-03-16 by me-no-dev