espressif/arduino-esp32 #4933: BLE kills Esp32MQTTClient.h

PDE-Paul, 2021-03-11:
Hi,

Why does BLE prevent Azure connection using Esp32MQTTClient.h?

I have tried adding delays with no luck.

I have managed to get the following working with Tasks

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLE2902.h>

but not with 

#include <Esp32MQTTClient.h>

I can get Azure connection if I remove the BLE and visa versa.

Any pointers would be most appreciated...

Kind Regards
Paul

chegewara, 2021-03-11:
Did you check free heap/RAM?

PDE-Paul, 2021-03-11:
No, will do shortly and revert back to you

me-no-dev, 2021-03-11:
there was a coexist issue in 1.0.5 if you are using that, please test against the current master branch

PDE-Paul, 2021-03-12:
Hi,

I cloned the current master branch, recompiled, programmed and powered up.... results are the same

> Total Free Heap: 36092
> Connect to Azure
> IoT Hub
Info: Initializing SNTP

Info: SNTP initialization complete

Info: IoT Hub SDK for C, version 1.1.23

Error: Time:Fri Mar 12 07:24:49 2021 File:C:\Users\User\Documents\Arduino\hardware\espessif\esp32\libraries\AzureIoT\src\az_iot\c-utility\pal\src\tlsio_openssl_compact.c Func:create_ssl Line:493 SSL_new failed

Info: >>>Connection status: disconnected

Info: >>>Connection status: disconnected

Info: tlsio_openssl_close has been called when in neither TLSIO_STATE_OPEN nor TLSIO_STATE_ERROR.

Error: Time:Fri Mar 12 07:24:50 2021 File:C:\Users\User\Documents\Arduino\hardware\espessif\esp32\libraries\AzureIoT\src\az_iot\c-utility\pal\src\tlsio_openssl_compact.c Func:create_ssl Line:493 SSL_new failed

Info: >>>Connection status: disconnected

Info: >>>Connection status: disconnected

Info: tlsio_openssl_close has been called when in neither TLSIO_STATE_OPEN nor TLSIO_STATE_ERROR.

Error: Time:Fri Mar 12 07:24:51 2021 File:C:\Users\User\Documents\Arduino\hardware\espessif\esp32\libraries\AzureIoT\src\az_iot\c-utility\pal\src\tlsio_openssl_compact.c Func:create_ssl Line:493 SSL_new failed

Info: >>>Connection status: disconnected

Info: >>>Connection status: disconnected

Info: tlsio_openssl_close has been called when in neither TLSIO_STATE_OPEN nor TLSIO_STATE_ERROR.

Error: Time:Fri Mar 12 07:24:55 2021 File:C:\Users\User\Documents\Arduino\hardware\espessif\esp32\libraries\AzureIoT\src\az_iot\c-utility\pal\src\tlsio_openssl_compact.c Func:create_ssl Line:493 SSL_new failed

Info: >>>Connection status: disconnected

Info: >>>Connection status: disconnected

Info: tlsio_openssl_close has been called when in neither TLSIO_STATE_OPEN nor TLSIO_STATE_ERROR.

Error: Time:Fri Mar 12 07:25:03 2021 File:C:\Users\User\Documents\Arduino\hardware\espessif\esp32\libraries\AzureIoT\src\az_iot\c-utility\pal\src\tlsio_openssl_compact.c Func:create_ssl Line:493 SSL_new failed

Info: >>>Connection status: disconnected

Info: >>>Connection status: disconnected

Info: tlsio_openssl_close has been called when in neither TLSIO_STATE_OPEN nor TLSIO_STATE_ERROR.

> Initializing IoT Hub :- Failed
> Total Free Heap: 30904

If I remove the BLE functions Azure connection is successful.. so any ideas or pointer on how to resolve this would be greatly appreciated.

Cheers
Paul

chegewara, 2021-03-12:
You are short on heap to perform SSL connection. Solution may be @h2zero NimBLE library.

PDE-Paul, 2021-03-12:
Thank you, will try your suggestion over the weekend.

PDE-Paul, 2021-03-13:
Thank you, this now works.

closed 2021-03-13 by PDE-Paul