espressif/arduino-esp32 #4934: Stop when change the baudrate

qlalfdu, 2021-03-11:
Board: ESP32 Dev Module
Version/Date: 1.0.5, today's master
IDE name: Arduino IDE

Below is my sketch

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  Serial.println("bauds=9600");
  delay(1000);
  Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
 delay(1000);
 Serial.println(millis());
}

### Debug Messages:
No debug messages
Just STOP and do nothing
[image]

chegewara, 2021-03-11:
Dont try to `begin()` twice, better update bitrate:
`Serial.updateBaudRate(bitrate);`

qlalfdu, 2021-03-12:
It worked in 1.0.4, but it didn't work in 1.0.5, so I was embarrassed.
Serial.updateBaudRate(bitrate);
It works fine.
Thank you.

closed 2021-03-12 by qlalfdu