espressif/arduino-esp32 #4939: Install 1.0.4 on raspbian 64-bit / arm64

AkbarRamzan, 2021-03-13:
How can I install arduino-esp32 1.0.4 on a raspberry pi 4 64-bit?

1.0.5 release candidates can be installed through the board manager very easily. However, it has some known bugs which prevents some of the libraries I am using from working (undefined references such as to GAP_ConnOpen when compiling). What I would like to do is install 1.0.4 instead. Unfortunately, this gives an error: "Tool xtensa-esp32-elf-gcc is not available for your operating system". Can you please advise on what to do to install 1.0.4 on a pi (64-bit)? 

If I need to compile something myself, I can do that but I will need a few instrucitons.
Thanks

lbernstone, 2021-03-13:
You could try to backport from https://github.com/koendv/xtensa-esp32-elf-raspberrypi.
Or, you could move forward to a supported platform and solve your issues in 1.0.5

AkbarRamzan, 2021-03-13:
1.0.5 itself is buggy. I will try and backport from the link you provided.
I know 1.0.4 works as I have tested it on my windows machine 
Thanks

lbernstone, 2021-03-13:
If nobody reports issues in 1.0.5, it will remain buggy.  Free code isn't free.

AkbarRamzan, 2021-03-13:
I believe the author of the library who found the bug is planning on raising the issue

AkbarRamzan, 2021-03-14:
I seem to be in a catch-22 situation: https://github.com/koendv/xtensa-esp32-elf-raspberrypi/issues/1

closed 2021-07-09 by stale[bot]