espressif/arduino-esp32 #4940: I2C maximum clock speed 800 kHz

tshcherban, 2021-03-13:
### Hardware:
Board:							WEMOS TTGO Mini D1 ESP32 4MB flash, 240 MHz CPU, Partition 1.2MB APP/1.5MB SPIFFS
Core Installation version:			1.0.5
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
Trying to set I2C clock above 800 kHz:

### Code
```
Wire.begin(-1, -1, 1000000); // -1 for default pins
```
it fails:

### Debug Messages:
```
[D][esp32-hal-i2c.c:1676] i2cSetFrequency(): i2c freq(1000000) out of bounds.vs APB Clock(80000000), min=9765, max=800000
[D][esp32-hal-i2c.c:1681] i2cSetFrequency(): APB Freq too slow, Reducing i2c Freq to 800000 Hz
```
however according to chip specs it is capable up to 5 MHz (with limitations of SDA pullups). Is it an arduino core limitation?

lbernstone, 2021-03-13:
Yes, as it quite clearly says. Max of the device is 1MHz in IDF, so you aren't missing much.

tshcherban, 2021-03-13:
So that
[image]
is not reachable even via their SDK?

lbernstone, 2021-03-13:
AFAIK the driver only supports fast mode.  If you wrote a driver for HS or US, then the hardware could likely handle it.

tshcherban, 2021-03-13:
Crap... I'm afraid my knowledge is far beyond this.
So ill try to look into seeduino xiao (hope there situation is better).
Thanks for explanation.

closed 2021-03-13 by tshcherban