espressif/arduino-esp32 #4945: problem uploading any code on my esp32cam board

SUperMattew043, 2021-03-14:
Hi I Buy from amazon (https://www.amazon.it/ESP32-CAM-Bluetooth-Development-Adattattore-Seriale/dp/B0893FBLFR/ref=sr_1_8?__mk_it_IT=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=esp32+cam&qid=1615721112&sr=8-8)

this esp32cam board which include the adaptor and some jumpers, I connect correctly the jumpers to FTDI module, then I connect a jumper bewteen GND and IO0 and I go in the arduino IDE (I'm on windows) I select the AI thinker board but when I go on ports there is only the native port even if the usb cable is connected from the computer to the adaptor, so I tried to upload the sketch in examples>esp32>camera>cameraWebServer I selected the AI thinker module and I insert my Wifi credentials but after compliling the sketch it says:

esptool.py v3.0-dev
Serial port COM3
Connecting...

Traceback (most recent call last):
File "esptool.py", line 3682, in
File "esptool.py", line 3675, in _main
File "esptool.py", line 3330, in main
File "esptool.py", line 512, in connect
File "esptool.py", line 492, in _connect_attempt
File "esptool.py", line 431, in sync
File "esptool.py", line 369, in command
File "esptool.py", line 332, in write
File "site-packages\serial\serialwin32.py", line 323, in write
serial.serialutil.SerialTimeoutException: Write timeout
Failed to execute script esptool

I tried to install some drivers, restarting pc but I cant upload that sketch so I tried another examples for this board and it gives me same error, I search on internet and there is a similar case but he solved the problem by installing a driver, so i tried to install thah driver but doesn't work. I tried to change usb cable and even pc but it doesn't work. When I plug my usb cable the pc makes a sound and if I go to device manager there is FT23R2 USB UART connected. What can I do?

lbernstone, 2021-03-14:
You could follow the instructions in the [issue template](https://github.com/espressif/arduino-esp32/blob/master/docs/ISSUE_TEMPLATE.md)

Al-Imran, 2021-03-14:
Perhaps you are new here. You better stick to the template when you ask for solutions (see previous comment). For this time, I am going to give you a head start. Without the actual board info, IDE, and other settings, it is somewhat difficult to suggest you something. Reply with a templated query if the following does not solve.

A few things to consider to troubleshoot:

1. Make sure you have selected the right board. Perhaps its a clone. Try choosing `#define CAMERA_MODEL_WROVER_KIT` or `#define CAMERA_MODEL_AI_THINKER` within the Arduino sketch. Similarly, select your board as **ESP32 Wrover Module** or **ESP32 Dev Module** from Tools.
2. If you are new to this hobby world, make sure your wiring to the FTDI/ESP is correct. Pay special attention to voltage/logic levels. Although ESP is 3.3V, while programming via FTDI modules, a supply of 3.3V may not be able to push enough current. If so, supply 5V from the FTDI to the 5V pin of ESP. In case you have two separate power supplies, you better hope you know what you are doing.
3. Shorting the IO0 to the GND pin on its own does nothing unless: (a) You have shorted those pins and then supplied power, (b) You have shorted afterwards of supplying voltage and pressed the reset button. The whole idea of this shorting of IO0 pin is to enable upload of new program. You could simply open up the serial monitor (perhaps at 115200 baud rate) and look for `waiting for download`.
4. It might have something to do with the board configuration. I will wait for you reply with the relevant information. 

**_Make sure understand these points before you proceed_**. Good luck!

SUperMattew043, 2021-03-15:
> Perhaps you are new here. You better stick to the template when you ask for solutions (see previous comment). For this time, I am going to give you a head start. Without the actual board info, IDE, and other settings, it is somewhat difficult to suggest you something. Reply with a templated query if the following does not solve.
> 
> A few things to consider to troubleshoot:
> 
> 1. Make sure you have selected the right board. Perhaps its a clone. Try choosing `#define CAMERA_MODEL_WROVER_KIT` or `#define CAMERA_MODEL_AI_THINKER` within the Arduino sketch. Similarly, select your board as **ESP32 Wrover Module** or **ESP32 Dev Module** from Tools.
> 2. If you are new to this hobby world, make sure your wiring to the FTDI/ESP is correct. Pay special attention to voltage/logic levels. Although ESP is 3.3V, while programming via FTDI modules, a supply of 3.3V may not be able to push enough current. If so, supply 5V from the FTDI to the 5V pin of ESP. In case you have two separate power supplies, you better hope you know what you are doing.
> 3. Shorting the IO0 to the GND pin on its own does nothing unless: (a) You have shorted those pins and then supplied power, (b) You have shorted afterwards of supplying voltage and pressed the reset button. The whole idea of this shorting of IO0 pin is to enable upload of new program. You could simply open up the serial monitor (perhaps at 115200 baud rate) and look for `waiting for download`.
> 4. It might have something to do with the board configuration. I will wait for you to reply with the relevant information.
> 
> **_Make sure understand these points before you proceed_**. Good luck!

SUperMattew043, 2021-03-15:
im sorry if I waste your time but i resolved, i think the problem was the ftdi adaptor, i try to program esp32 with arduino and everything succefully worked. i understand it was a port ptoblem because even when if i see a new devide on windows device manager, arduino still not recognise the esp32 module so as I said, I try to code it with arduino and it worked, hope this can help someone, thanks everybody that tried to help me

SUperMattew043, 2021-03-15:
i think I accidentaly closed the issue... im sorry

closed 2021-03-15 by SUperMattew043