espressif/arduino-esp32 #4946: BLE stack hangs in unrecoverable way on ESP32 DEVKIT V1

aovestdipaperino, 2021-03-14:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.3, 1.0.4, 1.0.5
IDE name:						Arduino IDE, Platform.io
Upload Speed:					115200
Computer OS:                                        Mac OSX

### Description:
As soon as BLEDevice::init() is invoked, the device hangs and the serial port goes haywire.
Tried on 2 identical dev boards.
Tried on 2 different computers.
Tried different versions of platform (2.1.0, 3.0.0, 3.1.0) and core (1.0.3, 1.0.4, 1.0.5).
Tried 2 different cables
Tried both on Arduino and Platformio

Different model of boards work, including a dev4.

I already bought newer replacement boards but I won't be able to sleep at night as this was my main dev-board and BLE was working fine up to few weeks ago.

### Sketch:
```cpp

#include <Arduino.h>
#include <BLEDevice.h>

void setup() {
   Serial.begin(115200);
   BLEDevice::init("");
   Serial.println("never here");
}

void loop() {
}
```

### Debug Messages:
```Exception in thread rx: ```
(this happens because the Serial port becomes unresponsive).

chegewara, 2021-03-14:
@ possible issues:
- power supply, including cable,
- some conflicting pins used on that board, but it is odd, because BLE library does not use and/or configure any pins.

aovestdipaperino, 2021-03-14:
Yeah, tried cable.
And tried with a board on an empty breadboard.

I debugged it a bit and it's in the first call that hangs, it never gets to the line that waits for idle.

aovestdipaperino, 2021-03-14:
[This is the one](https://www.amazon.com/gp/product/B07Q576VWZ) I bought from Amazon which doesn't work.
An HW issue might also be a possibility.

h2zero, 2021-03-17:
I'm betting on a bad voltage regulator on that board. If you can, try supplying 3.3v from an external source.

closed 2021-07-11 by stale[bot]