espressif/arduino-esp32 #4948: ESP32 1.0.5 not working with M5Stack lib. Serial.begin() with M5.begin() stops running the code

PatKeel, 2021-03-15:
After updating esp32 board package from 1.0.4 to 1.0.5 my projects with M5Stack Core Gray and Black (Basic) did not work anymore. It compiles but did not came over setup into the loop.
Trying several combinations of installed libs i found one combination makes this issue: esp32 1.0.5! All other constellations worked for me.
If i comment the line with Serial.begin() it works and serial output works tough at 115200baud.
If i remove all M5 stuff (M5.Lcd) the Serial.begin() statement works.

### Hardware:
Board:							M5Stack Core GRAY and BASIC
IDE name:						Arduino 1.8.13
Flash Frequency:					80Mhz
Upload Speed:					921600
Computer OS:                                        Windows 10

`#include <M5Stack.h>

int i = 0;

void setup(void) {
  Serial.begin(115200);     // comment this line if installed esp32 lib 1.0.5!

  M5.begin();
  M5.Power.begin();
  M5.Lcd.setCursor(10, 20, 4);
  M5.Lcd.setTextColor(TFT_YELLOW);  
  M5.Lcd.setTextSize(1);
  M5.Lcd.print("Hello World!");
  delay(2000);
  M5.Lcd.clear(TFT_BLACK);
}

void loop() {
  M5.Lcd.setCursor(10, 20, 4);
  M5.Lcd.fillRect(10, 20, 100, 19, TFT_BLACK);
  M5.Lcd.println(i);

  Serial.println(i);
  i++;
  delay(1000);
}`

Regards Patrick

felmue, 2021-03-15:
Hello @PatKeel 

if you look into `M5.begin()` you'll see that there already is a `Serial.begin()` call.
```
  if (SerialEnable == true) {
    Serial.begin(115200);
    Serial.flush();
    delay(50);
    Serial.print("M5Stack initializing...");
  }
```
In my experience calling `Serial.begin()` twice is a bad idea and can lead to very strange errors.
Thanks
Felix

PatKeel, 2021-03-15:
Thank you Felix

closed 2021-03-15 by PatKeel

me-no-dev, 2021-03-15:
I investigated this possible issue with multiple `Serial.begin()` and could not get the code to not run. What I do get is that the `Serial` is not immediately available for writing. Also some prints in between `Serial.begin()` calls could also get lost for some reason. I poked at registers, tried more complex `Flush` and all... Everything looked normal but prints were not coming through.

felmue, 2021-03-15:
Hello @me-no-dev 

thank you for your feedback and taking the time to do extra tests. Much appreciated.

Cheers
Felix

SuGlider, 2021-07-19:
Issue is solved with PR #5385 using version 2.0.0Alpha