espressif/arduino-esp32 #4949: If bluetooth disconnected when BluetoothSerial::write() is working, BluetoothSerial::write() does not working but BluetoothSerial::read() works.

JinLuckyboy, 2021-03-15:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			master 2021-03-14 (UTC)
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
If bluetooth disconnected when BluetoothSerial::write() is working, BluetoothSerial::write() does not working but BluetoothSerial::read() works.
In this case(disconnected when BluetoothSerial::write() is working),
Sometime BluetoothSerial::write() does not work without freeze.
Sometime BluetoothSerial::write() freeze.
See Below code.

Duplicate:
connect(bluetooth).
send "1"(bluetooth).
disconnect(bluetooth).
notice BluetoothSerial::write() work in loop() yet.
connect.
notice BluetoothSerial::write() freezes in loop().
send "3".
notice BluetoothSerial::read() works.
send "2".(or "1")
send "3".(or "2" or "1")
notice BluetoothSerial::write() freezes when you send "2".
disconnect.
connect.
send "3".
notice BluetoothSerial::read() works.
send "2".(or "1")
send "3".(or "2" or "1")
notice BluetoothSerial::write() freezes when you send "2".

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
#include "BluetoothSerial.h"

BluetoothSerial SerialBT;

bool taskBluetooth;
TaskHandle_t btHandle;

void setup() {
    Serial.begin(115200);
    
    SerialBT.enableSSP();
    SerialBT.register_callback(BTCallback);
    SerialBT.begin("Test");

    xTaskCreatePinnedToCore(TaskTimer0, "TaskTimer0", 1024, NULL, 1, NULL, 0);
    xTaskCreatePinnedToCore(TaskTimer1, "TaskTimer1", 1024, NULL, 1, NULL, 1);
}

void loop() { 
  Serial.println('a');
  SerialBT.print('a');
  Serial.println('A');
  delay(100);
}

void TaskBluetooth(void *pvParameters) {
  (void) pvParameters;

  taskBluetooth = true;

  for(;;) {
    if(!taskBluetooth) vTaskDelete(NULL);
    
    if (SerialBT.available()) {
      byte b = SerialBT.read();
      
      if(b == '1') {
        Serial.println("read 1 start");
        while(taskBluetooth) {
//          Serial.print('1');
          SerialBT.print('1');
//          delay(1000);
        }
        Serial.println("read 1 end");
      } else if(b == '2') {
        Serial.println("read 2");
        SerialBT.print('2');
      } else if(b == '3') {
        Serial.println("read 3");
      }
    }
  }
}

void BTCallback(esp_spp_cb_event_t event, esp_spp_cb_param_t *param){
  if(event == ESP_SPP_SRV_OPEN_EVT){
    Serial.println("Client Connected");
    
    while(SerialBT.read() >= 0);
    xTaskCreatePinnedToCore([](void *pvParameters){
      SerialBT.flush();
      Serial.println("flush end");
      vTaskDelete(NULL);
    }, "TaskFlush", 1024, NULL, 1, NULL, 1);
    xTaskCreatePinnedToCore(TaskBluetooth, "TaskBluetooth", 1024, NULL, 1, &btHandle, 1);
  }
 
  if(event == ESP_SPP_CLOSE_EVT ){
    Serial.println("Client disconnected");
    
    vTaskDelete(btHandle);
    taskBluetooth = false;
  }
}

void TaskTimer0(void *pvParameters) {
  for(;;) {
    Serial.println("TaskTimer0");
    delay(1000);
  }
}

void TaskTimer1(void *pvParameters) {
  for(;;) {
    Serial.println("TaskTimer1");
    delay(1000);
  }
}
```

me-no-dev, 2021-03-15:
can you please enable debug to level `Debug` and post the full UART log?

me-no-dev, 2021-03-16:
done :)

closed 2021-03-16 by me-no-dev