espressif/arduino-esp32 #4950: With BLE and WIFI coexisting allways STA_DISCONNECTED reason 8  and reboot after a few seconds

noukee, 2021-03-15:
Using NimBLE-Arduino and PubSubClient with WiFiClientSecure with TLS (and We are using PSRAM)

We can connect to the MTT broker but after a few seconds we always receive a SYSTEM_EVENT_STA_DISCONNECTED reason 8 that always reboot the system. 

If we do not use bluetooth the problem does not occur or when we use a very small bluetooth service. 

Can someone help us? 

platformio.ini

```
platform = espressif32
framework = arduino
board =  esp-wrover-kit
```
Customized SDKconfig to use PSRAM:

```
# PSRAM
CONFIG_SPIRAM_USE_CAPS_ALLOC=y
CONFIG_SPIRAM_CACHE_WORKAROUND=y

# SW COEXIST WIFI AND BLE
CONFIG_SW_COEXIST_ENABLE=y

# SINGLE CORE
CONFIG_FREERTOS_UNICORE=y

# TLS USES PSRAM
CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC=y
CONFIG_MBEDTLS_TLS_CLIENT_ONLY=y
```

[sdkconfig.TXT](https://github.com/h2zero/NimBLE-Arduino/files/6143750/sdkconfig.TXT)

Debug messages:

```
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[E][ssl_client.cpp:36] _handle_error(): [data_to_read():287]: (-76) UNKNOWN ERROR CODE (004C)
[D][mymqtt.cpp:102] task(): === MQTT NOT CONNECTED ===
[E] Received event:
[D][mywifi.cpp:85] reconnect(): [!] WIFI Disconnected
[E] Received event:
[!] MQTT Disconnected
[!] WIFI Disconnected
[D][mymqtt.cpp:108] task(): === MQTT RECONNECTING ===
[I][mymqtt.cpp:125] reconnect(): Attempting MQTT connection...
[I][mymqtt.cpp:126] reconnect(): Connecting to cloud.noukee.com
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[D][mywifi.cpp:85] reconnect(): [!] WIFI Disconnected
[E][ssl_client.cpp:98] start_ssl_client(): Connect to Server failed!
[D][mymqtt.cpp:223] sendData(): Sending message to MQTT topic gateway/setup/40f52055dd80
{
  "v
[E][WiFiClientSecure.cpp:127] connect(): start_ssl_client: -1
[!] WIFI Disconnected
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40094014  PS      : 0x00060230  A0      : 0x80122a30  A1      : 0x3ffe2670  
A2      : 0xfffffffe  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x3f402c5c  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0xffffffc0  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xfffffffc  LBEG    : 0x40094029  LEND    : 0x40094039  LCOUNT  : 0xfffffffd  
ELF file SHA256: 0000000000000000
Backtrace: 0x40094014:0x3ffe2670 0x40122a2d:0x3ffe2680 0x4012b84e:0x3ffe2990 0x4012b88a:0x3ffe2a20 0x400ddab7:0x3ffe2a60 0x400db111:0x3ffe2ac0 0x40095cae:0x3ffe2af0
Rebooting...

```

chegewara, 2021-03-15:
Few things:
- use backtrace decoder to understand why it crash,
- i can only guess, but logs are giving some clue, you are sending data when wifi is disconnected (no valid client)
```
[D][mymqtt.cpp:223] sendData(): Sending message to MQTT topic gateway/setup/40f52055dd80
{
  "v
  ```
- you didnt say how you are using BLE, maybe radio is occupied to much and wifi has no time to use it

noukee, 2021-03-16:
@chegewara decoded backtrace

```

decode ESP stack trace
Exception Cause: Not found
0x40094014: strlen at /home/mak/e/p/newlib_old/newlib_xtensa-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strlen.S:53
0x40094029: strlen at /home/mak/e/p/newlib_old/newlib_xtensa-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strlen.S:84
0x40094039: strlen at /home/mak/e/p/newlib_old/newlib_xtensa-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strlen.S:96
0x40094014: strlen at /home/mak/e/p/newlib_old/newlib_xtensa-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strlen.S:53
0x40122a2d: _svfprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:1529
0x4012b84e: _vsnprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:72
0x4012b88a: vsnprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:41
0x400ddab7: log_printf at /home/cs/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:533
0x400db111: noukee::mqtt::task(void*) at /home/cs/imatia/noukee/lib/mqttt/mymqtt.cpp:159
 (inlined by) noukee::mqtt::task(void*) at /home/cs/imatia/noukee/lib/mqttt/mymqtt.cpp:110
0x40095cae: vPortTaskWrapper at /Users/fjsualop/Code/esp/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:143
```
Sometimes the trace change to:

```
mbedtls
decode ESP stack trace
Exception Cause: Not found
0x4008efd5: lock_acquire_generic at /Users/fjsualop/Code/esp/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:143
0x40095180: invoke_abort at /Users/fjsualop/Code/esp/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:156
0x400953f9: abort at /Users/fjsualop/Code/esp/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:171
0x4008efd5: lock_acquire_generic at /Users/fjsualop/Code/esp/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:143
0x4008f101: _lock_acquire_recursive at /Users/fjsualop/Code/esp/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c:171
0x40125f1b: _vfiprintf_r at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vfprintf.c:860 (discriminator 2)
0x4012049d: fiprintf at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/fiprintf.c:50
0x40113ccd: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:59 (discriminator 8)
0x40099d91: multi_heap_free at /Users/fjsualop/Code/esp/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c:218 (discriminator 1)
0x400872ce: heap_caps_free at /Users/fjsualop/Code/esp/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:268
0x40087499: esp_mbedtls_mem_free at /Users/fjsualop/Code/esp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/port/esp_mem.c:35
0x40132c72: mbedtls_free at /Users/fjsualop/Code/esp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/platform.c:98
0x401c0d16: mbedtls_ssl_free at /Users/fjsualop/Code/esp/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:8992
0x4017d7ab: stop_ssl_socket(sslclient_context*, char const*, char const*, char const*) at /home/cs/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/ssl_client.cpp:272
0x4017d30c: WiFiClientSecure::stop() at /home/cs/.platformio/packages/framework-arduinoespressif32/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:87
0x401d6977: PubSubClient::connected() at /home/cs/imatia/noukee/.pio/libdeps/noukee/PubSubClient/src/PubSubClient.cpp:694
0x400dafc9: noukee::mqtt::task(void*) at /home/cs/imatia/noukee/lib/mqttt/mymqtt.cpp:102
0x40095cae:
``` vPortTaskWrapper at /Users/fjsualop/Code/esp/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:143

Is there any known issue to simultaneous use of BLE and TLS?  It seems TLS stuff seems to take too much time and BLE events interrupt it which stuffs up the stacks, however We're using NimBLE_Server with Advertisement (without BLE scan), should we stop advertisement to do MQTT? Because that happens without any BLE client

Thanks for your support

chegewara, 2021-03-16:
Yes, that is highly recommended to stop advertising, but it all depends what are advertising settings (window and interval).

It looks like something is trying to free memory that has been already freed, probably in mbed library, but i think for now it may help you solve the problem when you stop advertising, or at least please try that.

closed 2021-07-11 by stale[bot]