espressif/arduino-esp32 #4952: [BLE Client] Deadlock when calling writeValue after registerForNotify sometimes

wesee-dev, 2021-03-16:
Describe your system
Hardware: Custom ESP32 Module 
Computer: Windows 10
Core version: 1.0.4
Environment: Arduino IDE

### Hardware:
Board:							?SP32 Dev Module
Core Installation version:			1.04
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
My task locks when calling writeValue right after calling registerForNotify. 
This is very random and seems to happen most frequently when the client disconnects from the sever and reconnects, and then again tries to register for notifications from server and write a value in the characteristic. 
I got from the logs that when calling writeValue it waits forever for a mutex to be released:
m_semaphoreWriteCharEvt.take("writeValue");
It seems that registerForNotify  take the writeValue mutex and never releases it, probably waiting for an event release it.
When writeValue is called, it tries to take the mutex and keep waiting forever causing the deadlock.
As a temporary solution a set a task watchdog to panic and reset the device if the situation above happen, but this may be affecting other people projects that a reset is impracticable.
Also, the client is connected and the code is executed right after a scan and successful connection to server.

### Sketch:```cpp
bool WeAcessBLEClient::elevateConnectionStatus(notify_callback response_callback){

	LOG_MSG("[Info] Informing server that the device is a permanent provider...");

	if(!isClientConnected()){
		LOG_MSG("[Error] We are not connected, or the client is NULL");
		return false;
	}

	/* First we get the tag service */
	BLERemoteService *tag_serv =
			bleClient->getService(TAG_SERVICE);
	if(!tag_serv){
		LOG_MSG("[Error] Could not get service for TAGs");
		return false;
	}

	/* Then we write the permanent provider key in the related characteristic */
	BLERemoteCharacteristic *perm_char =
			tag_serv->getCharacteristic(PERM_TAG_PROVIDER_CHARACT);
	if(!perm_char){
		LOG_MSG("[Error] Could not get characteristic for permanent TAG provider.");
		return false;
	}

	/*Register to grab authentication status from server */
	perm_char->registerForNotify(response_callback);
	delay(100);
	LOG_MSG("Registered to server notifications!");
	/* At last We send the permanent provider key to server */
      *** LOCKS IN THE LINE BELOW ****
	perm_char->writeValue(BLE_PERM_CONN_PWD.c_str());
	LOG_MSG("[Info] Registering in server as a permanent provider...");
	return true;
}
```

h2zero, 2021-03-17:
I really need to find my old code fixes for this library and PR them.

The issue here is your device has been disconnected while you were waiting for the write to complete and are now stuck on the semaphore as that event is not handled properly to release it in the BLEChracteristic class.

Here is the fix; Add this code to [this line](https://github.com/espressif/arduino-esp32/blob/a451c9ef0de45fee5f73b77f73723566b6949eac/libraries/BLE/src/BLERemoteCharacteristic.cpp#L248):
```
case ESP_GATTS_DISCONNECT_EVT: {
	m_semaphoreConfEvt.give();
	break;
}
```
That should resolve this issue.

wesee-dev, 2021-03-17:
Thank you for providing the fix. 
It would be nice to have this fix included in the next release.

closed 2021-03-17 by wesee-dev

me-no-dev, 2021-03-17:
it will :)

h2zero, 2021-03-17:
Nice :)

ninja-, 2021-05-17:
nice