espressif/arduino-esp32 #4953: 'Unable to sync with device' error when using rosserial

akmakaiau, 2021-03-16:
### Hardware:
Board: DOIT ESP32 DEVKIT V1
IDE name: Arduino IDE 1.8.12
Flash Frequency: 80 MHz (I have also tried 40MHz)
PSRAM enabled: no
Upload Speed: 921600
Computer OS: Ubuntu 18.04

### Description:
I am using an ESP32 with the HelloWorld example sketch from the ros_lib library. When I start the node using:
````cpp
rosrun rosserial_python serial_node.py _port:=/dev/ttyUSB0 _baud:=115200
````
I get the following output and error:

````cpp
[INFO] [1615876946.675566]: ROS Serial Python Node
[INFO] [1615876946.680395]: Connecting to /dev/ttyUSB0 at 115200 baud
[INFO] [1615876948.785994]: Requesting topics...
[ERROR] [1615876963.789433]: Unable to sync with device; possible link problem or link software version mismatch such as hydro rosserial_python with groovy Arduino
[INFO] [1615876963.792668]: Requesting topics...
````
And the "Unable to sync..." error repeats every time it goes back to "Requesting topics..."

I have tried changing the programmer, the baud rate (9600, 57600) and the USB cable I am using. I have tried everything in [this](https://answers.ros.org/question/210875/arduino-rosserial-unable-to-sync-with-device/) thread, except for changing the size of the message, but I believe this is already a small message. Does anyone know how to resolve this error so that the topic get published? Please let me know if there is any more information needed.

### Sketch:
````cpp
/*
 * rosserial Publisher Example
 * Prints "hello world!"
 */

#include <ros.h>
#include <std_msgs/String.h>

ros::NodeHandle  nh;

std_msgs::String str_msg;
ros::Publisher chatter("chatter", &str_msg);

char hello[13] = "hello world!";

void setup()
{
  nh.initNode();
  nh.advertise(chatter);
}

void loop()
{
  str_msg.data = hello;
  chatter.publish( &str_msg );
  nh.spinOnce();
  delay(1000);
}
````

### Serial Monitor Output (Set to 115200 baud)
This output only appears once:
[](url)
````cpp
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
````

closed 2021-07-09 by stale[bot]

Bopp1312, 2021-08-23:
I had the same issue, in the ros_lib directory there is a file called ros.h 
I modified my file to look like this so that the serial interface was used rather than the TCP one. 
`#if defined(ESP8266) or defined(ESP32) or defined(ROSSERIAL_ARDUINO_TCP)
  //#include "ArduinoTcpHardware.h"
  #include "ArduinoHardware.h"
#else
  #include "ArduinoHardware.h"
#endif`

Mjidda, 2022-01-05:
i have the same issue too. i couldnt resolve it sadly enough. were u able to @Bopp1312?