espressif/arduino-esp32 #4956: Can't connect directly to ESP using ethernet cable

GeorgeFlorian, 2021-03-16:
Hello.

I have an [ESP32-EVB](https://www.olimex.com/Products/IoT/ESP32/ESP32-EVB/open-source-hardware) capable of ethernet connection.
I have configured it to be in Station Mode with a given Static IP. I am trying to connect to it using an Ethernet cable from my laptop to the ESP but I can't find it at the given IP address.

How can I connect to the ESP without a router/modem ?

felmue, 2021-03-16:
Hello @GeorgeFlorian 

do you mind sharing a little more information?

- what is the static IP address of your ESP32?
- what IP address does your laptop have?
- how does your laptop get its IP address? Static or DHCP?
- Are the two devices (laptop and ESP32) in the same network?
- And while you have your ESP32 connected via Ethernet to your laptop, does your laptop still have a WiFi connection as well? If yes, does it help if you temporarily disconnect from WiFi?

Thanks
Felix

GeorgeFlorian, 2021-03-24:
@felmue 

Here is the code:
```
void WiFiEvent(WiFiEvent_t event) {
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname(host_name);
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      eth_connected = true;
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
      }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
  }
}

void EthernetConfig(String x[]){
  if(x[0] == "WiFi") return;
  ETH.begin();
  int ki = 0;
  while(!eth_connected && ki <20) {
    Serial.println("Establishing ETHERNET Connection ... ");
    delay(1000);
    ki++;
  }
  if(!eth_connected) {
    logOutput("(1) Could not access Network ! Trying again...");
    logOutput("Controller will restart in 5 seconds !");
    delay(5000);
    ESP.restart();
  }
  if(x[1] != NULL &&      //Local IP
     x[1].length() != 0 &&
     x[2] != NULL &&      // Gateway
     x[2].length() != 0 &&
     x[3] != NULL &&      // Subnet
     x[3].length() != 0 &&
     x[4] != NULL &&      // DNS
     x[4].length() != 0) {
       local_IP_STA.fromString(x[1]);
       gateway_STA.fromString(x[2]);
       subnet_STA.fromString(x[3]);
       primaryDNS.fromString(x[4]);
       if(!ETH.config(local_IP_STA, gateway_STA, subnet_STA, primaryDNS)) {
         logOutput("Couldn't configure STATIC IP ! Obtaining DHCP IP !");
       }
       delay(50);
  } else {
    logOutput("Obtaining DHCP IP !");
  }
  local_IP_STA = ETH.localIP();
  gateway_STA = ETH.gatewayIP();
  subnet_STA = ETH.subnetMask();
  primaryDNS = ETH.dnsIP();
  logOutput((String)"IP addres: " + local_IP_STA.toString());
  logOutput((String)"Gateway: " + gateway_STA.toString());
  logOutput((String)"Subnet: " + subnet_STA.toString());
  logOutput((String)"DNS: " + primaryDNS.toString());
}

void setup() {
// ...
  WiFi.onEvent(WiFiEvent);
// ...
EthernetConfig(v);

}
```

I am reading the IP values from a text file into `v` array, which I then pass to `EthernetConfig` method.  

I am literally connecting the ESP to the PC using an Ethernet cable. There is no network. The devices are not in a network. They are connected to each other.

When this happens, the PC doesn't have any IP address and the ESP only outputs:
```
Establishing ETHERNET Connection ... 
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 21 - ETH_START
ETH Started
Establishing ETHERNET Connection ... 
Establishing ETHERNET Connection ... 
```

felmue, 2021-03-26:
Hello @GeorgeFlorian 

thank you for sharing your code. I do not have this board so I cannot test it myself, but here are some points that might help:
- `ETH.begin()` tries to get an IP address via DHCP, in your setup there is no router and no DHCP server therefore that will never succeed. And that is why your code is stuck. I suggest you try to call `ETH.config()` first and then call `ETH.begin()`. This should give your board a static IP and the underlying code should then not try to do DHCP.
- You'll need to assign your PC a static IP as well. Without IP address devices on an TCP/IP network cannot communicate. (That is true even if only two devices are connected with an Ethernet cable, like in your case.)

If the above doesn't help I suggest you start with a ready made example for your board using a router and DHCP to get more familiar with the hardware. This would also verify that a) the board is ok and works and b) the Ethernet cable is ok as well.

As a next step I 'dsuggest you modify the example to still use a router but with the board using a static IP. If that works as well you can go back to your minimalistic setup. The only change there is then the PC with its static IP.

Thanks
Felix

GeorgeFlorian, 2021-03-26:
@felmue 

I've tried calling `ETH.config()` before `ETH.begin()` some time ago (NOV 2019) and I've made a [post on ESP32 forum](https://www.esp32.com/viewtopic.php?f=19&t=13254) noting that if I put `ETH.config()` before `ETH.begin()` the code crashes admirably:
```
E (1384) tcpip_adapter: tcpip_adapter is not initialized!
abort() was called at PC 0x4011615b on core 1

Backtrace: 0x4008c618:0x3ffb1c20 0x4008c849:0x3ffb1c40 0x4011615b:0x3ffb1c60 0x400d8047:0x3ffb1ca0 0x400d2ae1:0x3ffb1cf0 0x400d34e2:0x3ffb1d70 0x400e2c93:0x3ffb1fb0 0x40088d05:0x3ffb1fd0

Decoding stack results
0x400d8047: esp_task_wdt_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/task_wdt.c line 206
0x400d2ae1: setCpuFrequencyMhz at /home/george/Arduino/hardware/espressif/esp32/cores/esp32/esp32-hal-cpu.c line 133
0x400d34e2: uw_update_context_1 at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/unwind-dw2-xtensa.c line 394
0x400e2c93: _svfprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 1536
0x40088d05: vTaskPlaceOnEventList at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/tasks.c line 2923
```

And, as noted in that post: calling `ETH.begin()` before `ETH.config()` gets an IP from the DHCP server then it changes that to what ever value I give it.

I don't know if this was fixed or not. Since no one answered on that post I've stopped trying to `config` before `begin`. I will try it again next week.

TD-er, 2021-03-27:
When using static IP you won't receive the event  `SYSTEM_EVENT_ETH_GOT_IP`

N.B. receiving ETH events do seem a bit 'flaky' to me.
For example unplugging the cable will trigger `SYSTEM_EVENT_ETH_DISCONNECTED` just fine and plugging in will give `SYSTEM_EVENT_ETH_CONNECTED`
But not when you later re-plug the cable.

If someone has any idea how to reset this, I would be very pleased to know.

lbernstone, 2021-03-27:
The network layer was completely reworked in IDF4.  Give the idf-release/v4.2 branch a look and see if it is more responsive to those events.

GeorgeFlorian, 2021-03-28:
> The network layer was completely reworked in IDF4. Give the idf-release/v4.2 branch a look and see if it is more responsive to those events.

What about Arduino Core ? I've never used IDF.

closed 2021-07-11 by stale[bot]