espressif/arduino-esp32 #4961: WIFI AP+STA DHCP crash on certain ESP32 modules

WebDust21, 2021-03-18:
### Hardware:
|||||||
|:---|---|---|---|---|---|
|<B>Board</B>ESP32-WROOM-32 module on custom PCB (designed with default KiCad symbol, manufactured with JLCPCB)|
|<B>Version</B>|1.0.4|
|<B>IDE name</B>|Arduino IDE 1.8.11|
|<B>Flash Frequency</B>|80Mhz|
|<B>PSRAM enabled</B>|no|
|<B>Upload Speed</B>|460800|
|<B>Computer OS</B>|Ubuntu 16.04 LTS x64|

### Description:
**Problem overview:**
I have well over a hundred boards in the field using the ESP32-WROOM-32 module.  95% of them work perfectly.  Most of the problematic ones with this (fairly fatal) problem can be caught in testing before the boards get shipped out.  The solution thus far is to simply replace the ESP32 module--this so far has resolved the issue every single time.
That alone makes me think that it is a hardware issue with the soldering / overheating the ESP32 module.  (There have been a couple boards that have developed this problem over time.  Frustrating, to be sure.)  We recently switched from hand-soldering the ESP32 modules with a hot-air gun, to having them properly assembled via the SMD factory...thinking this would be the Holy Grail solution.

Whoops, not so fast.  Had one SMD reflow-oven ESP32 board fail...AFTER being tested functional.

**What happens:**

1. User requests "WiFi setup mode."
2. ESP32 generates a hotspot (STA mode) for the user to connect to.  
3. User finds said hotspot, and tries to connect it with an smartphone.  (Does not seem to matter what kind of device tries to connect....Android phone, iPhone, laptop)
4. ESP32 instantly crashes, and restarts (connection never completes).  100% of the time completely reproducible on faulty units.

Yes, this sounds like a hardware problem.  But my arguments against a hardware fault are:
1. If it was indeed a hardware fault, wouldn't it fail when entering AP mode?  It's already transmitting data for the smartphone to be able to recognize the ESP32's network.  (If some of the internal components shifted during soldering, and shorted the antenna circuit out, I could easily see it crash faulting.  But if that was the case, it would not be able to generate an AP.)
2. Conversely, if it was a PCB fault, replacing the ESP32 would not solve the problem.  (But replacing the ESP32 is presently the solution.)
3. If it was an ESP32 hardware fault, I would expect there to be either no crash message, OR the crash message to be completely random every single time (i.e. CPU "mistake" due to power supply brownout, etc.)  But the crash report points to exactly the SAME place EVERY time--indicating that even if there is a hardware fault, it is triggered by software.

I did notice today that ESP32 firmware 1.0.5 has been released, which I am not using here.  Maybe a problem of this sort has already been addressed?

### Sketch:  
fragments of it anyway.  Please note that all of the boards have been encrypted with an exported ESP-IDF bootloader that had encryption enabled.  However, if the ESP32 is connected to WiFi, I can upload different firmware to it for testing.

```cpp
....
IPAddress apIP(192, 168, 1, 1);
....

if (WiFi.status() != WL_CONNECTED) {WiFi.disconnect();}   // doesn't shut anything else off.
    
      //specify AP mode.
      WiFi.mode(WIFI_AP_STA);   //Note: this is the AP + Station mode, so it has full capability of dual network connection.
      WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
      WiFi.softAP(APName, NULL, 1, 0, 1);    // NULL = no password, channel "1", not hidden, and only 1 device connection permitted

```
Please note that I did not provide more code because the crash log does not reference ANY of my project code.  It appears to be entirely internal to the ESP-IDF framework, and it is crashing long before any of the AP handling code ever gets called.  (And yes, the SPIFFS AP data is present, otherwise the board will throw an error at startup.)

### Debug Messages:
It always crashes in exactly the same place.  Here are 2 crash logs.  It is interesting to note that the number of DCHP messages are different between the 2...however, I would assume that this has to do with the RTOS and RS-232 buffers.

> dhcps: send_offer>>udp_sendto result 0
> Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
> Core 0 register dump:
> PC      : 0x00000000  PS      : 0x00060630  A0      : 0x80152979  A1      : 0x3ffd9660  
> A2      : 0x3ffee184  A3      : 0x3ffed8dc  A4      : 0x3ffdffec  A5      : 0x3ffedf4c  
> A6      : 0x0201a8c0  A7      : 0x0c01a8c0  A8      : 0x8015281c  A9      : 0x3ffd9620  
> A10     : 0x3ffee194  A11     : 0x3ffed8dc  A12     : 0x3ffd966c  A13     : 0x00000044  
> A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
> EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  
> 
> Backtrace: 0x00000000:0x3ffd9660 0x40152976:0x3ffd96a0 0x4015fa31:0x3ffd96c0 0x40164a39:0x3ffd9700 0x40169cd6:0x3ffd9720 0x4015337f:0x3ffd9740 0x40089469:0x3ffd9770
> 
> Rebooting...
> _[startup gibberish, etc.]_
> dhcps: send_offer>>udp_sendto result 0
> dhcps: send_offer>>udp_sendto result 0
> Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
> Core 0 register dump:
> PC      : 0x00000000  PS      : 0x00060630  A0      : 0x80152979  A1      : 0x3ffd9660  
> A2      : 0x3ffedd1c  A3      : 0x3ffedf50  A4      : 0x3ffb5948  A5      : 0x3ffedcfc  
> A6      : 0x0201a8c0  A7      : 0x0c01a8c0  A8      : 0x8015281c  A9      : 0x3ffd9620  
> A10     : 0x3ffedd2c  A11     : 0x3ffedf50  A12     : 0x3ffd966c  A13     : 0x00000044  
> A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
> EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  
> 
> Backtrace: 0x00000000:0x3ffd9660 0x40152976:0x3ffd96a0 0x4015fa31:0x3ffd96c0 0x40164a39:0x3ffd9700 0x40169cd6:0x3ffd9720 0x4015337f:0x3ffd9740 0x40089469:0x3ffd9770
> 
> Rebooting...

Stack trace decode (both stack traces are identical):
 
> PC: 0x00000000
> EXCVADDR: 0x00000000
> 
> Decoding stack results
> 0x40152976: handle_dhcp at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/apps/dhcpserver/dhcpserver.c line 1019
> 0x4015fa31: udp_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/udp.c line 394
> 0x40164a39: ip4_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/ipv4/ip4.c line 733
> 0x40169cd6: ethernet_input at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/netif/ethernet.c line 172
> 0x4015337f: tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c line 125
> 0x40089469: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
> 

PC at zero looks quite suspicious.

Can anyone shed any light on this?  Is there anything that I can provide that would help further?  I'm sure there's a way to improve the firmware so whatever error is encountered, won't cause a crash (even if it has to return an error code and fail).

Yes, I know this is a type of problem that developers don't like to see, maybe that's why I'm passing it up the chain ;-).  Haha, thanks in advance for any help.

me-no-dev, 2021-03-18:
1. Very strange issue indeed... the stack trace does not point to a hardware issue per say, but also quite weird that happens on some boards and not others
2. You should definitely try the current `master` or 1.0.6 when it comes out some time next week (based on IDF v3.3.5) on some "faulty" modules. Even 1.0.5 should be OK to test with, given that you are not using BT/BLE as well and that you are aware that some things might not work as expected (some things broke along the way but are fixed in `master`)
3. Issue does not seem to be related to power or RF. AP was already running, Device has already connected and the ESP has already sent DHCP offer. Best option is to try and reproduce with current arduino-esp32, so that current ESP-IDF version is referenced and new log is obtained (we can look at the dhcps source and try to figure out where the exception happens). Best case is, it will already be fixed.

WebDust21, 2021-03-19:
Thanks for your reply!

I am not using BT/BLE at all.  
Other peripherals in use include 2 of the 3 serial ports...LED PWM modules (4 channels), external PLL (rtc_clk_apll_enable), I2S (I2C), and ADC.  I'm using 4 threads on the RTOS.

Have to admit I don't have many "faulty" modules, as I've always X-marked the ESP32s, and had them replaced ;-)

I will see if I can get the latest faulty board back...maybe 1.0.6 will be released by then, and I can do a full update/test report with the latest core.

me-no-dev, 2021-03-19:
That would be great :)

WebDust21, 2021-03-28:
So I got a faulty board back (had over 4,000 hours on it)...verified that it crashed with the exact same DHCP fault:

`dhcps: send_offer>>udp_sendto result 0
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x00000000  PS      : 0x00060d30  A0      : 0x8014edd5  A1      : 0x3ffafbb0  
A2      : 0x3ffee1c4  A3      : 0x3ffee3f8  A4      : 0x3ffee840  A5      : 0x3ffee820  
A6      : 0x0204a8c0  A7      : 0x6504a8c0  A8      : 0x8014ec78  A9      : 0x3ffafb70  
A10     : 0x3ffee1d4  A11     : 0x3ffee3f8  A12     : 0x3ffafbbc  A13     : 0x00000044  
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffafbb0 0x4014edd2:0x3ffafbf0 0x4015be8d:0x3ffafc10 0x40160e95:0x3ffafc50 0x40166132:0x3ffafc70 0x4014f7db:0x3ffafc90 0x40089469:0x3ffafcc0
`

...managed to get an updated firmware installed (based on the new ESP32-Arduino Core 1.0.6)...and...

**THE PROBLEM IS GONE!**

Thanks a million!
Have at least 2 more boards to test the fix on (hopefully tomorrow)...if the update miraculously fixes both of them, I will consider this issue closed.

me-no-dev, 2021-03-28:
Awesome :) Please close the issue if you find everything working :)

WebDust21, 2021-04-03:
One more board successfully updated to the 1.0.6 core...problem disappeared.  I will consider this issue successfully closed.
Thanks!

closed 2021-04-03 by WebDust21