espressif/arduino-esp32 #4964: Second and third NTP servers are not working

szerwi, 2021-03-21:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.5
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
I am using SimpleTime example. I have added second and third NTP server, but when I change first NTP server's address to wrong one, it cannot obtain time. ```configTzTime()``` and ```configTime()``` are not using second and third NTP servers.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
#include <WiFi.h>
#include "time.h"

const char* ssid       = "xxx";
const char* password   = "xxx";

const char* ntpServer = "wrong.ntp.org";
const char* ntpServer2 = "pl.pool.ntp.org";
const char* ntpServer3 = "3.pl.pool.ntp.org";
const long  gmtOffset_sec = 3600;
const int   daylightOffset_sec = 3600;

void printLocalTime()
{
  struct tm timeinfo;
  if(!getLocalTime(&timeinfo)){
    Serial.println("Failed to obtain time");
    return;
  }
  Serial.println(&timeinfo, "%A, %B %d %Y %H:%M:%S");
}

void setup()
{
  Serial.begin(115200);
  
  Serial.printf("Connecting to %s ", ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
  }
  Serial.println(" CONNECTED");
  
  configTzTime("CET-1CEST,M3.5.0,M10.5.0/3", ntpServer, ntpServer2, ntpServer3);
  //configTime(gmtOffset_sec, daylightOffset_sec, ntpServer, ntpServer2, ntpServer3); //same situation
  printLocalTime();

  WiFi.disconnect(true);
  WiFi.mode(WIFI_OFF);
}

void loop()
{
  delay(1000);
  printLocalTime();
}
```

### Debug Messages:
```
Connecting to xxx [D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.1.113, MASK: 255.255.255.0, GW: 192.168.1.1
. CONNECTED
Failed to obtain time
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 3 - STA_STOP
Failed to obtain time
Failed to obtain time
Failed to obtain time
Failed to obtain time

```

lbernstone, 2021-03-21:
Support for multiple ntp servers is not compiled into the arduino libraries.  You would need to roll your own to get this.
https://github.com/espressif/arduino-esp32/blob/1.0.5/tools/sdk/include/lwip/lwip/apps/sntp_opts.h#L58 or https://github.com/espressif/arduino-esp32/blob/1.0.5/tools/sdk/include/lwip/lwip/opt.h#L887

szerwi, 2021-03-21:
@lbernstone Thanks for the information. Do you plan to add it in the future?

lbernstone, 2021-03-21:
That comes from upstream (lwip), but I seriously doubt it.  It is a pretty good chunk of memory that most people won't use.  Using a pool address from ntp.org is pretty reliable.

fthiebolt, 2021-04-17:
Hello, i was facing the same issue ... moreover, i really need NTP FROM DHCP answer (yes our university is blocking outgoing ntp packets !!)
After struggling for a while, i was able to recompile lwip with support for 3 NTP servers along with NTP from DHCP answer:
```
git clone -b v3.3.5 --recursive https://github.com/espressif/esp-idf.git esp-idf-v3.3.5
export IDF_PATH=${HOME}/esp-idf-v3.3.5
cd esp-idf-v3.3.5
export IDF_TARGET="esp32"
./install.sh
. ./export.sh
cd examples/get-started/blink
cp ~/.arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/sdkconfig sdkconfig
make menuconfig
    <load sdkconfig>
vim components/lwip/lwip/src/include/lwip/opt.h
    880 #define LWIP_DHCP_GET_NTP_SRV           1
    887 #define LWIP_DHCP_MAX_NTP_SERVERS       3
idf.py build
```
Then you need to backport those mods to your arduino release:
```
apply same opt.h mods to:
    ~/.arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/lwip/lwip/
cp build/esp-idf/lwip/liblwip.a ~/.arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/li
```
need to mention i'm using 1.0.6 esp32 arduino so i chose esp-idf 3.3.5

Of course, i would have loved a much more simpler way to achieve this simple lwip mod (fairly simple on esp8266-arduino)
Hope this helps

Rotzbua, 2021-04-29:
I think it should be mentioned in the examples like:
https://github.com/espressif/arduino-esp32/blob/7856de7a57420e494176c16c5138174fe2c1dad0/libraries/HTTPClient/examples/BasicHttpsClient/BasicHttpsClient.ino#L59

Also if possible a compile warning or runtime log would be useful.

vortigont, 2021-06-28:
Ugh! This really took me a while to find out why the heck a function with 3 ntp server params just silently ignores two of it. At least a note in source comments would be nice.

closed 2021-09-19 by stale[bot]