espressif/arduino-esp32 #4965: ESP32 crash on SerialBT.begin() after a Wifi.setSleep(false)

mstrens, 2021-03-21:
Arduino 1.8.5
ESP32:  dev kit 1 (compile using wmos lolin32)
Version : development version installed from github on 20 march 2021

I had a sketch that was running with release 1.0.4 but SerialBT.connect() always returned true.
To fix it, I installed release 1.0.5 but then ESP32 crashed (on SerialBT.begin() I presume).
I installed development version but ESP32 still crashed.

I search a lot but I think I can reproduce the error with a minimal sketch.
In this minimal sketch, when the line "WiFi.setSleep(false);" is not commented, ESP32 crash.
When the line is commented, the script runs.
Note: when running this sketch, the wifi parameters are good for my access point and ESP32 can connect as STATION but I use a dummy  name for the other BT device in "SerialBT.connect("ESP32_BT")". Not sure if this play a role for crashing. 
Here the minimal sketch:
`
#include <WiFi.h>              // Built-in
#include <WebServer.h>
#include <BluetoothSerial.h>

BluetoothSerial SerialBT;
 
void setup() {
  uint8_t initWifiCnt = 40 ;
  bool btConnected;

  Serial.begin(115200); // init UART for debug and for Gcode passthrough via USB PC
  WiFi.begin("bbox2-58c4" , "your_password");   
  while (WiFi.status() != WL_CONNECTED)  { 
    Serial.print(".");       
    delay(250); 
    initWifiCnt--;
    if ( initWifiCnt == 0) {
      break;
    }
  }
  
  WiFi.setSleep(false);
  
  Serial.println("SerialBT.begin will be executed");
  SerialBT.begin("ESP32testm", true); 
  Serial.println("SerialBT.begin has been executed");
  btConnected = SerialBT.connect("ESP32_BT");
  if(btConnected) {
    Serial.println("Connected Succesfully in BT!");
  } else {
    Serial.println("not connected:  start while");
    while(!SerialBT.connected(2000)) {
      Serial.println("Failed to connect. Make sure remote device is available and in range, then restart app."); 
    }
    Serial.println("end of while !SerialBT.connected");
  }
}

void loop() {
  delay(500);
}
` 
What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2021-03-21:
Does it crash if you switch the order and init BT before WiFi?

mstrens, 2021-03-21:
When I change the sketch like this, it is still crash but with an error message:
`#include <WiFi.h>              // Built-in
#include <WebServer.h>
#include <BluetoothSerial.h>

BluetoothSerial SerialBT;
 
void setup() {
  uint8_t initWifiCnt = 40 ;
  bool btConnected;
  Serial.begin(115200);
  Serial.println("SerialBT.begin will be executed");
  
  SerialBT.begin("ESP32testm", true); 
  Serial.println("SerialBT.begin has been executed");
  btConnected = SerialBT.connect("ESP32_BT");
  
  Serial.begin(115200); // init UART for debug and for Gcode passthrough via USB PC
  WiFi.begin("bbox2-58c4" , "alineloriejulie");   
  while (WiFi.status() != WL_CONNECTED)  { 
    Serial.print(".");       
    delay(250); 
    initWifiCnt--;
    if ( initWifiCnt == 0) {
      break;
    }
  }
  
  WiFi.setSleep(false);
  
  if(btConnected) {
    Serial.println("Connected Succesfully in BT!");
  } else {
    Serial.println("not connected:  start while");
    while(!SerialBT.connected(2000)) {
      Serial.println("Failed to connect. Make sure remote device is available and in range, then restart app."); 
    }
    Serial.println("end of while !SerialBT.connected");
  }
}

void loop() {
  delay(500);
}`

Here the output

`SerialBT.begin will be executed
SerialBT.begin has been executed
`

me-no-dev, 2021-03-23:
remove `WiFi.setSleep(false);`?

mstrens, 2021-03-23:
Yes, it is what I did.
In previous version, this sentence was recommended. That is the reason why I used it.
So, for compatibility, it would perhaps be good to discard this instruction (or this value) in new version.

c8528830, 2021-06-26:
Maybe is voltage too LOW, try to use 5V -> VIN.

closed 2021-09-19 by stale[bot]