espressif/arduino-esp32 #4967: can not reconnect BluetoothSerial at 1.0.5 and hang problem at 1.0.4

smarq8, 2021-03-22:
### Hardware:
Board:							TTGO T-DISPLAY using board "ESP32 Dev module"
Bluetooth device:                                     Android + "Serial Bluetooth Terminal" or "RoboRemo"
Core Installation version:			1.0.5
IDE name:						Arduino IDE / PlatformIO
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Upload Speed:					?
Computer OS:                                         Windows 7 x64

### Description:
At the version 1.0.5 after connection lost/disconnect I'm not able to acquire connection anymore without restarting board.
At the version 1.0.4 under heavy load system there was similar problem but I fixed them with 2 patches described below.
Same thing happen using PlatformIO + arduino Framework 1.0.5, 1.0.4 not tested yet but probably same effect as Arduino IDE.
Currenlty 1.0.5 BluetoothSerial prevent from using it because I need to connect my device several times and I can not afford to hang my sketch in the middle of job (1 hour) or being unable to connect and lost control only beacuse i lost connection.

these 2 patches are mandatory to avoid most of troubles and I'm wonder why they was not implemented
[patch1](https://github.com/jcdevel/arduino-esp32/commit/9508b8bfffaa84c548c59fb10dc061ebd55cf3b4)
[patch2](https://github.com/jcdevel/arduino-esp32/commit/0f7de4de91fab00cfed19699b7eb24e4037380a2)
[from here](https://github.com/espressif/arduino-esp32/issues/4594#issue-753397538)

Sorry for my english

### Sketch:
```
#include <Arduino.h>
#include <BluetoothSerial.h>

BluetoothSerial BT;

void setup() {
  Serial.begin(115200);
  BT.begin("esp32Test");
}

void loop() {
//    vTaskDelay(200); // for 1.0.5 it does not help to unload system
    Serial.println(millis());
    BT.printf("time: %u\n"
    "pidSV %.3f\n"
    "pidPV %.3f\n"
    "pidP %.3f\n"
    "pidI %.3f\n"
    "pidD %.3f\n"
    "pidO %.3f\n"
    "fileCounter %i\n"
    "TEMPPERS_R %.2f\n"
    "fileSize %.2f\n"
    "programProc %.3f\n\n"
    ,millis()
    ,1.0
    ,2.0
    ,3.0
    ,4.0
    ,5.0
    ,6.0
    ,7.0
    ,8.0
    ,9.0
    ,10.0);
}
```

### Debug Messages:
```
Arduino framework v1.0.4
hang main loop right on second connect, sometimes also it hang right after disconnect

[V][BluetoothSerial.cpp:271] esp_spp_cb(): ESP_SPP_WRITE_EVT: 330 FREE
7349
[V][BluetoothSerial.cpp:271] esp_spp_cb(): ESP_SPP_WRITE_EVT: 330 FREE
7356
[V][BluetoothSerial.cpp:271] esp_spp_cb(): ESP_SPP_WRITE_EVT: 330 FREE
7363
[V][BluetoothSerial.cpp:271] esp_spp_cb(): ESP_SPP_WRITE_EVT: 330 FREE
7363
[I][BluetoothSerial.cpp:247] esp_spp_cb(): ESP_SPP_CLOSE_EVT // disconnect
7375
7375
.......
11206
11206
[I][BluetoothSerial.cpp:235] esp_spp_cb(): ESP_SPP_SRV_OPEN_EVT // reconnect, not anymore Serial.print() or BT.print() but BT library still respond to connect/disconnect
[I][BluetoothSerial.cpp:247] esp_spp_cb(): ESP_SPP_CLOSE_EVT    // disconnect
[I][BluetoothSerial.cpp:235] esp_spp_cb(): ESP_SPP_SRV_OPEN_EVT // reconnect
[I][BluetoothSerial.cpp:247] esp_spp_cb(): ESP_SPP_CLOSE_EVT    // disconnect
[I][BluetoothSerial.cpp:235] esp_spp_cb(): ESP_SPP_SRV_OPEN_EVT // reconnect
[I][BluetoothSerial.cpp:247] esp_spp_cb(): ESP_SPP_CLOSE_EVT    // disconnect
[I][BluetoothSerial.cpp:235] esp_spp_cb(): ESP_SPP_SRV_OPEN_EVT // reconnect
```

```
Arduino framework 1.0.4 + 2 patches
in this case everything seems to work fine

[V][BluetoothSerial.cpp:273] esp_spp_cb(): ESP_SPP_WRITE_EVT: 330 FREE
33348
[V][BluetoothSerial.cpp:273] esp_spp_cb(): ESP_SPP_WRITE_EVT: 330 FREE
33355
[V][BluetoothSerial.cpp:273] esp_spp_cb(): ESP_SPP_WRITE_EVT: 330 FREE
33362
33363[V][BluetoothSerial.cpp:273] esp_spp_cb(): ESP_SPP_WRITE_EVT: 330 FREE
33369
33370
[V][BluetoothSerial.cpp:273] esp_spp_cb(): ESP_SPP_WRITE_EVT: 330 FREE
33377
33377
[V][BluetoothSerial.cpp:273] esp_spp_cb(): ESP_SPP_WRITE_EVT: 330 FREE
33384
[I][BluetoothSerial.cpp:249] esp_spp_cb(): ESP_SPP_CLOSE_EVT // disconnect
33385
33391
33391
...........
34631
34631
34632
[E][BluetoothSerial.cpp:165] _spp_send_buffer(): SPP Ack Failed!
34633
34639
34640
...........
34736
34736
34737
[E][BluetoothSerial.cpp:165] _spp_send_buffer(): SPP Ack Failed!
34737
34744
34744
...........
34841
34842
34842
[E][BluetoothSerial.cpp:165] _spp_send_buffer(): SPP Ack Failed!
34849
34849
34850
...........
34946
34947
34947
[E][BluetoothSerial.cpp:165] _spp_send_buffer(): SPP Ack Failed!
34953
34954
34955
...........
35495
35495
35496
[I][BluetoothSerial.cpp:237] esp_spp_cb(): ESP_SPP_SRV_OPEN_EVT // connect
35497
35504
35505
35505
[V][BluetoothSerial.cpp:273] esp_spp_cb(): ESP_SPP_WRITE_EVT: 144 FREE
35506
35512
[V][BluetoothSerial.cpp:273] esp_spp_cb(): ESP_SPP_WRITE_EVT: 288 FREE
35518
35519
[V][BluetoothSerial.cpp:273] esp_spp_cb(): ESP_SPP_WRITE_EVT: 288 FREE
35526
[V][BluetoothSerial.cpp:273] esp_spp_cb(): ESP_SPP_WRITE_EVT: 288 FREE
35533
```

```
Arduino framework 1.0.5
In this case after connection lost/disconnect I'm not able to connect again.
On my main program it also sometimes hang right after disconnect however I can not
reproduce it at above example or it happes too rarely. Those 2 patcher from earlier at 
this verion seems to not take any effects
..........
15730
15731
15732
[I][BluetoothSerial.cpp:240] esp_spp_cb(): ESP_SPP_SRV_OPEN_EVT // connect
15738
15738
15739
15740
15740
15741
[V][BluetoothSerial.cpp:276] esp_spp_cb(): ESP_SPP_WRITE_EVT: 144 FREE
15748
15748
[V][BluetoothSerial.cpp:276] esp_spp_cb(): ESP_SPP_WRITE_EVT: 330 FREE
15755
15756
..........
20715
20716
[V][BluetoothSerial.cpp:276] esp_spp_cb(): ESP_SPP_WRITE_EVT: 330 FREE
20723
20724
[V][BluetoothSerial.cpp:276] esp_spp_cb(): ESP_SPP_WRITE_EVT: 330 FREE
20730
20731
[V][BluetoothSerial.cpp:276] esp_spp_cb(): ESP_SPP_WRITE_EVT: 330 FREE
20738
20738
[V][BluetoothSerial.cpp:276] esp_spp_cb(): ESP_SPP_WRITE_EVT: 330 FREE
[I][BluetoothSerial.cpp:252] esp_spp_cb(): ESP_SPP_CLOSE_EVT // disconnect
20751
[V][BluetoothSerial.cpp:276] esp_spp_cb(): ESP_SPP_WRITE_EVT: 4294967295 FREE
[V][BluetoothSerial.cpp:276] esp_spp_cb(): ESP_SPP_WRITE_EVT: 4294967295 FREE
[V][BluetoothSerial.cpp:276] esp_spp_cb(): ESP_SPP_WRITE_EVT: 4294967295 FREE
[V][BluetoothSerial.cpp:276] esp_spp_cb(): ESP_SPP_WRITE_EVT: 4294967295 FREE
[V][BluetoothSerial.cpp:276] esp_spp_cb(): ESP_SPP_WRITE_EVT: 4294967295 FREE
[V][BluetoothSerial.cpp:276] esp_spp_cb(): ESP_SPP_WRITE_EVT: 4294967295 FREE
[V][BluetoothSerial.cpp:276] esp_spp_cb(): ESP_SPP_WRITE_EVT: 4294967295 FREE
[V][BluetoothSerial.cpp:276] esp_spp_cb(): ESP_SPP_WRITE_EVT: 4294967295 FREE
[V][BluetoothSerial.cpp:276] esp_spp_cb(): ESP_SPP_WRITE_EVT: 4294967295 FREE
[V][BluetoothSerial.cpp:276] esp_spp_cb(): ESP_SPP_WRITE_EVT: 4294967295 FREE
[V][BluetoothSerial.cpp:276] esp_spp_cb(): ESP_SPP_WRITE_EVT: 4294967295 FREE
[V][BluetoothSerial.cpp:276] esp_spp_cb(): ESP_SPP_WRITE_EVT: 4294967295 FREE
[V][BluetoothSerial.cpp:276] esp_spp_cb(): ESP_SPP_WRITE_EVT: 4294967295 FREE
[V][BluetoothSerial.cpp:276] esp_spp_cb(): ESP_SPP_WRITE_EVT: 4294967295 FREE
[V][BluetoothSerial.cpp:276] esp_spp_cb(): ESP_SPP_WRITE_EVT: 4294967295 FREE
20854
20855
20855
20856
20857
20857
20858
20858
...........
// unable to connect anymore, at my Bluetooth device i got timeout connection
```

cattledogGH, 2021-03-22:
>At the version 1.0.5 after connection lost/disconnect I'm not able to acquire connection anymore without restarting board. 

This is a duplicate of 4915. The fix is in the current master and should be implemented in 1.0.6

> At the version 1.0.4 under heavy load system there was similar problem

Regarding the 10..4 hang problem it would be worth testing with the current master.

viniciusdesa, 2021-03-23:
I'm getting this issue too.. looking fwd for the fix..

closed 2021-07-11 by stale[bot]