espressif/arduino-esp32 #4969: BLE with external Oscillator and Arduino Core

seackone, 2021-03-23:
### Hardware:
Board:							ESP32 Dev Module
IDE name:						Platform.io
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					921600
Computer OS:                                         Mac OSX

### Description:
Im using PlatformIO with ESP-IDF and Arduino instance because I had to do some settings in menuconfig. Its my first time, that im working with ESP-IDF but I think my problem is maybe caused by the Arduino instance. For my next project, I need to run an ESP32 BLE Server with very low power consumption and I still prefer Arduino because I never used ESP-IDF before. There is already a solution which I followed step by step.

https://github.com/espressif/esp-idf/issues/947#issuecomment-500312453

Im using an external crystal oscillator and it works. Otherwise I receive a message at the startup log:

`BTDM_INIT: 32.768kHz XTAL not detected, fall back to main XTAL as Bluetooth sleep clock. Light sleep mode will not be able to apply when bluetooth is enabled.`

The menuconfig is edited as described in the link above and I am running the Arduino BLE_server example with the additional esp_pm_config. If im running my sketch with an empty loop function, the serial monitor prints every 5 seconds:

```
(11666) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
(11666) task_wdt:  - IDLE1 (CPU 1)
(11666) task_wdt: Tasks currently running:
(11666) task_wdt: CPU 0: IDLE0
(11666) task_wdt: CPU 1: loopTask
```

I am able to connect my phone with the ESP32 but the avg current consumption is roughly about 64 mA. If I am adding a delay of lets say 1 second in the empty loop function, the watchdog message disappears and the average current consumption decreased to 40 mA if client is connected or 30 mA with a disconnected client. That was the best value I could achieve, which is far away of the expected 3 mA. 

@mywang-espressif also wrote:

> To test whether bluetooth enters sleep, there are two functions that you can modify. In source file bt.c of ESP-IDF: btdm_sleep_enter_phase2_wrapper and btdm_sleep_exit_phase3_wrapper. The two functions are two hooks for bluetooth enters and exits from modem sleep. You can add some logic to toggle GPIO output in the two functions and use logic analyzer to see whether the signal toggles.

This is what I also did. I was adding a GPIO high state to the btdm_sleep_enter_phase2_wrapper and a low state to the btdm_sleep_exit_phase as followed (only for entering sleep mode):

```cpp
static void btdm_sleep_enter_phase2_wrapper(void)
{
if (btdm_controller_get_sleep_mode() == BTDM_MODEM_SLEEP_MODE_ORIG) {

    gpio_pad_select_gpio(OUTPUT_GPIO);
    gpio_set_direction((gpio_num_t)OUTPUT_GPIO, GPIO_MODE_OUTPUT);
    gpio_set_level((gpio_num_t)27, 0);
....
....
```

I was measuring the GPIO with an oscilloscope and a client connected to the server:

[image]

Im not very familiar with the settings for FreeRTOS. @nywang-espressif wrote:

>  FreeRTOS will enter light sleep if no tasks need to run for 3(by default) ticks, that is 30ms if tick rate is 100Hz.

Im using the default settings but for my understanding of this sentence: The ESP32 is going to light sleep if there is nothing to do for a timeframe of 30ms. Is that correct? What I was measuring with the scope is a period of 30ms with a sleep time of approximately 26ms and an on time of 4ms. Could this be right? The client is connected to the device. I could change the ticks or the tick rate but after that, I had some issues to establish the bluetooth connection. It was not easy to capture a good picture with the scope if no client is connected (due to the switching frequency).

Even if I dont understand the FreeRTOS, it seems that bluetooth is going to sleep. Now im wondering, why I can not get the ~2 mA of the measuring results of @nywang-espressif:

> CONNECTION(connection_interval = 960.0ms, slave_latency = 0) average current 2.11mA, max 106.01mA, min 0.94mA

His description was for ESP-IDF only. Im using the Arduino instance at Core 1 and I am wondering if this is the reason, why the ESP32 is not going to light sleep as expected. By the way: how can I set the connection_interval of 960ms? Im very new to the bluetooth topic as well. It would be great to get some help to decrease my current consumption since I have no idea what else I could do.

Best regards,
Chris

### Sketch:
```cpp

#include <Arduino.h>
[... 93 lines cut ...]
␛[0;32mI (438) cpu_start: ESP-IDF:          3.40001.200521␛[0m
␛[0;32mI (444) cpu_start: Starting app cpu, entry point is 0x40081774␛[0m
␛[0;32mI (0) cpu_start: App cpu up.␛[0m
␛[0;32mI (454) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (461) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM␛[0m
␛[0;32mI (467) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM␛[0m
␛[0;32mI (473) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM␛[0m
␛[0;32mI (480) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM␛[0m
␛[0;32mI (486) heap_init: At 3FFC9230 len 00016DD0 (91 KiB): DRAM␛[0m
␛[0;32mI (492) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (498) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (505) heap_init: At 400946D0 len 0000B930 (46 KiB): IRAM␛[0m
␛[0;32mI (511) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (582) spi_flash: detected chip: generic␛[0m
␛[0;32mI (583) spi_flash: flash io: dio␛[0m
␛[0;32mI (583) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (0) cpu_start: Starting scheduler on APP CPU.␛[0m
Starting BLE
␛[0;32mI (30) BTDM_INIT: BT controller compile version [c1cbe45]␛[0m
␛[0;32mI (666) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE␛[0m
␛[0;32mI (786) phy: phy_version: 4180, cb3948e, Sep 12 2019, 16:39:13, 0, 0␛[0m
Characteristic defined! Now you can read it in your phone!
␛[0;32mI (1646) pm_esp32: Frequency switching config: CPU_MAX: 80, APB_MAX: 80, APB_MIN: 40, Light sleep: ENABLED␛[0m
␛[0;33mW (1656) BT_BTM: BTM_BleWriteScanRsp, Partial data write into ADV␛[0m
```

h2zero, 2021-04-01:
There are a lot of things going on to accomplish what you're looking for.

First is the connection parameters, if the client decides to use the ones suggested in your code that will result in possibly 7.5ms connection interval - very far from the 960 quoted above. Look up the server method `updateConnParams`. although this will ask the client for a connection parameter change, there is not guarantee of it, the client is in charge here. You can start with changing the `setMinPreferred` and `setMaxPreferred` to something like 80, 160 - these are 1.25ms units which correspond to 100ms and 200ms (BLE spec).

Second is the advertising interval, the methods `setMaxInterval` and `setMinInterval` control this. I don't remember the default values but if they are small you could be sending many advertising packets in a short period of time, increasing current consumption.

Third is the scan response, though a minor issue. If the scanning device detects a scan response is available it will request it, increasing the current consumed by the advertiser. If you do not need to provide data in a scan response you should change the `setScanResponse` parameter to false.

That should keep you busy for a while, I'll be interested to hear the results of your efforts.

eccessivo75, 2021-06-17:
@seackone
Is there any updates?

seackone, 2021-06-17:
@eccessivo75 
Unfortunately not. I have had little time to continue working on the project but will continue later this year.

eccessivo75, 2021-06-17:
@seackone 
Your work is really useful and probably you are solving one of the most important ble issue for arduino users. I will try to investigate and share. I hope to have news from you shortly .
P.s. i think that the different power consumption changing the delay time can be understood by watching this video: https://www.youtube.com/watch?v=k_D_Qu0cgu8
Thank you for your hard work

closed 2021-09-06 by stale[bot]